openscad/openscad #3023: ERROR: CGAL error in CGAL_Nef_polyhedron3():

mogra09, 2019-08-01:
Hello!
I have a problem.
Please help me to solve the problem.

platform: Windows10-64bit, OpenSCAD version 2019.05

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329

the code:
```
module cover()
{
 CubePoints = [
  [ 9.5, 0  ,1  ], //0
  [-9.5, 0  ,1  ], //1
  [-9.5,21.3,1  ], //2
  [-6.0,26  ,1  ], //3
  [ 6.0,26  ,1  ], //4
  [ 9.5,21.3,1  ], //5
  [ 1.8, 6  ,2.5], //6
  [-1.8, 6  ,2.5], //7
  [-1.8,15  ,2.5], //8
  [ 1.8,15  ,2.5], //9
  [ 9.5, 0  ,0  ], //10
  [-9.5, 0  ,0  ], //11
  [-9.5,21.3,0  ], //12
  [-6.0,26  ,0  ], //13
  [ 6.0,26  ,0  ], //14
  [ 9.5,21.3,0  ]  //15
 ];
 CubeFaces = [
  [0,1,7,6],[1,2,8,7],[2,3,8],[3,4,9,8],[4,5,9],[0,5,9,6],[6,7,8,9],
  [0,1,11,10],[1,2,12,11],[2,3,13,12],[3,4,14,13],[4,5,15,14],[5,0,10,15],
  [10,11,12,13,14,15]
 ];
 difference(){
  polyhedron( CubePoints, CubeFaces );
  hull(){
   translate([0, 6,-1]) cylinder(d=4, h=4, $fn=16 );
   translate([0,15,-1]) cylinder(d=4, h=4, $fn=16 );
  }
 }
} 
cover();
```

t-paul, 2019-08-01:
Have you checked that all faces have the correct winding order? -> [FAQ Entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?)

mogra09, 2019-08-01:
Hello t-paul
Thank you very much for your kind advice.
I don't understand the writing order and now solved with your advice.

```
 CubeFaces = [
  [6,7,1,0],[7,8,2,1],[8,3,2],[8,9,4,3],[5,4,9],[0,5,9,6],
  [0,1,11,10],[1,2,12,11],[2,3,13,12],[3,4,14,13],[4,5,15,14],[5,0,10,15],
  [9,8,7,6],
  [10,11,12,13,14,15]
 ];
```

closed 2019-08-01 by mogra09

nophead, 2019-08-02:
These issues crop up all time because presenting cgal exceptions to the
user looks like an internal program error. Can they be wrapped to day your
geometry is broken?