openscad/openscad #3031: OpenSCAD window display completely broken

thehans, 2019-08-05:
Just tried starting up the latest nightly OpenSCAD version (2019.08.05.nightly) on Linux Mint 19 and the openscad window comes up but does not draw anything inside the window manager decorations, the contents of the window are sort of just a screengrab of what was last on my desktop.
This is across the entire window, not any particular pane like editor or 3D view.

Starting from command line I see these messages:
[code]
hans@hans-ThinkPad-P50:~/projects/primes/mprime$ openscad-nightly
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
QOpenGLWidget: Failed to create context
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
[/code]

t-paul, 2019-08-05:
Please add the full library output.

thehans, 2019-08-05:
The top menu isn't displayed either so I can't tell where to click, but I was able use keyboard to navigate to help -> Library info and got this popup:
"A critical error was caught. The application may have become unstable:
basic_string::_M_construct null not valid"

using openscad --info from command line:
```
QObject::startTimer: Timers can only be used with threads started with QThread
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  25
  Current serial number in output stream:  26
OpenSCAD Version: 2019.08.05.nightly (git e731c04)
System information: Linux 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 Linux Mint 19 8 CPUs 62.74 GB RAM
User Agent: OpenSCAD/2019.08.05.nightly (git e731c04) (Linux x86_64; Linux Mint 19)
Compiler: GCC "7.3.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.5
QScintilla version: 2.10.2
InputDrivers: 
GLib version: 2.56.1
lodepng version: 20180910
libzip version: <not enabled>
fontconfig version: 2.12.6
freetype version: 2.8.1
harfbuzz version: 1.7.2
lib3mf version: (not enabled)
Application Path: /usr/bin
Documents Path: /home/hans/.local/share
User Documents Path: /home/hans/Documents
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/hans/.local/share/OpenSCAD/libraries
User Config Path: /home/hans/.config/OpenSCAD
Backup Path: /home/hans/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/hans/.local/share/OpenSCAD/libraries
  /home/hans/Desktop

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/hans/.local/share/fonts
  /home/hans/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/hans/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/mathjax
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/abyssinica
  /usr/share/fonts/truetype/ancient-scripts
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/emoji
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/lao
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/nanum
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/openoffice
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/sinhala
  /usr/share/fonts/truetype/takao-gothic
  /usr/share/fonts/truetype/tibetan-machine
  /usr/share/fonts/truetype/tlwg
  /usr/share/fonts/truetype/ttf-bitstream-vera
  /usr/share/fonts/truetype/ttf-khmeros-core
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/truetype/wqy
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/X11/encodings/large

```

thehans, 2019-08-05:
Hmm, I am seeing some other GL programs being unstable, may need to just restart this machine.

thehans, 2019-08-05:
Rebooted and things are back to normal, sorry for the panic.  Probably from a graphics driver update or something like that.  Closing.

closed 2019-08-05 by thehans