openscad/openscad #3034: Adding a circle yield  "Expr: e->incident_sface() != SFace_const_handle() "

Freddan-67, 2019-08-09:
I am a noob on OpenScad. I am using it because someone posted a CC-BY-SA licensed file for a Bahtinov mask generator for my telscope. I am using windows 64-bit version 2019.05.
Anyway the 4 lines in my source code at line 77:
           difference() {                  // makes the horizontal outer margin
                circle(r=outerDiameter/2);
                circle(r=aperture/2-ap_margin);
            }
yield a assertion Expr: e->incident_sface() != SFace_const_handle() in M_const_decorator.h Line: 329
The only meaningful change to the original file I have done is adding line 99-102:
   difference() {
    cylinder(h=14,r=aperture/2-1);
    cylinder(h=14.1,r=aperture/2-ap_margin); // should be 14 but the preview looks ugly then
    }
I made the change to keep the the printed mask stay in place when I use it to colimate my telescope. 
Anyway changing the circle to a cylinder with height=4 removes the bug, so a workaround exists. 
But I do not understand what is going in, and why my change of the source code made it fail, when the original did not fail.
Complete source code:
/*                                                 -*- c++ -*-
 * A Bahtinov mask generator.
 * Units in mm, default values are for
 * a Celestron NexStar GPS 8.
 *
 * Copyright 2013-2014, Brent Burton
 * License: CC-BY-SA
 *
 * Updated: 2014-06-23
 * Update: 2019-08-09 Fredrik Agert add a little extra
 * support inside the tube for skywatcher 200P
 */

// This is the diameter of the mask.
outerDiameter = 235; // [80:400]

// The telescope light's path diameter.
aperture = 215; // [80:400]
ap_margin = 4; // where the vertical circle starts inside the aperture

// Diameter of secondary mirror holder. If no secondary, set to 0.
centerHoleDiameter = 35; // [0:90]

// Width of the gaps and the bars.
// f=1200 mm focal length => 
gap = 6; // [4:10]

/* create a series of bars covering roughly one half of the
 * scope aperture. Located in the +X side.
 */
module bars(gap, width, num=5) {
    num = round(num);
    for (i=[-num:num]) {
        translate([width/2,i*2*gap]) square([width,gap], center=true);
    }
}

// triple support

support_l=20;

for (giro=[0:120:240]){
rotate([0,0,giro])
translate([(aperture+support_l)/2,0,2])
difference(){
cube([support_l,15,4],center=true);
cube([support_l-10,4,4],center=true);
}
}

module bahtinovBars(gap,width) {
    numBars = aperture/2 / gap / 2 -1;
    // +X +Y bars
    intersection() {
        rotate([0,0,30]) bars(gap, width, numBars);
        square([outerDiameter/2, outerDiameter/2], center=false);
    }
    // +X -Y bars
    intersection() {
        rotate([0,0,-30]) bars(gap, width, numBars);
        translate([0,-outerDiameter/2]) square([outerDiameter/2, outerDiameter/2], center=false);
    }
    // -X bars
    rotate([0,0,180]) bars(gap, width, numBars);
}

module bahtinov2D() {
    width = aperture/2-ap_margin;
    difference() {                          // overall plate minus center hole
        union() {
            difference() {                  // trims the mask to aperture size
                circle(r=aperture/2-ap_margin);
                bahtinovBars(gap,width);
            }
            difference() {                  // makes the horizontal outer margin // these 4 lines gives the openscad assetion
                circle(r=outerDiameter/2);
                circle(r=aperture/2-ap_margin);
            }
            // Add horizontal and vertical structural bars:
            square([gap,2*(aperture/2-ap_margin)], center=true);
            translate([aperture/4,0]) square([aperture/2-ap_margin,gap], center=true);
            // Add center hole margin if needed:
            if (centerHoleDiameter > 0 && centerHoleDiameter < outerDiameter) {
                circle(r=(centerHoleDiameter+2*gap)/2); 
            }
        }
        // subtract center hole if needed
        if (centerHoleDiameter > 0 && centerHoleDiameter < outerDiameter) {
            circle(r=centerHoleDiameter/2+1);
        }
    }
}

    
union() {
    linear_extrude(height=4) 
    bahtinov2D();
    // add a cylinder that fits inside the tube; my change to the original source
    difference() {
    cylinder(h=14,r=aperture/2-1);
    cylinder(h=14.1,r=aperture/2-ap_margin); // should be 14 but the preview looks ugly
    }
    // add a little handle
    //translate([outerDiameter/2+gap*1.6,0,0]) 
    //   cylinder(r=gap/2, h=14);
    
    //translate([-outerDiameter*0.273,110,0]) 
    //   cylinder(r=gap/2, h=14);

    //translate([-outerDiameter*0.273,-110,0]) 
    //   cylinder(r=gap/2, h=14);
}

nophead, 2019-08-09:
Change line 72 to
```
                circle(r=aperture/2-ap_margin + 0.1);
```
The problem is caused by cropping the bars to a circle with exactly the same radius as the ring it is subsequently unioned with. When you intersect the bars with a diagonal line it isn't possible to have the intersection points exactly on that line without infinite precision numbers. In practice they will be rounded slightly leaving minute gaps. These get snapped to a grid which merges them but breaks the topology and then CGAL barfs.

By overlapping slightly it ensures there are no gaps between the ring and the bars and the problem goes away.

kintel, 2024-03-31:
Status as of 2024-03-24: Still broken with CGAL, but works with Manifold