openscad/openscad #3035: warning on undef custom special variable in module imported with use

GillesBouissac, 2019-08-10:
I didn't find any issue on this problem anywhere
The behavior pointed in the title is different with predefined and user special variable this is why i report this as a bug.

# Context
- Windows 10
- OpenSCAD version 2019.05 (64 bits)

# Demonstration
## Files
- lib.scad
- main.scad

Content of lilb.scad:
```
module echo_fn() {
	echo ( "$fn=", $fn );
}
module echo_my_var() {
	echo ( "echo_my_var $my_var=", $my_var );
}
```
Content of main.scad:
```
use <lib.scad>
echo_fn();
echo_my_var();
```

## Result

Each time you preview main.scad or render it you get this in the console:
```
ECHO: "$fn=", 0
WARNING: Ignoring unknown variable '$my_var', in file lib.scad, line 5.
ECHO: "echo_my_var $my_var=", undef
```
As you can see you don't get warning on $fn but you get one on a user special variable.
Even if you try to test it before ( `if($my_var!=undef)` ) you get this warning.

nophead, 2019-08-10:
$fn is a built in variable that defaults to 0 that is why you don't get a warning if you don't define it.

t-paul, 2019-08-10:
As @nophead said already, it is defined, so no warning. The [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#$fa,_$fs_and_$fn) also states
> The default value is 0.

closed 2019-08-10 by t-paul

GillesBouissac, 2019-08-10:
Ty for the fast answer.

Ok i agree with you maybe looks like i simplified too much my usecase sorry :)

Actually i need to use a user special variable in a module imported with "use" keyword.
I didn't manage to set a defaut value in this module because when i do this the caller can't override this default value.

```
// File lib.scad
module echo_my_var() {
	echo ( "echo_my_var $my_var=", $my_var );
}
```
```
// File main.scad
use <lib.scad>
echo_my_var();
```
If you do this you get the warning. To get rid of it you must set a value in main.scad...
... or you can try to set a default value in lib.scad:
```
// File lib.scad
$my_var=1;
module echo_my_var() {
	echo ( "echo_my_var $my_var=", $my_var );
}
```
Then you don't have the warning but you are unable to change the value in the main file
```
// File main.scad
use <lib.scad>
$my_var=2;
echo_my_var();
```
Output:
```
ECHO: "echo_my_var $my_var=", 1
```
With $fn or any other bultin special variable i'm able to use the variable without any warning:
```
// File lib.scad
module echo_my_var() {
	echo ( "echo_my_var $fn=", $fn );
}
```
```
// File main.scad
use <lib.scad>
$fn=2;
echo_my_var();
```
Output with no warning:
```
ECHO: "echo_my_var $fn=", 2
```

nophead, 2019-08-10:
I don't understand why you think you should be able to use a variable before you have defined it. You can use the built in special variables anywhere because they are defined before any user code runs.

You can also guard against using a variable before its is defined with ```is_undef()```.

And you can set a special variable when you call a module e.g.

```
let($my_var = 2) echo_my_var();
```

Have you tried setting $my_var before you use lib.scad? I can't remember if that works or not.

GillesBouissac, 2019-08-10:
I don't want to use an undef variable i just want to check if its defined before i use it.
I tryed if ( $my_var == under) and this don't work, but is_undef() actually works and solve my problem, thank you.