openscad/openscad #3047: GSoC 2019 Project - Improvement of OpenSCAD Text-Editor Features

RomitKumar, 2019-08-21:
# About the Project
OpenSCAD is a powerful open-source software for 3D CAD modelling. While the integrated editor provides various text-editing tools, it lacks many useful features. The aim of this project was to add more IDE-like functionality to the editor, thus, improving the workflow and user-experience. 

The project proposed following deliverables:
1. Multi-file editing support
2. Autocomplete suggestion of OpenSCAD keywords

# Work Completed
## Multi-file Editing
The editor previously supported one file per window. This sub-project enabled working on multiple documents under different tabs in a single window. The console, viewport and Parameter widget responds to the active tab. Opening of file in new Window has been shifted under the option of New Window in File Menu. The support of deprecated Legacy Editor has been removed.

Related PRs:
1. #2949 [closed] - A prototype of multi-file editing in tabbed environment
2. #2955 [merged] - Actual code implementing the feature

## Autocomplete Suggestions
This sub-project added the feature of autocomplete suggestion list of OpenSCAD keywords as it is being typed in the editor. A calltip of function signatures is displayed after the keyword is typed and is triggered by `(`. Settings to enable/disable autocomplete and character threshold in Preferences has been added. User-defined and pre-defined template feature is also added which can be triggered by `Ctrl+Insert` shortcut.

Related PR:
1. #2975 [merged] - Code implementing above features

# Other PRs made during GSoC period
1. #2989, #2993 [merged] - Fix for issue #2988
2. #3028 [merged] - Fix to correct reading of Preference settings
3. #3029 [merged] - Fix to display correct filename in export dialog box
4. #3022 - Added 'never' option to reload confirmation dialog box
5. #3045 - Prototype code for hyperlink feature of use'd files

# Future Work
1. Autocomplete suggestion of user-defined module/variable name in current scope
2. Completion of hyperlink feature for use'd and include'd files
3. Implementing correct behavior of Parameter Widget in multi-tab environment

# Related Information
Mentors: Ryan Colyer, Torsten Paul 
[Development Log](https://brlcad.org/wiki/User:RomitKumar/GSoC19/logs)

Look and feel of new editor
[image]

thehans, 2019-08-30:
Small multi-file issue/request: With multi-file editing implemented now, I would expect the Ctrl-W shortcut to close just the current file/tab.  Similar to how it functions in most browsers etc.  

Currently it closes the entire OpenSCAD window regardless of how many editor tabs are open.

t-paul, 2019-08-30:
Please add comments as separate issue, this one is meant as final project summary and will be closed and locked soon.

t-paul, 2019-08-31:
Thank you for an interesting time and a successful conclusion of the GSoC project. With GSoC now finished, I'm closing this issue which is meant a documentation of the final state.

closed 2019-08-31 by t-paul