openscad/openscad #3052: ALSA related segmentation fault

atoms118, 2019-08-29:
Version 2019.05 on Gentoo Linux, without PulseAudio (*only* ALSA).

When rendering, immediately after the render completes the program crashes with a segfault and the message `QAudioOutput: alsa only supports single period!`. The segfault is triggered in libasound according to GDB (I can't give much more info as I didn't build with debug flags).

t-paul, 2019-08-29:
Looks like this might a Qt bug: https://bugreports.qt.io/browse/QTBUG-43256.

thehans, 2019-08-29:
As a quick workaround, you should be able to disable the "render complete" audio cue to avoid another crash.

Go to:  Edit -> Preferences... -> Advanced -> User Interface -> Un-check "Play sound notification on render complete"