openscad/openscad #3054: Editor cursor is white on white background

aricblunk, 2019-08-31:
Hi OpenSCAD team, thanks for the nice program. On Windows 10, when I mouse over the editor, the cursor turns white, over the white background of the editor, making it nearly impossible to see. Much appreciated if you could look into this, thanks again.

t-paul, 2019-08-31:
Please add some more info including the version you are using and specifics about the environment (e.g. using some kind of dark style?).
See also this: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?

aricblunk, 2019-08-31:
Observed the issue using the 2019.05 release and 2019.08.04 snapshot, windows x64 zip. Issue occurs with default windows cursor theme. I can switch to the windows dark cursor theme and the issue is fixed. Normally, with the default cursor theme, the insertion cursor is a black vertical line, not white. Let me know if you need anything else.

MichaelAtOz, 2019-08-31:
That link above says
```
The bug report should give as much information as possible to help with reproducing it, including but not limited to

    ...
    In case of graphics issues, the OpenGL driver information
    ...

Most of the technical version information can be found in menu Help -> Library Info, or via the command line openscad --info. 
```

This is not happening for me with W10 (in a VM).

However please clarify;
> when I mouse over the editor, the cursor turns white, over the white background of the editor

> with the default cursor theme, the insertion cursor is a black vertical line, not white.

Is this both, mousing over (but not editor selected, so no insertion point) AND also when you click in the editor? Neither happening for me with `Windows Default (system scheme)`

aricblunk, 2019-09-01:
Pre-script: I just noticed this is even happening in Google Chrome in this very input box, so this issue can be closed, it's apparently an issue with my system. Sorry about that. If you're curious, here's the additional info I was going to provide to help troubleshoot.

The flashing vertical line where typed characters will appear is fine, it's just the mouse cursor, regardless of whether or not I've clicked. It does seem to work fine on my laptop. It even looks fine in a screenshot:

[image]

But here it is in a photo:

[image]

I'm also using the Windows default system scheme. I was using 125% DPI scaling since I'm on a 4k monitor, but I tried reverting to default 100% and that did nothing.

Using fully updated Windows 10 x64, openscad 2019.05 release

closed 2019-09-01 by aricblunk