openscad/openscad #3057: Use after free

mgehre, 2019-09-01:
Detected by clang trunk:
```
src/Network.h:48:10: warning: returning address of local temporary object [-Wreturn-stack-address]
                return errorMessage.toStdString().c_str();
```
because `toStdString()` yields a temporary `std::string`, and thus the return value of `c_str()` will dangle after the end of this line.

See 
https://github.com/openscad/openscad/blob/c060053d3b4818bf6d0620b0711be845795c4157/src/Network.h#L48

t-paul, 2019-09-03:
I don't have a clang setup currently but I think the change in #3058 fixes the issue. Please reopen if you see some issues flagged still.

closed 2019-09-03 by t-paul

t-paul, 2019-10-18:
@mgehre I just watched your [CppCon 2019 talk](https://www.youtube.com/watch?v=d67kfSnhbpA), awesome stuff. Thank you.

mgehre, 2019-10-19:
Thanks :-) Feel free to play around with https://github.com/mgehre/llvm-project if you want to follow the progress.

commit 8a945af2 (2019-09-03): Merge pull request #3058 from openscad/fix-use-of-temporary

Use std::string as storage (fixes #3057).
commit f44e7439 (2019-09-03): Use std::string as storage (fixes #3057).
commit 05e286d2 (2019-10-22): Use std::string as storage (fixes #3057).