openscad/openscad #3065: Towards STEP importing

agordon, 2019-09-06:
Hello,

I've been experimenting with code to read STEP files, and have some working prototype (without openSCAD). It uses "openCASCADE" to do the actual reading (just like almost any other STEP-capable free software out there), and generates either STL or SCAD scripts.

Currently, the output is very similar to STL (i.e., the object is first triangulated, then OpenSCAD draws the triangles using `polyhedron`) - but there's some code to explore the class heirarchy  of OpenCASCADE too - if someone wants to give it a try.

A fun side-effect of reading the STEPs is being able to seperate the planes, and give each one a different color:

[image]

Code and examples here https://github.com/agordon/openscad-step-reader .

Comments and feedback are very welcomed.

gordon

t-paul, 2019-09-10:
Very cool. OpenCASCADE is a bit scary though as we already have quite some trouble getting the CI builds into the allowed time frames. I was looking at stepCODE for exporting STEP but did not have enough time to really dig into it. Having OpenCASCADE available would be nice also from perspective of maybe allowing it as alternative CSG engine.

I tried compiling the reader but got some linker error hinting at some c++abi incompatibility, not yet sure where that's coming from. I'm on Debian/testing using gcc9 which should not be much different yet from Buster mentioned in the build notes.

agordon, 2019-09-10:
I agree OpenCASCADE is a PITA.
That's why I was thinking of keeping it as a separate executable, which can be optionally built if the needed libraries are there, or skipped other wise.

Later, we can add OpenSCAD menu option to import/convert by executing the binariy. If it's missing - not a big problem - just show error and explain.

kintel, 2019-09-11:
I like the idea of an executable converting files to .scad (or .csg) format, and possibly some UI feature to insert (or reference using import/include) in the editor.

If we generalize this, we could do the same for other 2D and 3D file formats, which could be useful as well.

That said, calling out into a separate process is a new feature that may create new challenges, but the only way to find out is to try I guess ;)

jack9603301, 2021-09-24:
That's a good idea. When can we support it?

t-paul, 2021-09-24:
Yep, the idea to start decoupled from the main application is great as the topic is complicated enough on it's own. But unfortunately I'm not aware of anyone working on this.

tpimh, 2024-05-20:
There's a new library for working with STEP: [ruststep](https://github.com/ricosjp/ruststep) (yes, it's being written in Rust)

cadprobs-a11y, 2026-10-26:
Until native STEP lands, the pragmatic workflow for most makers is still: open/measure the STEP in FreeCAD or CAD Assistant (or a browser STEP viewer), export STL/3MF for OpenSCAD `import()`, and keep a known dimension as a scale check.

Disclosure: I'm Ethan Tan, CADProps developer. Free [STEP viewer](https://www.cadprops.com/tools/step-viewer/) (92 extensions). Native CATIA/NX/Inventor/SolidWorks files are processed on CADProps servers.