openscad/openscad #3067: if statements do not seem to work as expected

billy1380, 2019-09-09:
I am getting some odd behaviour that I cannot explain:
This code sample works well... and you can see 14 printed in the console
```
a = 12;
b = 12;

c = (b==12 ? 14 : 0);

echo (c);
```
but when I change c to a, things start to become unexpected
```

nophead, 2019-09-09:
If you assign ```a``` twice it moves the second definition to the first one and uses that value throughout. I.e. a variable can only have one value. In your case it moves the statement that uses ```b``` before the definition of ```b```, hence the confusing error message.

Simply remove the first assignment to ```a``` and it will work.

billy1380, 2019-09-09:
is there anyway to treat `a` as variable or define something with the intent of changing it?

nophead, 2019-09-09:
No OpenSCAD variables are actually more like constants in other languages. Rather than thinking of it as a executing program it is more like a static description of geometry. In a description of a solid nothing needs to change during that description. If you want another value you have to give it a new name.

The only exception is ```$t``` changes to allow animations but each frame is a separate execution of the script, so it only has one value everywhere for each frame.

You can override a default value. For example a library could define a variable ```a = 3``` and after including it you can do ```a = 2``` but a will be 2 everywhere.

billy1380, 2019-09-09:
Thanks again @nophead, that makes sense.

closed 2019-09-09 by billy1380