openscad/openscad #3070: GitHub Wiki - where are images?

MichaelAtOz, 2019-09-11:
I'd like to add images to a git wiki doc.
If I click the image icon in the editor it asks for a URL ala-markdown, I'd prefer not to externally host them somewhere.

I see this doc has an image:
https://github.com/openscad/openscad/wiki/Editor-Improvements
If I edit it it has the image defined as:
` [[images/notepad-plus-find-replace.jpg]] `
I looked in code/images, it is not there.
Looking at the html shows:
`<img src="/openscad/openscad/wiki/images/notepad-plus-find-replace.jpg">`
Loading that shows it is at:
`https://raw.githubusercontent.com/wiki/openscad/openscad/images/notepad-plus-find-replace.jpg`
So where/how can I upload like that?

t-paul, 2019-09-11:
The only way I know is to clone the wiki as git repo and add it through direct git commit. https://help.github.com/en/articles/adding-or-editing-wiki-pages#cloning-wikis-to-your-computer

MichaelAtOz, 2019-09-11:
I don't have git locally. Been on to do list for a while.

Testing something I saw on stackexchange, using images from issues...

[image]

MichaelAtOz, 2019-09-11:
No good.
Closing.

closed 2019-09-11 by MichaelAtOz

t-paul, 2019-09-11:
You can send me the images via mail and I'll add them.