openscad/openscad #3072: strange one - triangles param in polyhedron fails

MichaelAtOz, 2019-09-12:
This is weird, on my usual Windows 7 (VM) PC with 2019.05 & 2019.08.05 (latest snapshot) using polyhedron with the deprecated 'triangles=' fails:
```
WARNING: variable triangles not specified as parameter, in file tests, line 1
TRACE: called by 'polyhedron', in file tests, line 1.
```
No geometry is shown in preview.
[image]

However nightly 2019.08.20 on Debian works:
```
WARNING: variable triangles not specified as parameter, in file mebd, line 1
DEPRECATED: polyhedron(triangles=[]) will be removed in future releases. Use polyhedron(faces=[]) instead.
```

I think the Warning is a flaw in that recent change on parameter checks.
 
I then tried 2019.08.05 on my other (non VM) Windows 7 box. It works, pyramid is shown.
```
WARNING: variable triangles not specified as parameter, in file openscad-2019.08.05.ci3226, line 1
DEPRECATED: polyhedron(triangles=[]) will be removed in future releases. Use polyhedron(faces=[]) instead.
```

So how would the same executable on two Windows PCs (one a VM), with identical scad code, produce two different results?

`TRACE: called by 'polyhedron', in file tests, line 1.`
v's
`DEPRECATED: polyhedron(triangles=[]) will be removed in future releases. Use polyhedron(faces=[]) instead.'

kintel, 2019-09-12:
> On Sep 11, 2019, at 22:03, Michael <notifications@github.com> wrote:
> 
> This is weird, on my usual Windows 7 (VM) PC with 2019.05 & 2019.08.05 (latest snapshot) using polyhedron with the deprecated 'triangles=' fails:
> 
> WARNING: variable triangles not specified as parameter, in file tests, line 1
> TRACE: called by 'polyhedron', in file tests, line 1.
> 

Perhaps "Preferences->Advanced->Stop on the first warning" is set on one of the machines?

 -Marius

MichaelAtOz, 2019-09-12:
I compared the help/library-info.
I doubt the OpenGL version explains this.
(VM with fault on left)
[image]

MichaelAtOz, 2019-09-12:
> 
> On Sep 11, 2019, at 22:03, Michael ***@***.***> wrote: This is weird, on my usual Windows 7 (VM) PC with 2019.05 & 2019.08.05 (latest snapshot) using polyhedron with the deprecated 'triangles=' fails: WARNING: variable triangles not specified as parameter, in file tests, line 1 TRACE: called by 'polyhedron', in file tests, line 1.
> Perhaps "Preferences->Advanced->Stop on the first warning" is set on one of the machines? -Marius

Yep. Thanks, Closing.

MichaelAtOz, 2019-09-12:
Re-opening.
Parameter warning incorrectly warns of `triangles=` parameter in `polyhedron`
I'll edit the title.

kintel, 2019-09-12:
Not sure I follow. `triangles=` is deprecated and attempts to use it yields a warning. This is  correct behavior.
..unless I misunderstood your updated report.

MichaelAtOz, 2019-09-12:
```
WARNING: variable triangles not specified as parameter, in file pyri-poly.scad, line 1
DEPRECATED: polyhedron(triangles=[]) will be removed in future releases. Use polyhedron(faces=[]) instead.
```
The `DEPRECATED:` is correct, the `WARNING:` is incorrect as `triangles=` is valid until it is removed.

commit 497baa7a (2019-09-15): Merge pull request #3076 from openscad/fix-polyhedron-warning

Accept triangles as valid parameter (deprecated warning remains) (fixes #3072).
commit 4fa150e7 (2019-09-15): Accept triangles as valid parameter (deprecated warning remains) (fixes #3072).
commit 10c98719 (2019-09-22): Accept triangles as valid parameter (deprecated warning remains) (fixes #3072).
commit a8ddbe5b (2019-10-14): Accept triangles as valid parameter (deprecated warning remains) (fixes #3072).
closed 2019-09-15 by kintel