openscad/openscad #3078: Saving file on network drive writes NUL to file in Windows

SevenD7D, 2019-09-16:
Looking at the other "saving files" bugs, I might not have enough knowledge of file streams and Qt to make a good bug report. It is very likely that the issue lies with Qt. Please ask for clarification or more information. I can run anything from "Process Explorer" to "Visual Studio" if it helps.

**Background:**
I had not used my openSCAD for maybe 1-2 years and never had this issue before. I was on openSCAD 2015 when I first noticed this issue, but have since upgraded to 2019.05 and the issue is still the same.

My openSCAD files are saved on a mapped network drive.

**Issue:**
When re-saving any openSCAD file all text is replaced by "NUL", deleting all content. The editor keeps all text and the content is lost once openSCAD has been closed.

I have used Notepad++ and Notepad to look at the files. Notepad++ shows "NUL" and Notepad just shows empty space.

This issue does not appear when saving the file to a new file on the network drive.
The issue can also be fixed by opening the file in wxHexEditor, which shows the correct content. After opening the file (not saving) in wxHexEditor the file content is correct in Notepad/Notepad++.

This issue also does not appear when saving to a local drive.

MichaelAtOz, 2019-09-16:
I could not reproduce this. Windows 7 host, and both Netgear ReadyNAS mapped drive, and another Windows 7 box mapped drive.

> The issue can also be fixed by opening the file in wxHexEditor, which shows the correct content. After opening the file (not saving) in wxHexEditor the file content is correct in Notepad/Notepad++.

That sounds sus. Not saving, means the actual file is presumably there.
What does Notepad++ show under Encoding, both before/after using wxHexEditor.
Also suggest virus/malware scans and [sfc](https://answers.microsoft.com/en-us/windows/forum/windows_10-update/system-file-check-sfc-scan-and-repair-system-files/bc609315-da1f-4775-812c-695b60477a93).

SevenD7D, 2019-09-20:
I have tested it on another machine on the same network with similar settings, but without "Offline Files" enabled, so it works directly off the network drive.
I cannot reproduce it there. I guess I will check my files in another program before closing openSCAD from now on.
I do not believe this issue is related to openSCAD, but has something to do how the file is saved via Qt or something in Windows?
I will close this report as I don't think I will have the time to get to the bottom of this.

RE: virus/malware scan - in my 10 years of experience the culprit was more often the anti-virus software than a virus

closed 2019-09-20 by SevenD7D

mwhead, 2024-06-30:
"nul" files whilst saving to a NAS drive is not solved. It has just happened to me again. It is very, very occassional and so is extremely difficult if not impossible to replicate. It does not happen with any other program that I use when storing to the NAS drive. It happened to me a while ago and I always create separate copies now but this needs resolving.
Thanks and regards

mwhead, 2024-06-30:
Sorry, should have said using Windows 10 and Openscad version 2021.01. Also happened on Windows 7 with earlier version of Openscad.