openscad/openscad #3079: linear_extrude with substracted object inverts normals

shalnoff, 2019-09-16:
when you apply difference() between extruded object and any other object in scene (probably even not interlaced), the extruded object gets inverted transparency in concave sections. Which is looks like inverted normals issue. 

Probably it emerges on concave objects only. 

#### to reproduce: 

difference(){
	linear_extrude(height=h) 
		polygon([
			[-51.450190,-31.041605],
			[-66.189451,-23.208037],
			[-69.215470,51.405365],
			[-60.185894,-22.546817],
			[10.674778,-25.840511],
			[-1.898290,-31.041995]
		]);

	cube([100, 10, 10]);
}

#### version:  2019.01.20

nophead, 2019-09-16:
You need to set the convexity parameter of linear_extrude. I think 2 is the correct value for that shape.

See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion

t-paul, 2019-09-16:
Closing, as @nophead said, it's caused by convexity not being set. See also related [faq entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?).

closed 2019-09-16 by t-paul