openscad/openscad #3081: text(halign="center", spacing!=1.0) is not centered

davidhbrown, 2019-09-25:
Centered horizontal alignment of text is not correct except when spacing is at the default 1.0:

    text("XXXXX", halign="center"); // correctly centered
    translate([0,10]) text("XXXXX", halign="center",spacing=0.8);  // shifted to right
    translate([0,-10]) text("XXXXX", halign="center", spacing=1.2); // shifted to left

Version 2019.05, Windows.

MichaelAtOz, 2019-09-26:
I know it's annoying, but it is he nature of glyphs.
In LTR text spacing is to the right of a ghyph, and there is also usually a small amount of spacing to the left. See [here](https://www.freetype.org/freetype2/docs/glyphs/glyphs-3.html#section-1) for 'advance'.

This shows what happens:
```
scale(0.75) {
color("blue") translate([0,-12,-2.5]) 
linear_extrude(1)
  text("▄▀▄", halign="center");  
color("red") translate([0,5,-1.5]) 
linear_extrude(1)
  text("▀▄▀", halign="center",spacing=0.8);  // shifted to right
color("green") translate([0,-14,-1.5]) 
linear_extrude(1)
  text("▄▀▄", halign="center", spacing=1.2); // shifted to left
}

```
[image]

I added the yellow boxes and lines manually, so they a a little off.
The yellow shows the bounding box for a glyph, the short orange lines show the origin of the next glyph, the last of which is the end position, the black lines are centered.
The yellow of the green boxes shows the expanded glyph where the extra space is on the right. The long orange lines are centered.

I beleive when the glyph is 'painted' OpenSCAD does not know what actual content is painted inside the bounding box. That would require extra processing to map the actual geometry, not just the typography.

MichaelAtOz, 2019-09-26:
Also note in the above link 'width' is notional, the actual 'ink' of a glyph can be outside of the min-max box, intended to overlap with either the previous or next glyph.

t-paul, 2019-11-11:
As @MichaelAtOz visualized, the `text()` parameters are based on glyph properties, not generated geometry. So it works as intended.

closed 2019-11-11 by t-paul