openscad/openscad #3082: OpenSCAD crash when rendering (F6)

phrxmd, 2019-10-01:
Here's an interesting crash. Pressing F6 on the following little file crashes OpenSCAD v2019.01-RC2 on Kubuntu Linux 19.04 for me:

```
// OpenSCAD script to test various crashes
LENGTH = 50; 
WIDTH = 25;
HEIGHT = 2;
CORNER_H = 1;
CORNER_L = 1;
CORNER_W = 1.5; // if I set this to 1, it works  

// 45" pyramid over a rectangle of length l and width w
module pyramid45(l = LENGTH, w = WIDTH){
    polyhedron(
       points = [ [0,0,0], [l,0,0], 
                  [l,w,0], [0,w,0], // 0,1,2,3: base
                  [w/2, w/2, w/2], 
                  [l-w/2, w/2, w/2] // 4,5: top
                ],
       faces =  [ [0,1,2,3],  // base
                  [0,1,5,4],  // right
                  [1,2,5],    // back
                  [2,3,4,5],  // left
                  [3,0,4] ]); // front                ]
}

difference() {
  intersection() {
    cube(size=[LENGTH, WIDTH, HEIGHT]);
    pyramid45(LENGTH, WIDTH);
  };
  cube(size=[CORNER_L, CORNER_W, CORNER_H]); // this kills it
};
```

nophead, 2019-10-01:
It shouldn't crash and the first time I ran it on Win7 it gave a CGAL exception because the top faces of your pyramid are flipped.
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329
```

It did crash when I was playing about with it to find the problem.

This fixes it.
```
       faces =  [ [0,1,2,3],  // base
                  [4,5,1,0],  // right
                  [5,2,1],    // back
                  [5,4,3,2],  // left
                  [4,0,3] ]); // front                ]
```

t-paul, 2019-10-01:
Duplicate of #2847.

closed 2019-10-01 by t-paul

phrxmd, 2019-10-01:
Thank you @nophead & @t-paul. My bad.

t-paul, 2019-10-01:
No problem. I had to check the actual place where it crashes in a debugger to be sure. I believe it's matching the trace from #2847.