openscad/openscad #3088: Discuss Function literals & Objects etc.

MichaelAtOz, 2019-10-08:
I'm raising this issue so the technical issues don't get bloated with discussion.
Function literals, @doug-moen's [description](https://github.com/doug-moen/openscad2/blob/master/rfc/Functions.md), and initial discussion in @t-paul's implementation issue #3077, and Geometry Ojbects, doug's [description](https://github.com/doug-moen/openscad2/blob/master/rfc/Objects.md) and tp's initial work #3087, are proposed as the next evolution of OpenSCAD.

[posting now so this OP remains as the overview, next post continues discussion]

MichaelAtOz, 2019-10-08:
@nophead said:

>But why? What is wrong with modules?
>And how to I have a script to make a fan, that can also cut it holes and make fan assembly?

Nothing is wrong with modules, but as it stands you can't do object-oriented programming.

Having quickly reread Doug's doco, I need to go over it again, as I was interpreting some things with my own presumptions, so I need to think again about some of the fine print (like when the object is instantiated into a Thingi). I'm also not natively imbued with object-orientation.
But as a generalisation I see the adoption of these two changes as a means of better supporting libraries and class like containerisation. The namespace like 'fields' helps.

I'll re-read it today and comment further.

MichaelAtOz, 2019-10-08:
@doug-moen wrote: 
> I think there is some confusion about OpenSCAD2.
> 
> A "script" is an element of syntax. It's a mixture of statements (that create geometry) and definitions (that define variables, functions and modules). An entire OpenSCAD program is a script. You can also enclose a script (a sequence of statements and definitions) in curly braces.
> 
> An "object" is what is returned by a module. So it is a run-time concept, not a grammatical concept. For example, in
> 
> ```
> module lollipop() {
>   radius   = 10; // candy
>   diameter = 3;  // stick
>   height   = 50; // stick
>   translate([0,0,height]) sphere(r=radius);
>   cylinder(d=diameter,h=height);
> }
> ```
> 
> then the thingy returned by `lollipop()` is an object, and the text between the curly braces is a script.
> 
> Consider that a script contains both statements (that generate geometry) and definitions. Program scripts are sometimes used to generate geometry and specify a 3D model. Other times, program scripts are used as libraries: it is the definitions that are important, and libraries do not usually contain any geometry.
> 
> In OpenSCAD2, objects are generalized. An object is a first class value that is constructed from a script. An object contains geometry (if any was specified), and it contains named fields that are initialized from the definitions, if any definitions were present. You can select fields from an object using dot notation: `obj.field`.
> 
> Just as program scripts are used either for 3D models or libraries, so OpenSCAD2 objects can be used either as shapes or as library namespaces. You can import a program script into memory using script("filename.scad"), and that will return an object. If the imported script is a model, then you can use the object as a shape. If the imported script is a library script, then you can use the object as a library namespace. So now we have the concept of namespaces.
> 
> In OpenSCAD2, the object returned by `lollipop()` has named fields like `radius` and `diameter`. So now models can carry around the parameters used to generate them, and those parameters can be programmatically accessed when the object is passed as an argument to another module.
> 
> There's more about library objects here: https://github.com/doug-moen/openscad2/blob/master/rfc/Library_Scripts.md
> 
> When @kintel reviewed the OpenSCAD2 design in 2015?, he suggested that the `script()` module should be called `import()` instead. Perhaps that would eliminate some confusion.

nophead, 2019-10-08:
So instead of
include <screws.scad>
screw(M3_cap_screw, 10);

I could do?
screws = script(screws.scad);
screws.screw(screws.M3_cap_screw, 10);

Can I also do?
screws = script(screws.scad);
screw = screws.screw;
the_screw = screws.M3_cap_screw;
screw(the_screw, 10);

t-paul, 2019-10-08:
I think the answer with the design so far is: it depends... on how things are defined inside the `screws.scad`. Right now I don't yet see how to handle lets call them "old style module definitions" imported from a script.

**ext.scad**
```
// data
a = 42;
b = a + 8;

// function literal
func = function(x) x * x;

// object literal
o = {
    r = 2;
    h = 10;
    data = [ 1, 2, 3, 4];
    nested_func = function(x) x + r;
    cylinder(r = r, h = h);
};

// old style function
function func_old(x) = x + x;

module mod_old(x) {
    cube([x, x, a]);
}
```

```
ext = script("ext.scad"); // or maybe ext = import("ext.scad"); ?
echo(ext.a); // 42
echo(ext.func(4)); // 16

ext.o(); // generate cube with r = 2 and h = 10;
ext.o(r = 4); // customization description sounds like the radius of the cylinder is still 2?
```
I don't see how we would access `func_old` and `mod_old` unless they are handled like with `use<>` and are imported into the scope of the calling module.

nophead, 2019-10-08:
I thought one of the mains points was to import into a namespace, so I don't think it is correct to put modules into the scope of the calling module. What is wrong with the dot notation to access everything inside the script?

t-paul, 2019-10-08:
Nothing wrong with that, I like it very much as it automatically produces namespaces when importing multiple libraries. But we need to consider backward compatibility and this makes things a bit more complicated for existing constructs. If we treat those objects imported via `script("...");` similar to normal modules, they maybe can have their own legacy functions and modules.

That's the whole point of pushing the current prototype code into public to have that discussion and find a solution that both brings OpenSCAD into a much better future state while still maintaining compatibility with existing code.

nophead, 2019-10-08:
Are you expecting to phase out the use of old style functions and modules?

I would only use a new style function if I wanted the pass it, which is not often.

t-paul, 2019-10-08:
I would expect both old and new style to co-exist for quite some time, maybe even forever. We still need to define a way how to handle things if both exist and declare what the preferred usage is.

Right now, with the existing logic

**inc.scad**
```
function f(x) = x * x;
module m(r) sphere(r);
```
**main.scad**
```
use <inc.scad>
echo(f(3));
m(10);
```
Both function and module definitions are imported into the function and module lists of main.scad (which is internally called a file module defined in `FileModule.{h,cc}`). As result both can just be called in that file at top level without any other qualifier.

Whatever the name of the new style import will be (`script`, `import`, ...) we need some definition of how to handle those old style functions/modules at that point. As it's a new feature, it can't break backward compatibility so we are relatively free with the definition. There's at least 3 possible options:
1) both are not accessible at all
2) same behavior like `use<>`  so they would be imported in the enclosing module
3) call via the object, e.g. `echo(o.nested_func(4))` in the [example above](https://github.com/openscad/openscad/issues/3088#issuecomment-539461372) - which needs the shadowing defined if there's another o.nested_func variable is declared

doug-moen, 2019-10-08:
@nophead wrote:
> So instead of
> ```
> include <screws.scad>
> screw(M3_cap_screw, 10);
> ```
> I could do?
> ```
> screws = script(screws.scad);
> screws.screw(screws.M3_cap_screw, 10);
> ```
> Can I also do?
> ```
> screws = script(screws.scad);
> screw = screws.screw;
> the_screw = screws.M3_cap_screw;
> screw(the_screw, 10);
> ```
That is correct. Or at least, that is the vision of how OpenSCAD2 is supposed to work. There are still details to work out about how this interacts with the 3 namespaces (variable/function/module), as @t-paul has noted.

nophead, 2019-10-08:
@t-paul I don't see why it needs to emulate ```use```  as in 2 because one can still use ```use``` if one wants that behaviour.

@doug-moen 
Yes I often use ```screw = M3_cap_screw; screw(screw, 10);```

That isn't going to work if ```screws.scad``` is imported with ```script()``` instead of ```use```. 

Also I was assuming that one can assign a module to a variable as well as an object literal, which seems more like a module literal to me.

doug-moen, 2019-10-08:
@nophead said "Are you expecting to phase out the use of old style functions and modules?"

That is what the OpenSCAD2 proposal says should happen. Chris's objection to this (circa about 2015) is part of the reason why I dropped the proposal.

Today, I think that the 3 namespaces (variable/function/module) should be preserved. We should see if it's feasible to modify the OpenSCAD2 object proposal to be compatible with the 3 namespaces, without trying to deprecate them. The code fragments that Chris is posting should be taken as requirements that the new design needs to satisfy.

nophead, 2019-10-08:
Object literals don't seem to be a replacement for modules unless I am missing something. When you define a module you specify the parameters. Anything inside the module is private. E.g. variables derived from the parameters and functions and modules defined inside it.

With an object literal it seems you can access anything inside it with dot and there doesn't seem to be anyway to distinguish parameters that can be overridden from derived values. Also no way to have a parameter that doesn't have a default.

t-paul, 2019-10-08:
That came up in the IRC discussion too:

> Although I still feel like there is a redundancy between modules and these objects. They are almost identical. Let me ask this question: Is there any feature of objects that is harmful to introduce to modules?
`o = (x, y) { square(x, y); } // Possible syntax for parameters`
Okay, but then it's a class, not an object.
So maybe:
`C = module(x, y) { square(x, y); } o = C(5, 6);`
Then all existing modules are basically already named classes, while assignable modules that are assigned to a variable that behaves like a class. If I have a named module m, o = m(5, 6); would be functionally equivalent to o = { m(5, 6); }; in terms of what is displayed. So then a variable assigned to an unnamed module is behaving identically to a named module.

nophead, 2019-10-08:
If you have:
```C = module(x) { r = x / 2; sphere(r) } o = C(5);``` 

As I understand it you can do ```ECHO(o.x)``` and get 5 but can you also do ``` ECHO(o.r) ```?

t-paul, 2019-10-08:
I believe the idea with that was to have `C` as class definition and `o` as instance - which matches existing module behavior. That seems to collide with the currently described behavior of objects, at least I don't see that differentiation yet. So that needs to be sorted, specifically we need to specify what is evaluated at which point in time.

doug-moen, 2019-10-08:
@nophead: "Object literals don't seem to be a replacement for modules unless I am missing something."
IRC person: "Although I still feel like there is a redundancy between modules and these objects. They are almost identical."

Objects already exist in OpenSCAD. `cube(10)` is an object. `{cube(10);sphere(12);}` is an object literal (a piece of syntax) that evaluates to an object. If I seem to be inventing new terminology, I'm sorry, but there isn't consistent terminology for these concepts. In the C++ source code, an object is represented by class AbstractNode, and the result of evaluating an object literal is represented by class GroupNode. The word 'object' is commonly used in the documentation and forum, so I picked that word.

Modules and objects are orthogonal. The 'children' argument of a module is an object. A module instantiation returns an object. Object literals are also used to form the 'children' argument to a module. In this statement:
```
  translate([1,1,0]) { cube(10); sphere(12); }
```
the expression in curly braces is an object literal.

In OpenSCAD2, the redundancy is between modules and functions. In OpenSCAD2, objects are values, so it is now possible for a function to take an object as an argument, and return an object as a result. So functions can now do everything that modules do. One of the benefits is that it is now possible to write a function that takes an object as an argument, and returns numeric data as a result. There is currently no syntax for this, and there is a backlog of feature requests that are blocked because we don't have a syntax for this.

nophead, 2019-10-08:
I think the confusing thing is object literals can have values overridden like in the lollipops example. so then they are more like modules that get instantiated. 

I must say I find this very confusing and I would class myself as experienced OpenSCAD user, not sure how easy a beginner would find it, whereas I found it very easy to learn OpenSCAD 1.

t-paul, 2019-10-08:
Yes, that's the point where I also have not fully wrapped my head around. For consistency object literals must have lexical scoping too which seems clear enough when just looking at data, but I don't yet see how to handle the geometry instantiation.
```
o = {
    r = 5;
    sphere(r);
};
```
With lexical scoping that binds `r` passed to the sphere directly to the `r = 5`. If the `sphere` is also directly instantiated, how would `o(r = 10);` work?

doug-moen, 2019-10-08:
@nophead: "object literals can have values overridden like in the lollipops example"

You should not try to implement the full OpenSCAD2 proposal. It's too big and complicated. The "object customization" feature that you refer to is not essential (nobody has ever asked for it) so just leave it out.

The aspects of the OpenSCAD2 object type that are valuable, and possibly worth implementing, are the ones that implement features that people have asked for on the forum and in the githhub repo:
* library namespaces
* adding arbitrary named metadata to a shape
* functions that take a shape as an argument and return numeric data as a result
* a dictionary-like data structure, similar to a Python dictionary or a Javascript object or a Ruby hash

doug-moen, 2019-10-09:
@t-paul "For consistency object literals must have lexical scoping too which seems clear enough when just looking at data, but I don't yet see how to handle the geometry instantiation."

I did implement a version of this in Curv. To do it in OpenSCAD, you need to preserve the parent lexical scope of the object literal in a data structure (using whatever data structure is used for function closures). And you need to preserve the unexecuted code of the object literal (probably as a parse tree, or using whatever data structure is used by function closures). When an object is customized, you re-execute the code to regenerate the geometry, using a lexical environment that overrides local definitions with the key/value bindings passed as arguments.

I'm still experimenting with the idea. However, based on my experience with Curv so far, I don't think this is a critical feature, and I don't think it should be prioritized.

t-paul, 2019-10-09:
The lexical scoping itself is not a big problem anymore as I've implemented that for the function literals already. The function objects can now hold on to their lexical scope even after it's dropped from the dynamic evaluation scope stack. So that is basically all prepared. The part where I'm not clear about yet is how to handle the re-evaluation in a consistent way.

So for example
```
o1 = {
    a = 5;
    b = a + 2;
    c = a + 3;
    cube([a, b, c]);
}
```
would evaluate all variables but the `cube()` is not instantiated yet.
```
o1(); // instantiate cube([5, 7, 8]);
o2 = o1(a = 2); // customize data, but no geometry instantiation
o2(); // instantiate cube([2, ?, ?]);
o2(c = 1);
```
That would make a difference between statement and expression evaluation which is probably not intended.

doug-moen, 2019-10-09:
In the OpenSCAD2 proposal, after executing
```
o1 = {
    a = 5;
    b = a + 2;
    c = a + 3;
    cube([a, b, c]);
}
```
then `o1` is fully instantiated. It contains fields (a,b,c) and it contains geometry (`cube([5,7,8])`). There is no need to use `o1()` to instantiate it, because `o1` is already an object.

So you just write `o1;` as a statement to add the geometry in `o1` to some other object. Eg,
```
   translate([1,1,1]) o1;
```
When you write `o1(a=3)` then you are *cloning* a fully instantiated object to create another fully instantiated object. To do this requires re-executing the original script, except that the value of `a` is overridden to have the value `3`. The definitions of `b` and `c` must be re-executed before the geometry is instantiated. The net result is the same as executing
```
{
    a = 3;
    b = a + 2;
    c = a + 3;
    cube([a, b, c]);
}
```
If this seems confusing, it's because `o1` is simultaneously an object *and* it is also a module.

nophead, 2019-10-09:
Since OpenSCAD currently evaluates all expressions, then creates geometry, I don't see how any of this makes sense.  

What happens with:
```
translate([1,1,1]) o1;
o1.b = 7;
```

Or just:
```
o1(b = 7);
```

```b``` is a derived variable from ```a```, so it makes no sense to me it can be modified other than through changing a. In a conventional module this would be expressed and enforced by the argument list.

doug-moen, 2019-10-09:
@nophead said "b is a derived variable from a, so it makes no sense to me it can be modified other than through changing a."

In the context of OpenSCAD, of course it makes sense. What happens if you put
```
    a = 3;
    b = a + 2;
    c = a + 3;
    cube([a, b, c]);
```
into `o1.scad`, and then you run `openscad -Db=7 o1.scad`? Same thing, right? Or what if you write
```
module obj(n) {
   include <o1.scad>
   b=n;
}
```
and then instantiate `obj(7)`? Same thing, right?

The reason I put this weird feature into OpenSCAD2 is because OpenSCAD already has this capability (see examples above), and some people on the forum were concerned that this capability should be preserved in future versions of the language.

nophead, 2019-10-09:
Yes you can do that with -D and the command line, but I don't do customisation of objects that way.  I only use the command line defs to turn on BOM generation and explode views. I don't know what other people use it for.

However with the customiser, which is designed for customise objects in the GUI you can actually select which variables can be changed and in this example it could be just ```a```.  With the normal way of making customisable objects, i.e. modules you can't mess with their internal state. 

I don't think the language should be extended to make bad practice easier.

doug-moen, 2019-10-09:
Is it crazy to allow customization of a dependent parameter like `b`, that depends on another parameter `a`? When I was writing this design document in 2015, I looked at Mendel90 to see if there was code like that, because I didn't want to leave out a feature that people were using in real projects.

Mendel90 does contain code that customizes dependent parameters. For example, `scad/conf/config.scad` defines this:
```
nut_trap_depth = nut_trap_depth(nut);
```
So `nut_trap_depth` is a dependent parameter that depends on `nut`. Elsewhere, this dependent parameter is customized. In `scad/frame_edge_clamp.scad`, we see:
```
include <conf/config.scad>
nut_trap_depth = screw_head_height(M3_hex_screw);
```
I'm not trying to be a dick here, this is actual research I did in 2015 that convinced me that OpenSCAD2 needed to support customization of dependent parameters.

Also, the purpose of this post is to explain the historical document. In 2019, I think OpenSCAD2 is too complicated, and needs to be simplified. The project died because it was too big and complicated. Anything that gets implemented needs to be simpler, easier to understand, easier to implement.

nophead, 2019-10-10:
```scad/frame_edge_clamp.scad``` is ```use```d so it has its own namespace. ```nut_trap_depth``` is logically a new variable that is not related to the one in ```config.scad```. It does shadow it but that is an accident rather by design.

The one in config is the default screw, which varies between machine size variants. It is used where the size doesn't matter in an attempt to keep the BOM smaller.

The one in frame clamp for fixing a camera is a specific size for the job. So a bad choice of names that clash. Should have been ```default_nut_trap_depth``` in config but I am lazy at naming and typing. Often OpenSCAD would give a weird error as the RHS expression has to be in scope in config.scad, and then I would have fixed the name clash but in this case it went unnoticed.

nophead, 2019-10-10:
Rather than adding object literals, which seem to lurch towards more weirdness I would rather see some more simple additions:

module literals to match function literals that can be assigned to variables. So:
```
mycube = module(a,b,c) cube([a, b, c]);
translate(...) mycube(1,2,3);
```

That allows polymorphism when drawing things like hot ends. They have some common properties, like length, but when it comes to drawing them I have to switch on manufacturer as there is nothing in common when drawing a J-Head versus an E3D.

Dictionaries e.g. ``` dict = { 'a' : 3, 'b' : 1 };  echo(dict['a'], dict.a);```

I use lists to represent objects in an OO way but with these mods they will look a lot more like objects.

t-paul, 2019-10-10:
Agreed, what I'm trying to achieve at this time is pretty much the list @doug-moen mentioned in https://github.com/openscad/openscad/issues/3088#issuecomment-539626365 - it seems the current design of object literals goes in a different direction which I did not see (and maybe still don't).

I do like the part that allows structured data to be represented easily, including functions and the automatic namespace support. We will some way of importing the objects from another file.

Dictionaries are a separate topic that might be relatively trivial to implement once we have agreement on a syntax how to declare and create modified instances.

nophead, 2019-10-10:
I think objects literals have taken the normal dictionary syntax and have a syntax to create modified instances. They seem to do everything but in a very confusing way.

t-paul, 2019-10-10:
Well, normal as in Python maybe :smile:... yes, this might be the most common way, but there's still lots of other existing solutions.

For example **Swift** [uses](https://developer.apple.com/documentation/swift/dictionary)
```
var interestingNumbers = ["primes": [2, 3, 5, 7, 11, 13, 17],
                          "triangular": [1, 3, 6, 10, 15, 21, 28],
                          "hexagonal": [1, 6, 15, 28, 45, 66, 91]]
```
Which is not a bad fit either. It might collide with the range syntax though.

**Scala** uses arrow notation
```
var hashMapName = HashMap("key1"->"value1", "key2"->"value2", "key3"->"value3", ...) 
```

More examples in different languages: https://en.wikipedia.org/wiki/Comparison_of_programming_languages_(associative_array)

nophead, 2019-10-10:
Not a Java script programmer but it seems they use the Python syntax but you can use labels as keys and then use the .notation to access them, so they look more like objects, or perhaps they are, not sure.

For example I have:
```

var frequencies = {
   '5min'  : { major : 300,            minor : 60,            poll : 1},
   'day'   : { major : 3 * 3600,       minor : 3600,          poll : 60},
   'week'  : { major : 24 * 3600,      minor : 6 * 3600,      poll : 600},
   'month' : { major : 7 * 24 * 3600,  minor : 24 * 3600,     poll : 1440},
   'year'  : { major : 31 * 24 * 3600, minor : 7 * 24 * 3600, poll : 17280}
};

```
And can do ```frequencies[frequency].poll```

I managed to write a web page that draws graphs without learning any Java at all. I just used Google.

MichaelAtOz, 2019-10-10:
Ooops, I though I had submitted this yesterday

> The part where I'm not clear about yet is how to handle the re-evaluation in a consistent way.
>
> So for example
>
> ```
> o1 = {
>     a = 5;
>     b = a + 2;
>     c = a + 3;
>     cube([a, b, c]);
> }
> ```
>
> would [...]

This is where I'm having concerns too.
That is why I used the term Thingi, I think you need to have 'object' (as in o1 above) but unlike Doug's view, it is not instantiated into a Thingi on creation, that needs to be specifcally done, not just by defining the object. 
I can't see how you can have a class like object, when every time it pops more Thingi's into the result.
I was viewing the 'object' as virtual, where you can change stuff, copy it etc, until you want to instantiate it into an actual Thingi.
The 'shape's (as Doug uses in the RFC) e.g. `lollipop[1]` _[== cylinder(3,50)]_, is a definition, or virtual Thingi, can be used, until finally being instantiated, perhaps as part of other objects, by some call, such as `o1();` above.

Further - today.

Re the above, the RFC has

> The First Class Values principle requires object literals.
An object literal is a script surrounded by brace brackets.
```
// define the lollipop object
lollipop = {
  radius   = 10; // candy
  diameter = 3;  // stick
  height   = 50; // stick

  translate([0,0,height]) sphere(r=radius);
  cylinder(d=diameter,h=height);
};

// now add some lollipops to our geometry list
translate([-50,0,0]) lollipop;
translate([50,0,0]) lollipop;
```

From that, I take the sphere & cylinder in the 'object' do not 'add...to our geometry' list, ie are not Thingi's, only the lollipop calls later do. This then can be utilised with either `include` or `use`, if that was one file, is the 'object' lollipop accessible with `use` which does not add the two lollipop's to our geometry list?

That conflicts with Doug's [statement above](https://github.com/openscad/openscad/issues/3088#issuecomment-540157915)

> In the OpenSCAD2 proposal, after executing
>
> o1 = {
    a = 5;
    b = a + 2;
    c = a + 3;
    cube([a, b, c]);
}
>
> then o1 is fully instantiated. It contains fields (a,b,c) and it contains geometry (cube([5,7,8])). There is no need to use o1() to instantiate it, because o1 is already an object.

tp said:
> Agreed, what I'm trying to achieve at this time is pretty much the list @doug-moen mentioned 

Does that encompass class like `include` and/or `use`?

nathanielstenzel, 2019-10-12:
Is there a chance of getting something like rotate_extrude but around the edge of a 2D shape? This would be similar to the result of using a hand router with cabinetry bits to make the fancy edges around the edges.

MichaelAtOz, 2019-10-12:
@nathanielstenzel  that is off-topic, please raise a new issue for a feature request.

nophead, 2019-10-12:
Subtract the 2D shape from a large square to get a negative, extrude it a
small amount to make it 3D and then Minkowski that shape with a
rotate_extrude of the router profile. That will give a 3D shape that you
can subtract from a 3D sheet.

t-paul, 2019-10-12:
I was thinking of something which would allow things like `import()` or `text()` to be used in a expression context and returning _something_ which will have both raw data and also geometry to instantiate. That seems to be very similar to what object literals would produce, but not exactly the same? Something like
```
t = text("hallo", 14);
echo(t.width);
echo(t.chars[2].x);
t.text(); // or t.geometry() to actually instantiate the geometry
t(); // not sure if that would make sense on the top level
```

MichaelAtOz, 2019-10-14:
t.\<something to instantiate the geometry>
A quick thesaurus of instantiate and render gives;
manifest, realise (realize - maybe not), embody, create, compose, generate, draw, solidify, publish, show, makeItSo, evoke, depict, make, build, assemble, construct, express, ...
I like t.build() or maybe just reuse a term t.render()?

That, above post, would be handy, presumably restricted (maybe initially) to things were that data is readily available, as opposed to having to render it. I believe the bounding box of `text()` is part of the API, and `import()` has the geometry so BB/max/min should be easy.
These types of data would be read only. 

Don't know whether extending that to `render()` is good, e.g. 
```
// actually build the 3d geometry in the object, but not adding it to the global geometry
ob = render() something(); 
echo(wide=(ob.bb.maxX-ob.bb.minX) );
translate([0,0,-ob.bb.minZ])  // bring object to z-axis
  ob.build();  // add pre-rendered geometry (thingi) to the global model/shape/collections-of-thingi
// maybe allow points=ob.points?
ob2=doSomething(ob.points);
oc2.build();
```

MichaelAtOz, 2019-10-14:
Maybe that last should be
```
newpoints=doSomething(ob.points);
polyhedron(points=newpoints);
```

t-paul, 2019-10-14:
Yes, something like that is certainly on the wishlist (e.g. #1713 and more specifically https://github.com/openscad/openscad/issues/301#issuecomment-141852823). It would be great if we can get that part working, which is pretty much point 3 on the trimmed down list @doug-moen posted above. I don't think the full data access will be the first to be added as that is going to be another one of those API which will be very hard to change once put into the world. Things like bounding box might be easier to define in a way that does not need to be changed later.

One thing to note is that the `t.text()` in the example above is not calling some kind of member function of `t`. It would be: evaluate `t.text` which will result in some kind of value and then do a function call with no parameters on that.

For the function literals that is:
```
o = { a = 4; f = function(x) a + x; };
echo(o.f(10)); // ECHO: 14
func = o.f;
echo(func(12)); // ECHO 16
```
In this case `o.f` evaluates to a function-thingy value (a reference to the function literal including the captured lexical context which defines variable `a`). This value can be passed around just like a string value. It does have the additional feature that it's possible to call it via `(...)`.

What @nophead showed in https://github.com/openscad/openscad/issues/3088#issuecomment-540544200 would make that possible for geometry too. With the old style module definition, it's probably not possible without breaking backward compatibility (just as with functions).
```
o = { mycube = module(a,b,c) cube([a, b, c]); };
o.mycube();
```
This seems to overlap the object literals and I'm not sure how we could resolve the conflict there.

jordanbrown0, 2020-03-30:
[ I don't know exactly how these thoughts fit into this discussion, but I'll dive in... I think, but I'm not sure, that these points have been made. ]

[ Note:  in the following I use "object" in its computer-language sense, not in the sense of a geometric object. ]

Reading over this discussion and the discussion in PR #3087, three things come to mind.

First... Using the object's variables as its external interface seems like a bad idea.  It violates modularity, exposing the object's implementation to callers.  (As a simple example, should `echo()` dump the object's internal implementation variables?)  Better, I think, would be to extend the concept of a module's parameters and their defaults to make those parameters be the external interface, and the defaults their values.

Second... In many conventional languages (e.g. C, Java, JavaScript), uttering a function's name in a standalone statement is equivalent to uttering it in the context of an expression.  It's always an expression, but its return value might be discarded.  In a scheme like those described, the two could be quite different.  Uttering it in an expression might *not* execute it in any sense.  Maybe all it does is to create a new module with different default parameter values.  It is only when it is eventually uttered as a statement that it is actually executed and geometry created.

There should perhaps be a mechanism to execute a module and extract metrics from the execution, but that seems like a second evolution of the concept.

Third... Note that an important use case is for aggregate values analogous to C structures, Python dictionaries, or JavaScript or Java objects (without the methods).  That probably falls out of any scheme as the trivial case where the list of resulting geometric objects is empty, but it seems worthwhile to explicitly mention.

jordanbrown0, 2020-03-30:
In https://github.com/openscad/openscad/issues/3088#issuecomment-539611776 Doug Moen says:
> The 'children' argument of a module is an object.

No, it's not.  An object, in the sense of cube(10), has been turned into geometry.  Children are not turned into geometry until you use them with children().  You can confirm this by having an echo() in your child; it will execute as many times as that child is referenced.  You can also confirm it by having a child that references a $ variable and setting the $ variable to different values when invoking the child.

doug-moen, 2020-03-30:
>> The 'children' argument of a module is an object.

> No, it's not. An object, in the sense of cube(10), has been turned into geometry. Children are not turned into geometry until you use them with children(). You can confirm this by having an echo() [...]

I'm using the word "object" as an abbreviation of "geometric object", consistent with how "object" is used in the documentation. For example, `cube(10)` is an object.

OpenSCAD has 2 phase evaluation. In the first phase, objects are not turned into geometry (a triangle mesh) unless they are forced to using `render()`. This first phase of evaluation is when modules are instantiated, `children()` is evaluated, etc. During this evaluation phase, objects like `cube(10)` do not have geometry. The result of the first phase of evaluation is a CSG tree of geometric objects. The second phase of evaluation converts this CSG tree into geometry.

OpenSCAD 'modules' are a kind of _function_. When you 'instantiate' a module, eg by evaluating `translate([10,10,0]) cube(10)`, you are _calling_ this special kind of _function_. There are two kinds of arguments that can be passed to a module: values, and objects. The object arguments are referred to as 'children'. In all mainstream programming languages, arguments are always passed _by value_, which means they are evaluated before the function is called. But there is an older parameter passing mechanism known as _by name_, in which the argument expression is evaluated each time the parameter is referenced. OpenSCAD passes value arguments by-value, and it passes object arguments by-name. It's still a function call, even if the function call syntax is different from C-like languages, and even if the ancient and unfashionable call-by-name mechanism is used for passing some of the arguments. So in the module instantiation
```
translate([10,10,0]) cube(10)
```
I claim we are calling the `translate` function with two arguments: the value `[10,10,10]` and the object `cube(10)`.

Thinking about modules and children in this way is an important first step to extending the OpenSCAD language to solve the problems that need solving.

nophead, 2020-03-30:
Hmm, not sure that is totally consistent terminology because ```cube()``` is a built in module, like ```translate```. It happens to generate geometry when called (or should that be instantiated) but it could just as well be ```rotate```, that doesn't or a user module that could do or not.

doug-moen, 2020-03-30:
Well, anyway, in my terminology, `cube` is a function that takes only value arguments, and returns an object (a cube), while `translate` is a function that takes both value and object arguments, and returns an object. The point of describing things like this is that I think we ought to extend OpenSCAD so that objects are values: they can be stored in ordinary variables, passed as arguments to ordinary functions, and returned as results by ordinary functions. The module mechanism will not go away, of course, we will just have a new method for working with objects that admits a larger range of possibilities.

jordanbrown0, 2020-03-30:
You're right that it's call-by-name.  It's like you're passing a reference to a function.

It looks, from the outside, like there are at least three phases of execution:  assignment, module execution, and geometry evaluation.

When you have a module invocation
```
module x() {
    children(1);
    children(0);
}

x() {
    y = echo("assignment 1") 1;
    cube(echo("module execution 1") 1);
    cube(echo("module execution 2") 1);
    cube(echo("module execution 3") badvar);
    z = echo("assignment 2") 1;
}
```

The assignments are both executed first, and then the module executions *when and if* the children are referenced.  The sequence above yields:

```
ECHO: "assignment 1"
ECHO: "assignment 2"
ECHO: "module execution 2"
ECHO: "module execution 1"
```
Notes:
- The assignments are executed first, in order.
- Then the children, in the order referenced.
- The third `cube()` invocation is not executed, not even enough to evaluate its arguments and not even enough to detect that one of its arguments is an undefined variable.

Anyhow, my point is that the third `cube()` invocation is a child - in a sense, an argument to `x()` - but is not in any sense a geometric object.  I suppose that you could say that the argument is an expression that evaluates to a possibly-null geometric object.

Tying this back into the original topic, a module/object referenced in an expression context could be in a state similar to a not-yet-referenced child.  (I'm not clear on whether it should capture the values of $ variables at the time it's created, or at the time that it's actually executed.)

thawkins, 2020-08-15:
Slight divergence 

I see mentions of OpenSCAD2 above, is that a project i can go and look at and possibly contribute to. 

I'm interested in

1. Namespaces
2. Package Management 
3. Pluggable Model Renders/Processors for 3D printing, CNC, VR image format generation.

t-paul, 2020-08-15:
@thawkins it's not a separate project or version, it's a path for further improvements to the OpenSCAD language (main repo with the documentation is: [https://github.com/doug-moen/openscad2/](https://github.com/doug-moen/openscad2/)). If you would like to contribute, you are already in the right place here. Plus there's also a channel to chat on [IRC](https://webchat.freenode.net/?channels=openscad).

NateTG, 2021-03-30:
It's already possible to implement class-like behavior with function literals, but the syntax for it isn't great for it.  I'm also not sure how fast it is. 

With a syntax update that makes
```functionname.instance```
equivalent to
```functionname(dot="instance",this=function("self"))```
or something similar

Should allow for a lot of stuff, like:

```
include<rgb.scad>

color=rbg.new(r=128,g=128,b=128);
assert(128==color.r);
```

And:

```
include<class.scad>

myclass=class.new(
   variables=[
      ["name","value"]
  ],
  methods=[
     ["myname",function (this=undef) (this.name)]
  ]
);

assert(myclass.myname()=="value");
```

jordanbrown0, 2021-07-18:
I had some thoughts on how objects might relate to parameter passing, and some object-manipulation-function semantics that we might want.  Torsten thought it worthwhile to record them here.  I have *not* vetted them for whether they duplicate or conflict with ideas already proposed.

 [ This is a bit stream-of-consciousness, not a fleshed-out proposal. ]

So I'm just writing something moderately complex that uses textmetrics, and I need to pass the same text parameters to both textmetrics() and text().  I can obviously do that by passing text, font, size, et cetera, to both, but that's tedious.  Suppose that I had objects and could just pass an object to specify a bunch of parameters in a single bundle, e.g.:
```
    params = { text="Hello", size=10, font="Arial" };
    tm = textmetrics(params);
    text(params);
```
or perhaps there could be a named argument for the parameters, e.g.:
```
    text(params=params);
```
and maybe you could override some of the parameters:
```
    text(params=params, size=12);
```
It's easy to see how such a thing would be implemented internally.  It would be doubly cool if you could make it convenient for user modules to do the same sort of thing, but that might require specialty syntax so that either on the caller side the object was marked as "use the elements of this object as the named parameters for this call" or on the called side you could say "get the named parameters from <this> parameter".

For discussion purposes, let's steal JavaScript "spread" syntax:
```
    text(...params, size=12);
```
That is, if you had a parameter that was "...object", it would mean "get named parameters from this object if they are not otherwise specified in the call".

I don't really have any ideas for a called-side syntax.

Perhaps a similar concept could apply to arrays, so that if you have an array a, you could say "...a" and it would be equivalent to listing each of the elements of the array as an argument, e.g. max(...a) to get the maximum value from an array.  (But I see that max() already does that on its own when passed an array.)

---

So that leads me to think:  if you had an object, and you wanted to create a new object that copied some of the elements but added or overrode some, what might that syntax look like?

Maybe
```
    new = object(old, { val1=123, val2=456 });
```
which is of course a special case of
```
    new = object(old1, old2, ...);
```
or maybe you wouldn't need to wrap the new stuff in an object:
```
    new = object(old, val1=123, val2=456);
```
which would actually suggest a way to create objects without requiring new syntax:
```
    new = object(val1=123, val2=456);
```
The rules would probably be:

-  Start with an empty object.

-    If an argument is a named parameter, add that named parameter as a new element of the object, replacing the existing element if any.
-    If an argument is an unnamed object, add each of the elements of that object as new elements of the object, replacing existing elements if any.
-    (maybe, but probably not) if an argument is an array, add each of the elements using constructed names "0", "1", et cetera.
-    (maybe, but probably not) if an argument is unnamed, add it using constructed names "0", "1", et cetera.

My first thought was that this would be a new meaning for concat(), but today concat(obj1,obj2) would naturally yield [obj1,obj2], so I think a new function might be best.

jordanbrown0, 2021-10-05:
Continuing my comment above, or actually restarting since I had forgotten it, #3345 prompts me to think about sets.  Perhaps a set is just an object with a bunch of names with boolean true values.

If you stick to "true":

Union is the concatenation described above.
Intersection would be something new.  Not clear how the values in the second argument would be used.
Similarly difference.
Adding an item to a set is just a union.
Removing an item from a set is a difference, or maybe is a union with an object where the value of the item is explicitly undef.
Testing an item is is_bool(obj.item) && obj.item.
You'd want a way to query the number of items in an object, to test for emptiness.  Maybe that's len(obj).

I don't know how good a match this would be for proper set semantics, but it's at least sort of plausible.  The biggest problem I can think of is that { foo = false } would have unobvious behavior.  Maybe if you just say { foo, bar } without giving them values you get a set rather than an object.

jordanbrown0, 2022-08-10:
It seems straightforward that object members could be functions, e.g.

obj = { func = function (x) x^2; };
echo(obj.func(2));

However:
* Do we have module-references-in-variables yet?  I don't think so, which means you couldn't have an object with a module reference in a member.
* To be useful in an OO-ish way, it seems like the function would need a way to get access to the object that was used to refer to it, along the lines of the "this" reference in JavaScript.

closed 2026-04-07 by github-actions[bot]