openscad/openscad #3089: Feature Request: Weld function or modifier of union

titchy, 2019-10-08:
It is not simple to produce unions with decent joints. 
To improve this a feature, either new function or modifier for the union function, that takes a radius and width variable to provide a curve between surfaces rather than a hard angle would allow much better end results.
I suggest the term weld as I would expect the end result to look like a weld joint with depending on radius either a straight concave or convex link between the two objects stretching from the joint to a length along the objects defined by the width variable.
Thank you all for a great piece of software.

t-paul, 2019-10-08:
Marked as "Discussion" as there's so simple "just needs to be implemented". This needs research in how to do this with with meshes. I found some papers so far, one even claiming implementation with CGAL, but no details or source code linked.

TLC123, 2019-10-08:
> Marked as "Discussion" as there's so simple "just needs to be implemented". This needs research in how to do this with with meshes. I found some papers so far, one even claiming implementation with CGAL, but no details or source code linked.

Rounded union has been suggested multiple times. issue:884 
Was there perhaps in the paper, a person with a name that could be contacted?

t-paul, 2019-10-08:
One paper was https://ialhashim.github.io/files/documents/rollingball_fillet_report.pdf

Right, I think there's enough overlap with #884 to close this one as duplicate. If someone has additional info, please add to that issue.

closed 2019-10-08 by t-paul