openscad/openscad #3090: Discuss SSAO

TLC123, 2019-10-08:
Prompted by a discussion in the forum about shadows (shading) and shape perception i low light regions.
The issue was that it is some times hard  to distinguish the shape of geometry in the less exposed regions.  

True ambient occlusion is excellent to convey shape information.
Fake screen space ambient occlusion lack some of that excellence but is pkaussible usefull.
I don't have any experience of SSAO in cad environments. 
In its simplest form, implementation is as i understand it a medium trivial image postprocess. 
Give that a depth-map is accessible.

My question to community is what is there any obvious reason to belive this is a hard/impossible feature not worth investigation?

nophead, 2019-10-08:
I don't know that it needs anything more than a non-diffuse light source. The problem is that any flat surface is all the same shade exactly when lit with a diffuse light, so another overlapping face at the same angle can't be distinguished unless it is a different colour. 

If the light were a point source then the angle of incidence would vary across a flat face so the shade would vary.

I don't know if this is possible with OpenCSG as it seems to use trick of two diagonally opposed light to get around a problem with inside faces.

t-paul, 2020-02-16:
Closing as there's no discussion happening. Maybe repost on the forum to see if there's feedback to be collected. We can open a new issue if there's a clearer view of how this feature would fit into OpenSCAD.

closed 2020-02-16 by t-paul