openscad/openscad #3091: hull() render times massively increased by module and other bracketed operations

UserNameUnavalabe, 2019-10-12:
I have noticed that any sort of bracketed operation inside a hull() will increase its preview and render time 16 fold or more.
While it is often possible to work around this querk, if you're aware of it, I suspect many users aren't.
Also, the workarounds often involve duplicating larg chunks of code which rather defeats the object of having modules.

`$fn=80;

hull(){                 //preview = 0 min 3 sec
sphere(10);
translate([30,0,0])
    sphere(10);
}

*hull(){                 //preview = 1 min 15 sec
    translate([-15,0,0]){
        sphere(10);
        translate([30,0,0])
            sphere(10);
        }
}

*hull(){                //preview = 0 min 49 sec
    if(1==1){
        sphere(10);
        translate([30,0,0])
            sphere(10);
        }
}

*hull(){                //preview = 0 min 49 sec
    shape();    
}       

module shape(){
sphere(10);
translate([30,0,0])
sphere(10);
}`

Version 2019.05, Windows 10

nophead, 2019-10-12:
It is imply because any module can only return a single child, so if they need to return more there is an implicit union and that is a much slower operation than hull.

To avoid it place each object inside the hull individually. To avoid length repetition you can make a module to place one object and call it repeated times with a parameter to specify where. See an example here: https://github.com/nophead/NopSCADlib/blob/master/printed/handle.scad#L65

t-paul, 2019-10-12:
This is what #350 tried to improve on but got stuck. Reasons are exactly what @nophead described. Hopefully that PR gets picked up eventually or retried from scratch. Closing as duplicate as this is a subset of #350.

closed 2019-10-12 by t-paul

nophead, 2019-10-12:
Rather than the full lazy union I think a shortcut when the operands bounding boxes don't overlap would solve a lot of these speed issues.