openscad/openscad #3093: Wrong line number of included file in error message

MichaelAtOz, 2019-10-15:
Introduced after 2019.06.21

The error is at line 247, later versions say it is 281.
[image]

t-paul, 2019-10-24:
Is that with warnings as error? And is it actually the same place in the file causing the error? There could be changes to warnings between the versions causing the different behavior.

MichaelAtOz, 2019-10-24:
> Is that with warnings as error?

If you mean stop on first warning, no, and is is an ERROR. See prefs below.
Note the error (fabricated this time), was due to changing assign() to basic assignments, assign() between to other transformations was valid, but assignment is not. As people may be doing this more often when assign(0 gets dropped, I wonder is a clearer error message is possible?

> And is it actually the same place in the file causing the error?

Not sure what you mean, OP was two Openscads running at the same time with the same source files. Same error in the included file, later version reported it as a different line number.

[image]

t-paul, 2019-10-25:
I can't reproduce that so far. Maybe a caching issue when working with multiple tabs?

MichaelAtOz, 2019-10-25:
Here is the include library with the fabricated error at line 253.
[text_on.zip](https://github.com/openscad/openscad/files/3774170/text_on.zip)
With a fresh run of 2019.10.06, and `include <text_on.scad>` the only code at line 1, F5 gives error at line 253.
If I insert blank lines before it the reported line number goes up.
[image]

MichaelAtOz, 2019-10-26:
I can't see if it would matter, but the library is accessed via the first directory here
`OPENSCADPATH: C:\Users\MeB\Documents\3D-REPRAP\OpenSCAD\OpenSCAD Libraries;C:\Users;C:\test;C:\test2;C:\test-uc-?` (hmm, that ? was a unicode char...)

t-paul, 2019-10-26:
Ahh, I see, that error is not using the full location object but the separate `yylineno` which is not rest at `include<>`. Should be easy to fix.

t-paul, 2019-10-26:
#3106

t-paul, 2019-11-11:
Fixed via https://github.com/openscad/openscad/commit/0fd84dbc4634f3be1b904fd31b1b49b05137c923

closed 2019-11-11 by t-paul