openscad/openscad #3098: Allow to set the number of decimals in customizer

stenci, 2019-10-21:
I'm not able to set the number of allowed decimals in the customizer.

For example, in my model the `dia_hole` parameter represents the diameter of a hole. The value is usually an integer between 5 and 11 mm, but sometimes can be 6.5 mm or 11.1125 mm which represents a fairly common drilling bit of 7/16 inches.

If I set the initial value in the code to the most common value `dia_hole = 6`, then I can only type integer values.

Today I need to set the default value to any (usually) meaningless value with 4 digits and ask the user to (almost) always change that value.

It would be nice if I could do something like this:

    dia_hole = 6 // Min:5 Max:20 Decimals:4 SpinnerRounding:0.5

This would allow me to:
- constrain the value between 6 and 20 (included)
- allow the user to type any number with 4 decimals
- while keeping the spinner useful by rounding the next up/down value to the closest multiple of 0.5

t-paul, 2019-10-21:
The current customizer syntax is meant to be compatible with what [Thingiverse supports](https://customizer.makerbot.com/docs). We should have a native way of specifying additional information at some point. In addition the GUI could be extended to be more flexible.

stenci, 2019-10-22:
Here I am again, two days in a row: a cylinder with diameter = 10 mm is usually printed as 10 mm, but with some resins it may be 10.15 mm. So I added a parameter called `th_comp` used to compensate for thickness errors. I would like for the default to be 0, but that would prevent me from typing any decimal value. So I am forced to set the default to 0.01.

Setting the default values to the wrong values only to make the interface happy doesn't feel right.

Are there other workarounds?

t-paul, 2019-10-22:
Maybe
```
th_comp = 0; // [0:0.01:0.5]
cylinder(r = 10, h = 10 + th_comp);
```
For further discussion, please use the [forum/mailing-list](http://forum.openscad.org/).

buteomont, 2020-07-16:
This is a big problem for me as well.  Seems that you should be able to enter any number directly into the customizer's input fields, regardless of the original value.

pedjas, 2020-10-15:
I am supporting this request.

I even tried to set 

`thickness = 9.0;`

But GUI still shows only integer and does not allow user to enter decimal value. User can type in decimal point but no figures after that.

ronsmits, 2020-10-17:
I was just playing with this and setting the thickness to 9.01 makes it possible

1950sRobot, 2020-11-25:
Please fix.  Making stepper motor models, started with nema 17 that has a 5mm shaft.  Now I can't use a 6.35mm shaft for nema 23 through the customizer.  I'll have to multiply the shaft diameter by 100 and divide in the script to get the precision I need to enter a common value.

stephancom, 2020-12-01:
Here's a great example of a case that need supporting, and my suggestion for a syntax that might work. ( relevant to #1766 , #2316 )

I'm trying to modify [Customizable Case (Rectangular)](http://www.thingiverse.com/thing:384919) and of course the default settings for things are causing trouble, like the interior height, since the default in the file is `10`, I can set that to `21` but not `21.5`.  OK, fine, the argument made elsewhere that there needs to be a max and min I suppose makes some sense here, except 0 is too small and I see no reason to dictate a minimum besides that.  We could get by with, perhaps

```c
interiorHeight = 10; // [0.1:0.1:999]
```

but now that creates a slider which isn't really wanted.  A little further down, however, is the poster child for a rangeless spinbox:

```c
// Adjust the tightness of the closed case (larger value makes the case looser, negative value makes it tighter - try adjusting in 0.1mm increments)
lidInsetOffset = 0.0;
```

oh.  Well.  There is no sensible lower or upper limit.  Tsk, tsk.  `-1:0.1:1` would probably be OK, but maybe someone needs -1.1 or 1.1... or maybe -1/+1 are too high/low and we don't want to encourage them.

The sensible option for this case really would be, I think:
```c
lidInsetOffset = 0.0; // [:0.1:]
```

... and the first could be expressed reasonably well with
```c
interiorHeight = 10; // [0.1:0.1:]
```

I haven't looked over the code in depth, but I can't imagine this would be terribly hard to implement.

Now... what about cases where we want to constrain to a range but do NOT want a slider?  How to prevent that?  The argument is made above that we wish to maintain compatibility with the Thingiverse customizer _(which isn't even working right with Safari)_... I sympathize, but they don't even have spinboxes - which do exist in html5 - perhaps it would be OK to break that and push back a little.  They are, after all, beholden to OpenSCAD, not the other way 'round.  

I would propose, for a breaking change, something like 
```c
interiorHeight = 10; // [spinbox:0.1:0.1:999]
```
or even, since their syntax permits `[100]` as being equivalent to `[0:1:100]`, we could allow
```c
interiorHeight = 10; // [spinbox:999]
```
and for completeness' sake permit `[slider:min:step:max]`.  this most accurately maps, I think, their existing scheme and should cause them the least amount of trouble to support.  

For a non-breaking change, we could consider specifying the widget in the explanation line qs, so on Thingiverse it would just appear as part of the explanatory text, though that's a less appealing approach, as here: 
```c
// Adjust the tightness of the closed case (larger value makes the case looser, negative value makes it tighter - try adjusting in 0.1mm increments) [Spinbox]
lidInsetOffset = 0.0; // [-1:0.1:1]
```

... anyway, just my opinions, keep up the great work!

thehans, 2020-12-01:
I would say that the solution should involve checking how many digits are specified after the decimal in the original assignment.  
The current code tries to check this numerically... the problem is that the parser automatically converts number literals into actual numbers, so `x = 1.000` is indistinguishable from `x = 1` by the time the decimal precision is being determined.

That said, each expression including number literals, contain Location info which tells the span of start/end line and column from the source file.  So we might be able to grab the source based on that(and do a sort of second pass parse of just that value to determine digits after decimal), but I'm not sure if there's any existing interface to take ASTNode Location data and extract the original source text verbatim.  
Otherwise the main parser would need to be modified to provide this extra metadata for number assignments, but that feels like the wrong way to handle it, since number of decimals wouldn't be relevant for majority of cases (ie numbers that *aren't* customizer params).

buteomont, 2020-12-01:
Why can't you just type in whatever number you want, and let the spin buttons roll the rightmost digit?

Triangulix, 2021-09-03:
Yes, please fix this problem. It is annoying. Make the field a decimal field if the default value is specified with a decimal point, even if  the value is x.0. This would fix the problem without changing the syntax.

pattarini, 2021-11-07:
my workaround is as follow:
set a silly number, with the right amount of decimals:   a=6.6666
elsewhere, change this value, and if a==6.6666, it takes the default value... 0 or what's needed.

SomeDeveloper13, 2023-06-04:
> Yes, please fix this problem. It is annoying. Make the field a decimal field if the default value is specified with a decimal point, even if the value is x.0. This would fix the problem without changing the syntax.

This makes the most sense. In fact this is what I was expecting to happen when it didn't allow decimals at first.

- 96 -> Integer (zero decimal places.)
- 96.0 -> Decimal (1 decimal place.)
- 96.00 -> Decimal (2 decimal places)
- etc.

Especially since that's exactly what happens when there's a non 0 decimal.
96.01 does doe 2 decimal places after all.
[image]
**becomes**
[image]

**but**
[image]
**becomes**
[image]

SomeDeveloper13, 2023-06-04:
Personally, I'd classify this as a bug, rather than an enhancement.

BlackysBoss, 2023-06-16:
Soooo nothing changed in 2.51 years time?

gsohler, 2023-06-20:
Guys, you can exactly specify the amount of decimals to be displayed in the
customizer

-> Sets 3 decimals for parameter 'var'
var=6; // [4:0.001:10]

-> Sets 1 decimals for parameter 'var'
var=6; // [4:0.1:10]

-> Sets 5 decimals for parameter 'var'
var=6; // [4:0.00001:10]

gabriele-cannizzaro, 2023-11-28:
I had resorted to the slider method, but then I found the section for "spinbox" in the documentation [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer#Spinbox), and I tried with setting the parameter this way:

var = 3.00; // .01

The default value is "3.00" (so 3), but the user can input any number with 2 decimals. This seems to achieve the task of setting an integer default with a defined number of decimal places.

t-makaro, 2023-12-05:
> I had resorted to the slider method, but then I found the section for "spinbox" in the documentation [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer#Spinbox), and I tried with setting the parameter this way:
> 
> var = 3; // .01
> 
> The default value is "3.00" (so 3), but the user can input any number with 2 decimals. This seems to achieve the task of setting an integer default with a defined number of decimal places.

That solved the issue for me. I didn't look too into it, but that does seem like it would cover all the bases here.