openscad/openscad #3100: CGAL error in Polyhedron3...

Fuechsl, 2019-10-22:
Hi developers,

I encounter a strange error/behavior in OpenScad 2019.5 Mac today:
[image]
I reduced my construction as far as possible to reproduce the error. Those are the files needed:
[ErrorOpenScad.zip](https://github.com/openscad/openscad/files/3757590/ErrorOpenScad.zip)
plot_function.scad is a library I found on thingiverse.

If you try to generate CGAL in the ErrorOpenScad you get the error from above.
You can avoid it either by setting the var wellenquali in line 8 to a value bigger than 1 (so 1.01 i.e.) or delete line 19.

I don't know if I'm doing something wrong, the library has a problem or OpenScad itself, but it seems the latter to me.

Regards, Martin

MichaelAtOz, 2019-10-23:
Your problem is the shared face of the two objects.

 

It is non-manifold to share the same face of two objects.

It breaks the "all edges must be shared by exactly two polygons (usually triangles)" rule.

It can't tell weather something is 'inside' or 'outside' the object.

Which is why moving it a little fixes the error.

When dealing with multiple objects, it is best practice to either have them clearly separate or
embedded into a face, not just touching.

 

 

 

  _____  

 

 
<https://user-images.githubusercontent.com/56889899/67315362-b3ffb780-f506-11e9-9ec7-9bf2897d1fc9.p
ng> Bildschirmfoto 2019-10-22 um 19 52 25

<https://github.com/openscad/openscad/issues/3100?email_source=notifications&email_token=AAPBVYMY5R
CA66MXLYQEI33QP46RZA5CNFSM4JDTMBZ2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HTSQHOA>  it on
<https://github.com/notifications/beacon/AAPBVYOIYBSGFKZBI5FKYMLQP46RZA5CNFSM4JDTMBZ2YY3PNVWWK3TUL5
2HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HTSQHOA.gif> 

 

---
This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2019-10-23:
There was an image in that gap above, don't know where it went..., this one:
[image]

closed 2019-11-28 by MichaelAtOz