openscad/openscad #3101: Freeze with 3D mouse in Linux

xythobuz, 2019-10-23:
I'm using the newest official openscad Arch packages. They are now built with libspnav / Space / 3D mouse support. I already have a serial mouse connected and the spnav driver configured and running. It works in other applications.

I then enable Spacenav in the OpenSCAD settings and restart.
OpenSCAD now works fine as long as I don't touch the 3D mouse.
As soon as the first motion event is sent, OpenSCAD freezes completely and I can only kill it by forcing the process to exit.

I'm not sure if I can provide any relevant data. Any logs or stuff that could be useful?

t-paul, 2019-10-23:
I can't reproduce that on Debian (`libspnav` v0.2.3, `spacenavd` v0.6).

There is no special OpenSCAD log, but please add the information listed in the [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?) and details about the `spacenavd` / `libspnav` versions.

On Debian, `spacenavd` logs into `/var/log/daemon.log`

xythobuz, 2019-10-23:
I'm also running [libspnav 0.2.3](https://www.archlinux.org/packages/extra/x86_64/libspnav/) and [spacenavd 0.6](https://aur.archlinux.org/packages/spacenavd/). The arch packages should be following upstream closely, and as it is rolling release, I can not really give you a version number, but everything is up-to-date.

It shouldn't make a difference, but I'm using an old RS232 space mouse.

The spacenavd log does not provide anything interesting:

    Spacenav daemon 0.6
    adding device.
    using device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AH02FYDW-if00-port0

I'm not sure if it will help much, but here's the openscad info. Anything else you need?

> QObject::startTimer: Timers can only be used with threads started with QThread
> OpenSCAD Version: 2019.05
> System information: Linux 5.3.6-arch1-1-ARCH #1 SMP PREEMPT Fri Oct 11 18:28:05 UTC 2019 x86_64 Arch Linux 8 CPUs 15.55 GB RAM
> User Agent: OpenSCAD/2019.05 (Linux x86_64; Arch Linux)
> Compiler: GCC "9.2.0" 64bit
> MinGW build: No
> Debug build: Yes
> Boost version: 1_71
> Eigen version: 3.3.7
> CGAL version, kernels: 4.14.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
> OpenCSG version: OpenCSG 1.4.2
> Qt version: 5.13.1
> QScintilla version: 2.11.2
> InputDrivers: 
> GLib version: 2.62.2
> lodepng version: 20180910
> libzip version: 1.5.2
> fontconfig version: 2.13.91
> freetype version: 2.10.1
> harfbuzz version: 2.6.2
> lib3mf version: (not enabled)
> Application Path: /usr/bin
> Documents Path: /home/thomas/.local/share
> User Documents Path: /home/thomas
> Resource Path: /usr/share/openscad
> User Library Path: /home/thomas/.local/share/OpenSCAD/libraries
> User Config Path: /home/thomas/.config/OpenSCAD
> Backup Path: /home/thomas/.local/share/OpenSCAD/backups
> OPENSCADPATH: <not set>
> OpenSCAD library path:
>   /home/thomas/.local/share/OpenSCAD/libraries
>   /usr/share/openscad/libraries
> 
> OPENSCAD_FONT_PATH: <not set>
> OpenSCAD font path:
>   /usr/share/fonts
>   /usr/local/share/fonts
>   /home/thomas/.local/share/fonts
>   /home/thomas/.fonts
>   /usr/X11/lib/X11/fonts
>   /System/Library/Fonts
>   /Library/Fonts
>   /home/thomas/Library/Fonts
>   /usr/share/fonts/100dpi
>   /usr/share/fonts/75dpi
>   /usr/share/fonts/OpenImageIO
>   /usr/share/fonts/TTF
>   /usr/share/fonts/adobe-source-code-pro
>   /usr/share/fonts/adobe-source-sans-pro
>   /usr/share/fonts/cantarell
>   /usr/share/fonts/croscore
>   /usr/share/fonts/cyrillic
>   /usr/share/fonts/droid
>   /usr/share/fonts/encodings
>   /usr/share/fonts/gnu-free
>   /usr/share/fonts/gsfonts
>   /usr/share/fonts/liberation
>   /usr/share/fonts/mathjax
>   /usr/share/fonts/mathjax2
>   /usr/share/fonts/misc
>   /usr/share/fonts/noto
>   /usr/share/fonts/ttf-linux-libertine
>   /usr/share/fonts/ubuntu
>   /usr/share/fonts/util
>   /usr/share/fonts/encodings/large
>   /usr/share/fonts/mathjax/HTML-CSS
>   /usr/share/fonts/mathjax2/HTML-CSS
>   /usr/share/fonts/mathjax/HTML-CSS/Asana-Math
>   /usr/share/fonts/mathjax/HTML-CSS/Gyre-Pagella
>   /usr/share/fonts/mathjax/HTML-CSS/Gyre-Termes
>   /usr/share/fonts/mathjax/HTML-CSS/Latin-Modern
>   /usr/share/fonts/mathjax/HTML-CSS/Neo-Euler
>   /usr/share/fonts/mathjax/HTML-CSS/STIX-Web
>   /usr/share/fonts/mathjax/HTML-CSS/TeX
>   /usr/share/fonts/mathjax2/HTML-CSS/Asana-Math
>   /usr/share/fonts/mathjax2/HTML-CSS/Gyre-Pagella
>   /usr/share/fonts/mathjax2/HTML-CSS/Gyre-Termes
>   /usr/share/fonts/mathjax2/HTML-CSS/Latin-Modern
>   /usr/share/fonts/mathjax2/HTML-CSS/Neo-Euler
>   /usr/share/fonts/mathjax2/HTML-CSS/STIX-Web
>   /usr/share/fonts/mathjax2/HTML-CSS/TeX
>   /usr/share/fonts/mathjax2/HTML-CSS/Asana-Math/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/Gyre-Pagella/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/Gyre-Termes/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/Latin-Modern/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/Neo-Euler/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/STIX-Web/woff
>   /usr/share/fonts/mathjax2/HTML-CSS/TeX/woff
> 
> 
> GLEW version: 2.1.0
> OpenGL Version: 4.6.0 NVIDIA 435.21
> GL Renderer: GeForce GTX 760/PCIe/SSE2
> GL Vendor: NVIDIA Corporation
> RGBA(8888), depth(24), stencil(8)
> GL_ARB_framebuffer_object: yes
> GL_EXT_framebuffer_object: yes
> GL_EXT_packed_depth_stencil: yes
> GL context creator: GLX
> PNG generator: lodepng
> GLX version: 1.4
> OS info: Linux 5.3.6-arch1-1-ARCH #1 SMP PREEMPT Fri Oct 11 18:28:05 UTC 2019
> Machine: x86_64

t-paul, 2019-11-11:
I suspect that might have the same root cause as #2933.

closed 2026-04-08 by github-actions[bot]