openscad/openscad #3107: ERROR: CGAL error in CGAL_Nef_polyhedron3(

Ryan2601, 2019-10-26:
Full error:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426

I am trying to do difference on 2 parts. One stl file that was sent to me and the other (called "part to reduce" in the zip folder) is made by me.

==============================================================
code used:
    color("red") import("E:/Downloads_in_E/BombNew_TOP.stl");
        

    difference(){

            translate([0,0,7]){
                color("green") cylinder(r=18.5,h=20,center = true);
            }

            translate([0,10,7.5]){
                color("yellow") cube([37,18.5,22],center = true);
            }
        }
==============================================================

[3D_Parts.zip](https://github.com/openscad/openscad/files/3775287/3D_Parts.zip)

MichaelAtOz, 2019-10-26:
BombNew_TOP.stl is bad, it has self intersections.
[image]

Ryan2601, 2019-10-26:
how can I fix it?

MichaelAtOz, 2019-10-26:
Unfortunately Self-Intersections are hard with free tools. 
Netfabb Pro for example can do it automatically
I don't think anyone here has it????????????

Otherwise it is a manual process in the likes of MeshLab or maybe Blender, deleting the SI's and trying to close the holes that creates. I've rarely had success with finely detailed models like yours.

Sorry.

Ryan2601, 2019-10-26:
Can you please list all the tools, known to you, that does it?Free or not.

Netfabb Pro would surely fix it?

MichaelAtOz, 2019-10-26:
Nothing is guaranteed.
Netfabb used to have an online system where you could upload STLs and it would apply the auto-repair.
It had a high success rate, but not 100%. Whether that would increase using the Pro tool manually, who knows.
That appears to now be [here](https://service.netfabb.com/login.php). I have not signed up, I assume it is similar.
I'm not a tools expert, so in terms of assisted repair, Netfabb is the only one I'm familiar with.
For manually-guided fixing most mesh processing tools will help, like MeshLab, but it is iterative and a PITA, search for 'repair self intersections', but AFAIK there are no miracle guides.

MichaelAtOz, 2019-10-26:
FYI I checked on Pro pricing a while ago, one user licence $2500 AUD (I think the USD exchange rate back then was close to 1)

MichaelAtOz, 2019-10-26:
Seems there are other [Netfabb options](https://www.autodesk.com/products/netfabb/compare-products), but i don't know whether 'standard' fixes SI's, maybe try the 'premium' free trial?

MichaelAtOz, 2019-11-28:
Bad STL. Closing.

closed 2019-11-28 by MichaelAtOz

EmVeeWRK, 2021-08-10:
hmm bad stl? i reexported in blender, reverse engineered it in freecad but i am getting still same, finally i edited it in freecad without problems :(((