openscad/openscad #3108: Add support for the BlockSCAD interface in OpenSCAD

WillAdams, 2019-10-27:
Allow an option where one could enable a toggle between BlockSCAD modeling with blocks and OpenSCAD code as is allowed in the online editor for the former: https://www.blockscad3d.com/editor/

MichaelAtOz, 2019-10-27:
I have no idea what that means.
I suspect you need to detail it.
"as is allowed in the online editor for the former" if you mean modelling with blocks in OpenSCAD, I do not support this. OpenSCAD is OpenSCAD, as the tag line says "The Programmers Solid 3D CAD Modeller", it is not the GUI block modeller.
You can't have your programmed cake and eat it in GUI blocks.

t-paul, 2019-10-27:
Why not?

[Scratch](https://scratch.mit.edu/) has the slogan Imagine, Program, Share and is blocks only, [Icestudio](https://icestudio.io/) allows to creating blocks for Verilog modules, so it's not that unusual.
Additional graphical input is used in many tools, that does not necessarily mean any restriction in the use of text/code only. I bellieve code will always be the core of OpenSCAD, but also there's nothing wrong with providing more options for people to use.

Realistically that will only happen if there's someone coming in with time and motivation to drive that project.

WillAdams, 2020-08-18:
FWIW, I have been doing my best "have my cake" and eat it to.

When working on projects I start in BlockSCAD:

https://www.blockscad3d.com/editor/

once I've gotten as far as I can there, I export to OpenSCAD, find all the variable definitions cut/paste them up to the top of the file, modify them as necessary to work with the Customizer, then start in on adding features which the block interface doesn't support such as Projection()

MichaelAtOz, 2020-08-18:
What if the answer would be, that trying to re-orientate a 3D parametric modelling tool, is not suited to being a simulated CNC cutting machine?

> Performance was much better in BlockSCAD on the MacBook (though I got a warning about excessive RAM usage in Safari). 
Unfortunately things bogged down again in OpenSCAD after I exported the code, and even when I manually reduced the number of passes to get a rendering things look rather rough.

Differencing a cube from >10000 cylinders following cutting paths is not typical OpenSCAD.
We get back to OpenSCAD is a 3D CAD parametric modelling tool, not a CAM simulator.

WillAdams, 2020-08-18:
I'm open to suggestions on other tools to try --- but I rather like OpenSCAD and feel that it would benefit from improved performance and flexibility.

Now that we know what can't be done by OpenSCAD in its current state I'm looking forward to determining what can be.

t-paul, 2020-08-18:
Lets move discussions back to the mailing list if this issue here is supposed to stay open as feature request.