openscad/openscad #3112: CRASH - Display CSG Products with large number of objects

MichaelAtOz, 2019-10-31:
I was using Ronaldo's cache demo code from the [forum](http://forum.openscad.org/When-modules-are-cached-td27677.html), and a got a crash after F5, Design/Display-CSG-Products.
2019.05 & 2019.10.25 64bit, Window7/64.
[image]
Google says C...FD is Stack Overflow.
It does not crash with `cube();` or one of my complex models, I presume 50,000 is the problem.
Code:
```
//C();
B();
A(50000) 
C();
//B();

module C() square(10,center=true);
  
module B() { 
   projection() 
    linear_extrude()
      intersection(){
        projection() rotate([10,20,30]) cube(1000,center=true); 
        projection() rotate([20,30,10]) cube(1000,center=true); 
        projection() rotate([30,10,20]) cube(1000,center=true); 
        square(10,center=true);
      }
  }
  
module A(n=0)
  for(i=[for(i=0;i<n;i=i+1) i]) rotate(360*i/n) children();

```

thehans, 2019-11-01:
Can confirm that this crash also occurs on Linux (Mint 19/ Ubuntu 18.04)  
And yes, its a stack overflow issue due to recursion in `std::string CSGOperation::dump()` function from `src/csgnode.cc`.

Got this from a stack trace on a debug build:
```
[Thousands of frames at src/csgnode.cc:166]
...
#16360 0x00005555556f1722 in CSGOperation::dump[abi:cxx11]() (this=0x555561365e70) at src/csgnode.cc:166
#16361 0x00005555556f1722 in CSGOperation::dump[abi:cxx11]() (this=0x555561365f20) at src/csgnode.cc:166
#16362 0x00005555556f1776 in CSGOperation::dump[abi:cxx11]() (this=0x555561365fd0) at src/csgnode.cc:166
#16363 0x00005555559618a5 in MainWindow::actionDisplayCSGProducts() (this=0x5555566bff10) at src/mainwin.cc:2210
#16364 0x0000555555d2c2ab in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x5555566bff10, _c=QMetaObject::InvokeMetaMethod, _id=67, _a=0x7fffffffc500) at objects/moc_MainWindow.cpp:598
#16365 0x00007ffff31ab645 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16366 0x00007ffff4821122 in QAction::triggered(bool) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16367 0x00007ffff482380c in QAction::activate(QAction::ActionEvent) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16368 0x00007ffff49a00fc in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16369 0x00007ffff49a75ab in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16370 0x00007ffff49a83d3 in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16371 0x00007ffff4866048 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16372 0x00007ffff49aa6fb in QMenu::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16373 0x00007ffff482783c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16374 0x00007ffff482f65f in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16375 0x00005555559907c4 in OpenSCADApp::notify(QObject*, QEvent*) (this=0x7fffffffd4f0, object=0x5555566d8e90, event=0x7fffffffcba0) at src/OpenSCADApp.cc:55
#16376 0x00007ffff317c9c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16377 0x00007ffff482e632 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16378 0x00007ffff4880ea5 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16379 0x00007ffff48837da in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16380 0x00007ffff482783c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16381 0x00007ffff482f104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16382 0x00005555559907c4 in OpenSCADApp::notify(QObject*, QEvent*) (this=0x7fffffffd4f0, object=0x7fffe00135e0, event=0x7fffffffd020) at src/OpenSCADApp.cc:55
#16383 0x00007ffff317c9c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16384 0x00007ffff3d54583 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#16385 0x00007ffff3d56055 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#16386 0x00007ffff3d2d2eb in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#16387 0x00007fffea63a260 in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#16388 0x00007ffff6b8d417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16389 0x00007ffff6b8d650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16390 0x00007ffff6b8d6dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16391 0x00007ffff31d597f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16392 0x00007ffff317a9fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16393 0x00007ffff3183aa4 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16394 0x00005555559353e0 in gui(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, boost::filesystem::path const&, int, char**) (inputFiles=std::vector of length 1, capacity 1 = {...}, original_path=..., argc=1, argv=0x7fffffffdc58) at src/openscad.cc:732
#16395 0x000055555593818c in main(int, char**) (argc=1, argv=0x7fffffffdc58) at src/openscad.cc:1050
```

The problem is that CSGNodes are a Tree data structure where a huge single loop is treated as a union with a pathologically unbalanced tree (the tree becomes a linked list) which makes traversing it(typically done with recursion) difficult without blowing the stack.

This is similar in effect to the recursion issues addressed as part of #2343 (and later rewrite in #2886 ), but on a different recursive function this time.

thehans, 2019-11-01:
While this could be fixed by rewriting `std::string CSGOperation::dump()` to use iteration over recursion, I wonder if it would be a good time to re-think how the CSGTree is generated in the first place, to create a more balanced tree from the start.  

Maybe treat each boolean operation as a heap data structure to make it balanced.  Not sure if `std::make_heap` could be used for this somehow?

thehans, 2019-11-08:
I thought about changing how the tree is created, but wasn't sure if reordering the tree would interfere with geometry cache in some way.
So I implemented `CSGOperation::dump` without recursion.  

During testing that fix, I found that too many elements also causes stack overflow during reset of the CSG tree (for example if you do F5 twice with 200,000 elements), because `shared_ptr<CSGOperation>` for the first tree get recursively deleted.  So `CSGOperation` nodes now use a custom deleter to work around that.

It works up to 1million elements, which is the current iteration limit in `for` loops.  

I also tested temporarily raising that limit to 10million, and previewing twice in a row still worked without crashing(though it takes about 5.5minutes to preview and uses 20+GB of RAM.  I didn't test "Display CSG Products" at 10mil because I don't think I have the ram for it.

thehans, 2019-11-10:
Fixed by #3122 
Closing.

closed 2019-11-10 by thehans