openscad/openscad #3113: can't scale viewport with mouse wheel any more

Parkinbot, 2019-11-01:
Mouse wheel scale of viewport stopped to work with update to snapshot 2019.10.25.ci3851. I am using Windows 8.1. It worked in 2019.5.

MichaelAtOz, 2019-11-01:
If you mean zoom in/out, still works for me on Windows 7.

Parkinbot, 2019-11-02:
Yes, I mean zoom in/out

DaleSchultz, 2021-05-23:
I just updated to OpenSCAD from 2019.5 to 2021.01

**The mouse wheel no longer zooms the 3D object view area.**
When I roll the mouse wheel it scrolls the code window, even if I click in the 3D view area first.

This page in the manual that suggests it should work...
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface#Viewing_area
It used to work and I heavily rely on zooming in and out.

In the preferences for 3D view I have tried the checkbox "Mouse centric zoom" on and off. Same result.

I am running on **Windows 10 64 bit**
Rebooted, restarted, tried different files, even just a trivial cube()

DaleSchultz, 2021-05-23:
well I just found something more on it....

If the mouse pointer is to over the object, or to the left of it, it does not work. If the mouse cursor is towards the _right hand edge of the view area_, then zooming works!  It is as if the rectangle being monitored for mouse events is limited by an invisible windows over the drawn object.

This is a regression.

t-paul, 2021-05-23:
Is that on a HIDPI screen, e.g. 4K or similar?

DaleSchultz, 2021-05-23:
no plain old Dell monitor 1440x900

DaleSchultz, 2021-05-24:
The area where it does work is only available if the 3D view area is large enough. If one works with a small view area,  there is no area where it zooms. i.e. it is not a proportional space, it seems absolute.

With this screen layout, no area allows zooming:
[image]

oninoshiko, 2021-06-15:
enabling "scroll windows when I hover over them" under mouse settings seems to resolve this problem.

DaleSchultz, 2021-06-20:
> enabling "scroll windows when I hover over them" under mouse settings seems to resolve this problem, at least for me.

yes! thanks. This is a setting in the Windows mouse settings. What this seems to indicate is that for some reason the 3D view rectangle is no longer regarded as 'active' (or rather just the right hand part of it is regarded as 'active').

So this represents a workaround for the changed behavior.

bnesheim, 2022-05-05:
Since this is still a problem shifting the preview window over to the left will "fix" the problem.

The area for the Edit window is wrong so that it register some distanse outside the source window and grab the mouse scroll instance.