openscad/openscad #3114: Bad STL file exported for Pela Block socket panel

hartzell, 2019-11-01:
I'm working with objects from the [Pela Blocks project](https://github.com/LEGO-compatible-gadgets/PELA-parametric-blocks).  After calibrating for my Prusa printer with the current version of PrusaSlicer, I tried to print a socket panel, but one of the sockets has a layer of material about halfway through.  [I mentioned it on the Prusa forum](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/prusaslicer-bug-slicing-pela-block-socket-panel/#post-172042) and one of the other members commented that the STL file was "bad" (according to a Microsoft tool that I can't run).

So, I'm bringing it up here.

Here's what the file looks like when I import it into PrusaSlicer:

[image]

And here's what it looks like post-slicing (and indeed, how it prints):

[image]

I've attached a zipfile that contains just the required files from the [Pela Bricks project](https://github.com/LEGO-compatible-gadgets/PELA-parametric-blocks) that are required to recreate the issue.

Is this an OpenSCAD issue, or ???

t-paul, 2019-11-01:
There's no attachment here. As general note, please try to make the example as small as possible and don't just dump big projects here with questions. There might be people who can help on the openscad forum to narrow down issues.

nophead, 2019-11-01:
The STL looks fine to me in NetFabb studio and slices fine in Skeinforge.

hartzell, 2019-11-01:
Thanks for the attention.

This should be the bug demonstration: [bug-demo.zip](https://github.com/openscad/openscad/files/3799866/bug-demo.zip)

@t-paul -- I trimmed as much of the project away as I could reasonably do.  It's down to the model itself and some supporting files.

@nophead -- @t-paul is right, I don't seem to have managed to attach the file.  What did you slice?

t-paul, 2019-11-01:
The ZIP is empty. Did you use Linux `zip`? Then try `zip -9r bug-demo.zip bug-demo` to add files recursively.

nophead, 2019-11-01:
I sliced the STL you posted on the Prusa forum you linked to.

[image]

[image]

hartzell, 2019-11-01:
@nophead -- Thanks!

Sorry for flubbing the zip file, I'm more of a `.tgz` kind of guy.

Hopefully the third time will be the charm.  It also includes the "fixed" file that was posted in the Prusa forum: [bug-demo.zip](https://github.com/openscad/openscad/files/3799888/bug-demo.zip)

hartzell, 2019-11-01:
@nophead -- Can I double check that you didn't test with the "fixed" STL file?

nophead, 2019-11-01:
It was this one:  Here's a link to the file, temporarily hosted:
https://shrimp.alerce.com/tmp/PELA-socket-panel.stl

hartzell, 2019-11-01:
@nophead -- Yep, that's the "bad" one.  Just pulled it down and fed it to PrusaSlicer w/ the same screwy result.

Interestingly, it's a text file but the "fixed" file from the forum is some sort of binary.

Not sure where the bug lies....

nophead, 2019-11-01:
There are two formats of STL file: text and binary. OpenSCAD only exports the text version. The two formats should give the same data when read though.

t-paul, 2019-11-01:
`admesh` reports the file as ok:

```
$ admesh PELA-socket-panel.stl 
ADMesh version 0.98.3, Copyright (C) 1995, 1996 Anthony D. Martin
ADMesh comes with NO WARRANTY.  This is free software, and you are welcome to
redistribute it under certain conditions.  See the file COPYING for details.
Opening PELA-socket-panel.stl
Checking exact...
All facets connected.  No nearby check necessary.
No unconnected need to be removed.
No holes need to be filled.
Checking normal directions...
Checking normal values...
Calculating volume...
Verifying neighbors...

================= Results produced by ADMesh version 0.98.3 ================
Input file         : PELA-socket-panel.stl
File type          : ASCII STL file
Header             : solid OpenSCAD_Model
============== Size ==============
Min X =  0.099999, Max X =  23.900000
Min Y =  0.099999, Max Y =  31.900000
Min Z =  0.000000, Max Z =  4.000000
========= Facet Status ========== Original ============ Final ====
Number of facets                 :  4342                4342
Facets with 1 disconnected edge  :     0                   0
Facets with 2 disconnected edges :     0                   0
Facets with 3 disconnected edges :     0                   0
Total disconnected facets        :     0                   0
=== Processing Statistics ===     ===== Other Statistics =====
Number of parts       :     1        Volume   :  1423.480835
Degenerate facets     :     0
Edges fixed           :     0
Facets removed        :     0
Facets added          :     0
Facets reversed       :     0
Backwards edges       :     0
Normals fixed         :     0
```
And processing it though OpenSCAD (together with an additional `cube()` to force a CGAL union) also works withough issue.

So I'm going to close the issue as there seems to be no problem on OpenSCAD side.

closed 2019-11-01 by t-paul

hartzell, 2019-11-01:
@t-paul -- **Thanks!**

hartzell, 2019-11-01:
(and @nophead!)