openscad/openscad #3116: Support for lib3mf 2

diabonas, 2019-11-02:
Version 2 of lib3mf has been [released on 2019-09-26](https://github.com/3MFConsortium/lib3mf/releases/tag/v2.0.0). This is a major release that breaks ABI compatibility, see the changelog:

> - **autogenerated** language bindings for C++, C, Python, Pascal, C#, Golang and NodeJS have been added
(the defunct COM interface has thus been removed)

Since OpenSCAD relies on the mentioned COM interface to allow import and export of 3MF files, it doesn't build with the most recent version of lib3mf. It would be nice to make OpenSCAD compatible with the new version 2.0.0 in order to allow distributions to drop support for the legacy version 1.8.1.

t-paul, 2019-11-02:
Which distribution supports lib3mf 2.0? So far it's enough struggle to have 1.8 supported.

diabonas, 2019-11-02:
Well, I just added [lib3mf-1](https://www.archlinux.org/packages/community/x86_64/lib3mf-1/) to Arch Linux to get 3MF support for OpenSCAD. I would love to use the most recent version instead (see the new [lib3mf](https://aur.archlinux.org/packages/lib3mf/) Arch User Repository (AUR) package), but failed to compile OpenSCAD with it due to the different interfaces.

t-paul, 2019-11-02:
We can't update until 2.0 is available at least for: Debian/Ubuntu, MacOS, Windows (MXE). So without help, that's not going to be very soon.

diabonas, 2019-11-03:
Sure, I can have a look into that if I have some time. The build process itself is unchanged from the old version, the only potential issue I found is that the unit tests require the most recent version 1.10.0 of gtest.

As an interim solution, it might be possible to support both versions simultaneously with a version switch, although I haven't looked into that yet.

t-paul, 2019-11-03:
Sorry, I did mean help in general. I ended up as maintainer of lib3mf in Debian and for 2.0 I assume the interface generator needs to be packaged too. I hope to get some help for that. Also I have to look if the options to use system LIBZIP and ZLIB still exist as Debian will not accept packages with a local copy of those.

If the API difference is not too big it might be an option to support both. We don't use much of the API, really just the core mesh store. Not the perfect solution, but might help to get things earlier. There's even more work going on for the AppImage and Snap packages and doing that as just one person is not scaling too well.

diabonas, 2019-11-03:
> Also I have to look if the options to use system LIBZIP and ZLIB still exist as Debian will not accept packages with a local copy of those.

They do, I just tested this, although even in version 1.8.1 the situation is less than ideal because lib3mf is not explicitly linked against these libraries. I have opened https://github.com/3MFConsortium/lib3mf/pull/184 and https://github.com/3MFConsortium/lib3mf/pull/185 to fix the situation for the legacy and current branch, resp. Since you are the Debian packager, I would very much welcome any feedback on these PRs from you.

t-paul, 2020-11-11:
Implemented via #3481.

closed 2020-11-11 by t-paul