openscad/openscad #3117: Rendering not works on same module to be multiplied

cidylong, 2019-11-03:
OpenSCAD version 2019-08-10.
tried on both macOS 10.15 and windows 7.

single module rendering works fine, but if put the same module into a for loop to multiple it, 

Rendering not work at all; Error message as:

_Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_external_structure.h Line: 1153
Geometries in cache: 27
Geometry cache size in bytes: 237264
CGAL Polyhedrons in cache: 32
CGAL cache size in bytes: 37453600
Total rendering time: 0 hours, 0 minutes, 10 seconds
Rendering finished._

source code as: 

```
res_s= 32;
res_m= 64;
res_b=128;
height = 250;

/*turbine_wing();*/
twisted_turbine();

module twisted_turbine(){
    union(){
        for(i=[0:1:4]){
            rotate([0,0,72*i])turbine_wing();
        }
    }
}

module turbine_wing(){
    translate([0,0,0])union(){
        twisted_blade();
        mount_bracket();
        rotate([12,0,0])translate([0,-322,6])rotate([0,0,90])rotate([0,90,0])cylinder(h=565,d=20,center=true,$fn=res_s);
        rotate([-12,0,0])translate([0,-322,-6])rotate([0,0,90])rotate([0,90,0])cylinder(h=565,d=20,center=true,$fn=res_s);
    }
}

module mount_bracket(){
    union(){
        rotate([0,0,-36])bracket_wing();
        translate([0,0,-15])bracket_through();
        translate([0,0,-15])rotate([0,0,36])bracket_wing();
    }
}

module bracket_through(){
    rotate([0,0,-90-36/2])rotate_extrude(angle = 36, convexity =4)translate([10,0,0])square([30,30],$fn=res_m);
    }

module bracket_wing(){
    difference(){
        rotate([0,0,-90-36/2])rotate_extrude(angle = 36, convexity =4)translate([10,0,0])square([30,15],$fn=res_m);
        translate([0,-25,7.5])cylinder(h=15,d=8,center=true);
    }
}

module twisted_blade(){
    translate([0,-500,0])
    linear_extrude(height = height*3, center = true, convexity = 20, twist = -height/6)translate([0, 0, 0])newMoon();
    /*rotate([0,0,22.5])translate([0,0,-25])linear_extrude(height = 50, center = true, convexity = 20, twist = 22.5)translate([0, 0, 0])newMoon();*/
}
module newMoon(){
    translate([0,-height/10])difference(){
        scale([1.5,0.6])circle(d=height,$fn=res_b);
        translate([0,height/20])scale([1.45,0.5])circle(d=height,$fn=res_b);
    }
}
```

thehans, 2019-11-03:
Looks like your "newMoon" is not manifold.  I think the scaled ellipse that you subtract is causing an infinitely thin sliver of geometry.  It works ok if you offset it slightly:  `translate([0,height/20+0.3]) ...`

cidylong, 2019-11-03:
Thank you! Thehans. It works. But why? Could you please explain to me?

MichaelAtOz, 2019-11-03:
Changing `res_b=112` also fixes it.
As thehans mentioned, CGAL can't handle infinitely thin geometry. In this case a very thin wedge at the tip(s?) of the moon.
As the two circles are made up of segments, the $fn determines the precise angle of a segment. 
At that specific point the segment of one ellipse and the segment of the other line-up too close.
Changing the $fn changes the angle slightly, making a less pointy triangle.

I do find this interesting tho, any triangle will eventually have an infinitely thin end point. I wonder what causes a triangle to become too thin.

nophead, 2019-11-03:
Not sure what you mean. Any triangle will have a finite width unless it is degenerate.

Perhaps the tips of the moon become degenerate.

MichaelAtOz, 2019-11-04:
Hmmm,
If you use
```
translate([-178.98006,2.74147])
  newMoon();
```
(or +178...) the points will be near the origin, so zooming in is easy.
[image]

That is definitely a different angle at the arrows, but OpenSCAD is not showing edges.

Close up.
[image]

Is this a bug? Looks like one.

MichaelAtOz, 2019-11-04:
`linear_extrude(height = 1)` chops it up
[image]

With a `cube()` it does NOT get a CGAL error.

But 
```
res_b=128;
height = 250;
cube();
linear_extrude(height = height*3, center = true, convexity = 20, twist = -height/6)
translate([178.98006,2.54147])
    translate([0,-height/10])
      difference(){
        scale([1.5,0.6])
          circle(d=height,$fn=res_b);
        translate([0,height/20])
          scale([1.45,0.5])
            circle(d=height,$fn=res_b);
      }
```
Does. 
> ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152

There is definitely something fishy with linear-extrude(), it keeps popping-up in various CGAL errors.
I can't see why the above should not be handled properly.

cidylong, 2019-11-04:
Thank you for all of your time to explain to me, seems OpenSCAD is still under developing to handle what will see in the future.

cidylong, 2019-11-04:
I'm still very confused, if you comment out function `twisted_turbine();` and un-comment `/*turbine_wing();*/`, which means, only render a single module.

OpenSCAD will render it properly.

Only when rendering it with whole group of modules as turbine wheel, error was thrown.

Please advise.

MichaelAtOz, 2019-11-04:
If there is only one object (and it doesn't use some of the more complex operations), CGAL does not always need to be called. So you don't get the error, it seems to render, you may be able to export the geometry, but it will usually be broken. That is why you didn't get an error until you used the `for()`.

To force CGAL, add one more object, the simplest is `cube();`, this forces a `union()` and hence GCAL processes both objects, and if bad, will generate the error. That is why there is  `cube();` in my code snippet above. 

Also note that geometry is cached, so you may get an error, but if you do another F6 you won't always get the error again, as it is cached, so it doesn't re-process that bit. If you are investigating CGAL errors, you need to Design/Flush-Cache before F6.

cidylong, 2019-11-04:
Hi. Michael. Thank you very much for your advice.

MichaelAtOz, 2019-11-11:
> But
>
> ```
> res_b=128;
> height = 250;
> cube();
> linear_extrude(height = height*3, center = true, convexity = 20, twist = -height/6)
> translate([178.98006,2.54147])
>     translate([0,-height/10]) [...]

Removing the `twist` does not get the CGAL error.

MichaelAtOz, 2019-11-12:
> But
>
> ```
> res_b=128;
> height = 250;
> cube();
> linear_extrude(height = height*3, center = true, convexity = 20, twist = -height/6)
> translate([178.98006,2.54147])
>     translate([0,-height/10]) [...]

Removing the `twist` does not get the CGAL error.

With `twist` and the $fn being zero in `linear_extrude()`, it selects `slices=3` this gets the CGAL error, specifying `slices=4` does not.

Is there any debug= that will produce the polygons fed into CGAL??

bakerjvb, 2019-12-16:
Same issue: Replicating an apparantly clean stl results in  GCAL errors
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426
openscad code:
for  (i = [0:45:315])rotate ([0,0,i])
hmm you don't support STL imports.
thats a shame.

MichaelAtOz, 2019-12-16:
> an apparantly clean stl

Unless you share it, it is probably bad...

THIS issue is NOT about importing probably bad STLs.
This issue is about OpenSCAD itself, using only normal code, producing CGAL errors.

bakerjvb, 2019-12-16:
I agree. This stl is created with normal code. I read the exchanges abobe and yes I originally had a fine point to the pencil like object with a hole through it. I had already changed this to ensure the minimum diameter was .1mm.
I have also passed the stl through meshlab, meshmixer netfabb and prusaslicer repair and the GCAL error is still reported.
I dont want to move to Fusion because I know and love Openscad, but this does make things difficult.

MichaelAtOz, 2019-12-16:
If you are not willing to share the STL (and/or the OpenSCAD code you created it with) it is not possible to confirm if there is a bug or not.
If you don't/can't share it publicly you could email it to me, oz.at.michael`@`gmail.com, and I'll go through it.

MichaelAtOz, 2019-12-17:
Well @cidylong took me up on the invite, it was a bad STL (holes & SI's) causing the CGAL error on `import()`.
I fixed the OpenSCAD code which generated it; manifold issues with just-touching faces etc, plus the `linear_extrude()` with `twist` getting `slices=3` causing long thin triangles making SI's, increasing slices fixed the final issue.

So I think we need to select a better default `slices=`.

@bakerjvb do you want to share your code/STL?

bakerjvb, 2019-12-19:
Dear michael,
How very kind of you, I gave up on the openscad replication and remade this
strut with an open end and replicated it with PrusaSlicer.
The interesting thins you should know and probably the source of the GCAL
errors is that in sharpening the strut I used a difference between a cone
and a cube to create a sharpener and then used another difference between
the strut and the sharpener to create the pointed end.
funnily enough That seemed to replicate OK and I then used another
difference between the sharpened strut and an annulus (that was created
with a rotate extrude of a circle) to create the hole in the end, and it
seemed that it was this that failed..
Anyway I sense that openscad, much as I adore its programatical interface
is failing to greate solid STLs when using Difference.
Best wishes.
Julian Baker

bakerjvb, 2019-12-19:
Dear Michael, Ive regenerated the code to create the sharpened strut with a
hole in it.
  The strut is created in 4 stages. At each stage I created and stl and
then imported the stl in the next stage.
1 create a pillar by extruding two circles and a square
2 create the sharpener (just like a pencil sharpener)
3 take the sharpener away from the column using difference
4 add a hole.
heres the code
// to create a pillar: pillar 1

//linear_extrude (height=80)
//union(){
//translate ([0,-1.1,0])
//color ("blue")//rotate ([90,0,0])
//circle ($fn=40,d=4);
//translate ([0,1.1,0])
//color ("blue")//rotate ([90,0,0])
//circle ($fn=40,d=4);
//translate ([-2,-1.1,0])
//    resize ([4,2.2,0])
//square (4);}

//to create sharpner 1

//difference (){translate([-7,-7,60]) cube ([13,50,20]);
//$fn=60;
//translate ([0,0,60])
////rotate ([90,0,0])
//cylinder (r1=3.5,r2=0.1, h=25);}

// to sharpen the pillar

//difference(){
//import("C:/Users/baker/Documents/home/diy/rabbit hole/pillar 1.stl");
//import("C:/Users/baker/Documents/home/diy/rabbit hole/sharpner 1.stl");}

//to add a hole

difference (){
import("C:/Users/baker/Documents/home/diy/rabbit hole/sharpened strut mm
solid 1.stl");
union(){color ("red")
translate ([-4,-0,10])
rotate ([0,90,0])
cylinder ($fn=60,d=3.4,h=8);
translate ([-4,-0,-3.5])rotate ([45,0,0])cube (10);}}

rgds Julian

nophead, 2019-12-19:
You shouldn't create a perfect knife edge with CSG because it has zero
thickness at the tip. Make a real knife edge with a finite thickness and it
should work fine.

bakerjvb, 2019-12-19:
The tip is finite. note r2=0.1
cylinder (r1=3.5,r2=0.1, h=25);}
rgds Julian

thehans, 2019-12-19:
@nophead all edges have zero thickness.  The only problem is when an edge shares more than two faces eg: `cube(); translate([1,1] )cube();` but I don't think that is the issue here.

@bakerjvb why complicate with multiple exports and imports?

Here is a version of your code cleaned up and modified to produce the output in a single step.  I don't see any CGAL errors and I even changed it to come to a infinitesimal point.
```
difference() {
  //to create a pillar: pillar 1
  linear_extrude (height=86) 
    union() {
      color ("blue")
        translate ([0,-1.1,0])
          circle ($fn=40,d=4);
      color ("blue")
        translate ([0,1.1,0])
          circle ($fn=40,d=4);
      translate ([-2,-1.1,0])
        resize ([4,2.2,0])
          square (4);
    }
  // to sharpen the pillar
  difference () {
    translate([-7,-7,60]) 
      cube ([13,13,27]);
    translate ([0,0,60])
      cylinder (r1=3.5,r2=0, h=25, $fn=60);
  }
  //to add a hole
  union(){
    color ("red") 
      translate ([-4,-0,10])
        rotate ([0,90,0]) 
          cylinder ($fn=60,d=3.4,h=8);
    translate ([-4,-0,-3.5]) 
      rotate ([45,0,0]) 
        cube (10); 
  }
}
```

Note that to post code on github: use 3 back-ticks <fixed>```<fixed> on the line before and after a block of code so it can be formatted properly.

nophead, 2019-12-19:
> Note that to post code on github: use 3 back-ticks ``` on the line before

I don't know it applies to this case but a knife edge in CSG is what you
get when you subtract a circle from square that is the same size. You get
an infinitely thin wall at the edge and that breaks things.

MichaelAtOz, 2019-12-19:
@bakerjvb 
I recreated your scad code, see the zip file attached.
[strut.zip](https://github.com/openscad/openscad/files/3985950/strut.zip)
Using `!` I previewed then rendered each step, exporting the STLs [also in the zip], the STLs are clean.
The final step worked without any CGAL errors, and the resultant strut.stl is clean.

Could you check my scad code to ensure it is the same as yours.
I used OpenSCAD 2019.05, the current release version, what are you using?

bakerjvb, 2019-12-20:
Dear Michael,
Absolutely correct and checked, Thank you.

So to find out what went wrong, I have looked back at my original code and found I had an anulus creating a curved hole in the strut.
This renders fine and I can create an STL.
However when I replicate that stl and render it, OPENSCAD gets to 999 and simply closes without an error (or saving the code.)

So the source of the replication error is this code (I think!)
difference (){
    translate ([5,0,8.5])
   rotate ([90,0,0])import("C:/Users/baker/Documents/home/diy/rabbit hole/michaels sharpened strut.stl");translate ([0,0,8.4]) 
   rotate_extrude(angle=360,$fn=60)
translate ([11,0,0])
   circle (d=3.4);}
This renders fine.
but
The rotation is implemented in this code
for  (i = [0:45:315])rotate ([0,0,i])
import("C:/Users/baker/Documents/home/diy/rabbit hole/michaels sharpened strut with curved hole.stl");  
and the abend error is occurring when rendering.

All the very best for this Christmas time,
Yours,
Julian

From: Michael
Sent: 19 December 2019 23:05
To: openscad/openscad
Cc: bakerjvb; Mention
Subject: Re: [openscad/openscad] Rendering not works on same module to bemultiplied (#3117)

@bakerjvb
I recreated your scad code, see the zip file attached.
strut.zip
Using ! I previewed then rendered each step, exporting the STLs [also in the zip], the STLs are clean.
The final step worked without any CGAL errors, and the resultant strut.stl is clean.
Could you check my scad code to ensure it is the same as yours.
I used OpenSCAD 2019.05, the current release version, what are you using?

MichaelAtOz, 2019-12-23:
@bakerjvb 
(note, use three backticks ``` on the line before and after code)
```
    difference (){ 
    translate ([5,0,8.5]) 
      rotate ([90,0,0])
        import("sharpened-strut.stl");
    #translate ([0,0,8.4]) 
      rotate_extrude(angle=360,$fn=60) 
        translate ([11,0,0]) 
          circle (d=3.4);}
```

Yes that produces a STL with self-intersections :( 
[image]
So the subsequent `rotate()`'s probably get a CGAL error.

> simply closes without an error (or saving the code.)

If you are running Linux, run OpenSCAD from terminal, you'll probably see an error there.
And Render does a backup save just in case, see
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Paths (under misc)

Close-up shows where triangulation is wrong
[image]

Looks like a bug.

bakerjvb, 2019-12-23:
You wizzard.
BTW i'm on the latest version

OpenSCAD <https://www.openscad.org/> version 2019.11.20.ci3989 (git
eedfcc4e)

on 64 bit windows 10. Hardware is i7 64GB memory and 250GBSSD (a second
hand steal!)

If I remember rightly base 2015 version does give a GCAL error properly
without closing: I upgraded to see if the error would go away.

Thank you for the tip about backticks.

Best wishes for a great Christmas.

Julian

bakerjvb, 2019-12-23:
Just to delve into this problem more.

bakerjvb, 2019-12-23:
Just to delve into this problem a little more. The strut doesn't have to be
complex.
A simple cube will suffice.
A curved hole cut into it will create an STL successfully. EG
```
difference (){
    translate ([5,-2,0])
cube ([20,4,4]);
    translate ([0,0,2])
rotate_extrude(angle=360,$fn=60)
    translate ([11,0,0])
    circle (d=3.4);}
```
This creates an stl and renders it successfully.

However rotating this stl with this code (below) results in an Openscad
abend.
```
for  (i = [0:45:360])rotate ([0,0,i])
translate ([10,0,0])import("C:/Users/baker/Documents/home/3d/problem
investigation.stl");
```

Secondly
A simple rotate rotate ([0,0,45]) is rendered correctly.

Thirdly
a single "for" loop with a single replication causes the abend
eg: for  (i = [0:45:46])    rotate ([0,0,i])

Finally
replicating the cube first and then subtracting the anulus works and will
render properly.
eg using this code:
difference (){
for  (i = [1:45:360])
    rotate ([0,0,i])
    translate ([5,-2,0])
cube ([20,4,4]);
    translate ([0,0,2])
rotate_extrude(angle=360,$fn=60)
    translate ([11,0,0])
    circle (d=3.4);}
```
I hope this helps in narrowing down the problem.
best wishes for a happy Christmas,
Julian

On Mon, 23 Dec 2019 at 12:47, Julian Baker <bakerjvb@gmail.com> wrote:

> Just to delve into this problem more.
>
> On Mon, 23 Dec 2019 at 09:23, Julian Baker <bakerjvb@gmail.com> wrote:
>
>> You wizzard.
>> BTW i'm on the latest version
>>
>> OpenSCAD <https://www.openscad.org/> version 2019.11.20.ci3989 (git
>> eedfcc4e)
>>
>> on 64 bit windows 10. Hardware is i7 64GB memory and 250GBSSD (a second
>> hand steal!)
>>
>> If I remember rightly base 2015 version does give a GCAL error properly
>> without closing: I upgraded to see if the error would go away.
>>
>> Thank you for the tip about backticks.
>>
>> Best wishes for a great Christmas.
>>
>> Julian
>>
>> On Mon, 23 Dec 2019 at 08:08, Michael <notifications@github.com> wrote:
>>
>>> @bakerjvb <https://github.com/bakerjvb>
>>> (note, use three backticks ``` on the line before and after code)
>>>
>>>     difference (){
>>>     translate ([5,0,8.5])
>>>       rotate ([90,0,0])
>>>         import("sharpened-strut.stl");
>>>     #translate ([0,0,8.4])
>>>       rotate_extrude(angle=360,$fn=60)
>>>         translate ([11,0,0])
>>>           circle (d=3.4);}
>>>
>>> Yes that produces a STL with self-intersections :(
>>> [image: Capture#3117-sharpened-curved-hole-si]
>>> <https://user-images.githubusercontent.com/1972961/71344011-8fcb6000-25b5-11ea-8538-97a6eb86c890.JPG>
>>> So the subsequent rotate()'s probably get a CGAL error.
>>>
>>> simply closes without an error (or saving the code.)
>>>
>>> If you are running Linux, run OpenSCAD from terminal, you'll probably
>>> see an error there.
>>> And Render does a backup save just in case, see
>>> https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Paths (under misc)
>>>
>>> Close-up shows where triangulation is wrong
>>> [image: Capture#3117-sharpened-curved-hole-si-closeup]
>>> <https://user-images.githubusercontent.com/1972961/71344780-8f33c900-25b7-11ea-81b7-b8e864c7ac09.JPG>
>>>
>>> Looks like a bug.
>>>
>>> —
>>> You are receiving this because you were mentioned.
>>> Reply to this email directly, view it on GitHub
>>> <https://github.com/openscad/openscad/issues/3117?email_source=notifications&email_token=AKS7SOGXGG7GT366ZX6EZDLQ2BWZPA5CNFSM4JIJ47YKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHQQTIY#issuecomment-568396195>,
>>> or unsubscribe
>>> <https://github.com/notifications/unsubscribe-auth/AKS7SOCC6RRQDJYI2HTCHU3Q2BWZPANCNFSM4JIJ47YA>
>>> .
>>>
>>

thehans, 2019-12-23:
> (note, use three backticks ``` on the line before and after code)

Weird, I guess the backticks don't work for email replies? (vs replying through github website interface).  
Maybe it doesn't interpret markdown at all when it comes via email.

bakerjvb, 2019-12-23:
now this is interesting
I tried similar code on an old machine with

OpenSCAD <https://www.openscad.org/> version 2019.01-RC4

and the stl WAS rendered withoit error..

```for (i=[45:45:360])
    rotate ([0,0,i])
difference (){translate ([10,-3,6])rotate([ 0,90,0])cube ([6,6,10]);
translate ([0,0,3])
rotate_extrude($fn=60,angle=360)
translate ([15,0,0])circle(d=3.4,$fn=60);}
```

MichaelAtOz, 2019-12-24:
@bakerjvb the problem is with OpenSCAD internals, it uses a small grid to 'align' vertices, sometimes with very small numbers, things like very thin triangles, the point is 'offset' by the grid, and produces Self-Intersections and/or Degenerate Triangles. It has been a problem for a while, and no easy fix comes to mind.
Your code produces the segment of the torus, being cut by the edge of the cube at a fine angle;
[image]

So for now, when such things happen you look for such spots, then think how you can move the joining position. Not ideal.

In the example in your last post I had `rotate(2)` in front of the rotate extrude, which moves the segment joint away from the box edge, and it produces a clean STL.
Another workaround is to play with $fn as that will change the position of such joins.

Not ideal.

MichaelAtOz, 2019-12-24:
OP issue was down to default `slices=` producing thin triangles [and various other problems], should we be look to changing the default to a larger number?

2nd problem, the torus/cube, also thin triangles.

Grid/float root cause is tracked in #2686 or #2991.

MichaelAtOz, 2019-12-24:
I managed to reproduce the crash, using 2019.11.20.ci389.
[cube-torus2.zip](https://github.com/openscad/openscad/files/3996989/cube-torus2.zip)
cube-turus2.scad, F6, export-> rotate.stl, rotate.scad, F6, crash.
[image]

Eventviewer:
```
Exception code: 0x40000015
Fault offset: 0x000000000037c773
Faulting process id: 0x100c
Faulting application start time: 0x01d5ba015e5ce1fa
Faulting application path: C:\Program Files\openscad-2019.11.20.ci3989\openscad.exe
Faulting module path: C:\Program Files\openscad-2019.11.20.ci3989\openscad.exe
```
Google says 'unknown software exception' abort not handled.

Above was on Windows. On linux with 2019.11.24 it gets
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1149 

That is possibly #3150, but it also points to OpenSCAD bad geometry. Something is fishy...

MichaelAtOz, 2019-12-24:
@bakerjvb Looks like you are crashing with 2019.11.20, probably due to #3150, I'd use 2019.05 - at least there you get a CGAL error, not a crash... the cause is difficult to trace.