openscad/openscad #3119: bogus window with bspwm

nobrowser, 2019-11-04:
Hi, I use [bspwm](https://github.com/baskerville/bspwm) which is one of the "tiling" window managers for XWindow.  How it works: a newly created window on an empty workspace is full size, ie. almost maximized except for a few pixel margin at the edges of the screen.  This is true for most normal windows, however dialog windows will usually be their natural size, centered and "float" in front of all the normal windows.  How bspwm decides what is a dialog is more involved than I want to investigate and go into here.  But anyway, when I start openscad-2019.05 and open a file (as I am a beginner, this is always one of the example files), _sometimes_ but not always one of these centered windows appears.  Its background seems to be a blitted rectangle from the preview of the file, and it has no widgets or text in it - so I cannot even close it by normal means :-(

This doesn't happen with the same version of openscad under the Xfce desktop environment.

If really needed I think I could get a screenshot.

Finally, I want to mention that bspwm has a "rule" feature that is meant to help in corner cases like this.  But the rules match on the WM_CLASS and WM_NAME properties, and the values for this bogus window are "openscad" and "OpenSCAD" which isn't specific enough to be useful. :-(

BlitzKraft, 2019-11-04:
I use bspwm. Is there anything else specific needed to reproduce this issue? I am currently unable to reproduce this.

MichaelAtOz, 2019-11-04:
Is it perhaps text() usage that triggers it. It produces a brief pop-up as it builds the font cache, which takes 'no' time when nothing new needs doing.

nobrowser, 2019-11-04:
@MichaelAtOz , that is a very good guess.  I only get it when I open LetterBlock.scad or text_on_cube.scad , never with logo.scad.

nobrowser, 2019-11-12:
I am not sure if I agree with the New Feature classificiation.  Quite apart from its geometry and staying power which may be blamed on my unusual environment, the _contents_ of the dialog is wrong - basically garbage bits from somewhere in memory.  That seems to me to be a bug in need of a fix.

nobrowser, 2019-11-12:
Indeed, commenting out everything in `OpenSCADApp::showFontCacheDialog()` and `OpenSCADApp::hideFontCacheDialog()` makes it go away, so I think @MichaelAtOz is right.

stephen304, 2021-05-27:
Since I seem to be having this issue, and since nobody has shared a screenshot, this is what it looks like for me. It persists across all workspaces and is unable to be focused or moved, which is a bit annoying when working on anything else while openscad is open.

[image]
[image]

your-friend-alice, 2021-11-08:
I'm also seeing this issue, on Debian 11 with bspwm `0.9.10`

kwshi, 2021-12-13:
I can also reproduce this issue, bspwm 0.9.10 on Void Linux.  By the way, do we know whether this is an issue on bspwm's side or on OpenSCAD's side?

t-paul, 2021-12-13:
It's likely an OpenSCAD issue with the dialog shown for the font cache updates. This is supposed to run in a separate thread so the application can startup normally while updating the cache. There might be better options now that Qt4 support was dropped.

ryanw, 2022-04-23:
Is there anyway to move this window? It's right in the middle and really annoying. I've never seen a window that I couldn't at least move

netthier, 2023-01-16:
Is there any progress on this? It still occurs with the latest OpenSCAD and bspwm versions.