openscad/openscad #3120: Public accessible server and using user upload openSCAD code safely.

mmiscool, 2019-11-05:
Hello,

I am looking for the best practices when implementing openSCAD in a server environment. What commands would be best to filter? 

The include command is obvious but what other commands can access other files from the file system?

Just trying to build a good list.

t-paul, 2019-11-05:
I don't think there's any existing discussion on that yet. One option would be to look at [compiler explorer](https://godbolt.org/) which allows to compile and run programs in various languages (and I'm wondering if we could add OpenSCAD there too, but that's a different topic). This is mostly based on Linux + `firejail` with plan to move to `nsjail`. Otherwise a `systemd` based [isolation](http://0pointer.de/blog/projects/security.html) might work too.

thehans, 2019-11-05:
In addition to `include`, there's also `use` of course, and the `import` module.
`surface` also accesses the filesystem.

And then there's a bunch of deprecated variations on the import module:
"dxf_linear_extrude", "dxf_rotate_extrude" "import_stl" "import_dxf", "import_off"

Also deprecated is the `file` parameter for `rotate_extrude` and `linear_extrude`.

That's everything I can think of with file access.

But I would suggest if possible: instead of disabling a bunch of features, to allow them in some sort of sandboxed environment, with access only to relative paths within the "working directory".  
This is one thing I wish Thingiverse customizer for example would implement, because it can be difficult to integrate all dependencies into a single .scad file for some designs.

mmiscool, 2019-11-06:
I have been looking for how it might be possible to do with chroot. 

I need to dynamically set the folder it should be able to access each time. Have not really figured out how to do it as a single command in chroot. There could be a bunch of different instances of it running all at once with different working directories.

t-paul, 2019-11-06:
I'd suggest to have a look at the https://github.com/google/nsjail readme which shows a number of one line examples. This allows for full isolation from network, users, processes and other parts of the base system by using the Linux kernel namespace/cgroups features (which is also used by container tools like docker, lxd, ...). Using only `chroot` does not really give much isolation security wise.
(Disclaimer again... I have not used it myself yet, but it was mentioned in a CppCon talk as possible replacement of firejail, github reference should be https://github.com/mattgodbolt/compiler-explorer/pull/1440).

closed 2020-01-07 by kintel