openscad/openscad #3124: Allow include/use/import over http/https

lhartmann, 2019-11-10:
Allowing include, use and import statements to fetch remote resources could simplify usage of shared scad libraries. E.g.:

```
include <https://openscad.org/libs/MCAD/v2019.5/units.scad>
include <https://openscad.org/libs/MCAD/v2019.5/bearing.scad>
include <https://openscad.org/libs/MCAD/v2019.5/trochoids.scad>
include <https://cdn.weird.site/latest/three_sided_square.scad>
```

It is up to the user to choose reliable, permanent links. Remote files could be cached forever like [Deno](https://github.com/denoland/deno) does, or per session.

Alternatively include<> syntax could be changed to function like, and concatenate parameters like str(...) does:

```
MCAD = "https://openscad.org/libs/MCAD/v2019.5/"; // Simple version control
include(MCAD, "units.scad");
include(MCAD, "bearings.scad");
include(MCAD, "trochoids.scad");
```

P.S.: All URLs are fictional.

t-paul, 2019-11-11:
This seems tempting, but I believe that kind of file handling belongs into a separate package manager.

lhartmann, 2019-11-11:
The creator of nodejs later created deno, with an explicit goal of getting rid of packager managers.

Package management adds complexity, extra files to configure packages and build system. Direct linking could allow single file models most of the time, with virtually no extra setup by the end user.

t-paul, 2019-11-13:
We don't have a sandbox or secure runtime, so at this point it's a bit out of scope. It's still an interesting topic (as we don't have any clear path towards a package manager at all yet). I'll try to find some more information about how deno handles things.
In case someone want's to tackle the topic, we can always discuss and reopen.

closed 2019-11-13 by t-paul