openscad/openscad #3125: Command line failed to export images

gdlcf88, 2019-11-10:
[image]

```
Framebuffer Object extension not found by GLEW
Can't create OpenGL OffscreenView. Code: -1.
```

And then I got a zero size png file.

But it works well in visual client.

gdlcf88, 2019-11-10:
Also try ubuntu(desktop), win 10, win server 2012, got the same responses.

MichaelAtOz, 2019-11-11:
That works, with 2019.05, on my Windows 7/64 VM (VMware VM on W7, via debian via xfreerdp - with opengl32.dll in program-files - yes it is complicated...), & 2019.08.20 on Debian.
Although the two png images were at different orientations??
[image]

@gdlcf88 please see this [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?), for the info to provide. 

> Also try ubuntu(desktop), win 10, win server 2012, got the same responses.

VM's or separate boxes? Provide above info for at least one of the alternatives.

While it is unlikely to be scad code, please include demo.scad.

MichaelAtOz, 2019-11-11:
Note that the orientation on Debian in the GUI matches the Windows orientation.

gdlcf88, 2019-11-11:
The OpenSCAD version: `2015.03-x86-64.zip` `2019.05-x86-64.zip` `2019.10.25.ci3851-x86-64.zip`

The Operating System name and version: `Win10 Pro 64` `WinServer2012 R2 Datacenter 64` `Ubuntu Server 16.04.1 LTS 64(with Desktop)`

A description of the scenario that produces the issue: `Exporting png file in command line doesn't work, but it works in visual exe program.`

In case of graphics issues, the OpenGL driver information: `OpenGL version 1.1.0`

If possible, a trimmed down script reproducing the issue: 
```
@echo off 
echo import("./demo.stl"); >>demo.scad
.\openscad.com --imgsize=800,800 -o .\demo.png demo.scad
```

Yes, it is a VM from cloud provider. I found that the command line works well in my own PC.

Another clue:
[image]

MichaelAtOz, 2019-11-11:
> Warning: .... OpenCSG has been disabled

OpenCSG is what produces png output.

[Side-note: Perhaps a better cmd line warning re OpenCSG?] 

Try copying opengl32.dll from your W10 VM c:\windows\system32 (or search for it) to the directory where openscad.exe is installed.

gdlcf88, 2019-11-11:
Tried to copy opengl32.dll in system32, but got the same warning from visual exe program and the same error from command line.

Copying opengl32 in SysWOW64 didn't solve it too.

Is this caused by my low OpenGL version? But only command line can reproduce the problem, strange..

MichaelAtOz, 2019-11-11:
That fixes it in my W10 VM in Boxes on Debian.
(renaming opengl32.dll produces your error)
This is getting into technical OpenCSG details too low for me.
It could also be down to the graphics hardware on the virtual machine in the cloud.

Please provide the output from `openscad --info`.

gdlcf88, 2019-11-11:
> PS C:\Users\Administrator\Desktop\openscad> ./openscad.com --info
> OpenSCAD Version: 2015.03-2
> System information: Windows 8 (v6.2.0.0) 1 CPU 1023.61 MB RAM
> Compiler: GCC "5.2.0" 64bit
> Boost version: 1_57
> Eigen version: 3.2.5
> CGAL version, kernels: 4.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
> OpenCSG version: OpenCSG 1.4.0
> Qt version: 5.5.1
> QScintilla version: 2.8.4
> MingW build: MingW64
> GLib version: 2.44.1
> Application Path: C:/Users/Administrator/Desktop/openscad
> Documents Path: C:\Users\Administrator\Documents
> Resource Path: C:/Users/Administrator/Desktop/openscad
> User Library Path: C:/Users/Administrator/Documents/OpenSCAD/libraries
> User Config Path: C:\Users\Administrator\AppData\Local/OpenSCAD
> Backup Path: C:/Users/Administrator/Documents/OpenSCAD/backups
> OPENSCADPATH: <not set>
> OpenSCAD library path:
>   C:/Users/Administrator/Documents/OpenSCAD/libraries
>   C:/Users/Administrator/Desktop/openscad\libraries
> 
> OPENSCAD_FONT_PATH: <not set>
> OpenSCAD font path:
> 
> 
> Framebuffer Object extension not found by GLEW
> Can't create OpenGL OffscreenView. Code: -1. Exiting.
> 
> PS C:\Users\Administrator\Desktop\openscad>

MichaelAtOz, 2019-11-11:
On my W10 VM, without `opengl32.dll` `--info` gets this (tail end)
[image]

With `opengl32.dll` in the `c:\program files\openscad` directory
[image]ck

Check that the dll is there, W10 tries hard to stop you doing the copy.
I can't recall exactly what I had to do, I'll have a gogle...

gdlcf88, 2019-11-11:
[image]
I'm sure that `opengl32.dll` is in its correct path.

MichaelAtOz, 2019-11-11:
Well I'm at my technical limits. That fixed it on my VMs.
I don't know why openscad.exe is not picking-up the dll.

gdlcf88, 2019-11-11:
Thanks, so if command line doesn't work, do I have other way to make a script to export images instead of manually using visual openscad?

kintel, 2019-11-11:
The error message down indicates that you have a too old version of OpenGL to render it in preview mode.
Try doing a full render: `openscad file.scad --render -o out.png`

gdlcf88, 2019-11-11:
Hi @kintel 
[image]
I tried `openscad.exe demo.scad --render -o out.png`, it seems no errors but also no images were output.

kintel, 2019-11-11:
..and the difference is that you got no image instead of a zero-sized image?

gdlcf88, 2019-11-11:
> ..and the difference is that you got no image instead of a zero-sized image?

[image]
Yes, I got no files.

I try again, it throw an error now.
[image]

MichaelAtOz, 2019-11-11:
OK, I found out what I had to do by backtracking my browser/email history.
Details to follow shortly.

MichaelAtOz, 2019-11-11:
It was @t-paul who pointed me to the right opengl32.dll, via
http://forum.openscad.org/Linux-VM-Host-support-for-OpenGL-2-for-Windows-Client-tp27163p27168.html

Download https://packages.msys2.org/package/mingw-w64-x86_64-mesa?repo=mingw64
Extract mingw64\bin\opengl32.dll, put THAT into the openscad.exe directory.

MichaelAtOz, 2019-11-11:
The version I got was ...mesa-19.1.4-1, I'm presuming that newer version works...

gdlcf88, 2019-11-11:
[image]
The new opengl32.dll is used, but it doesn't seem to change much.
I found that openscad.exe is running. But I waited for 30 minutes and it didn't work out.

MichaelAtOz, 2019-11-11:
Try without `--render`

gdlcf88, 2019-11-11:
Coooool, Thanks for your help. @MichaelAtOz @kintel

closed 2019-11-11 by gdlcf88

MichaelAtOz, 2019-11-11:
@kintel do you think I should follow-up on the --render failure in this case?

hackwin, 2020-10-15:
Adding the following DLLs was necessary to get it working on Windows 10 Pro.

- libwinpthread-1.dll
- zlib1.dll
- libglapi.dll
- opengl32.dll

[openscad-dlls.zip](https://github.com/openscad/openscad/files/5386598/openscad-dlls.zip)