openscad/openscad #3128: Usability: Inconsistency of origins for uncentered cubes and squares

ringerc, 2019-11-12:
(Reporting some challenges I experienced when just starting with OpenSCAD, now I've had time to get more familiar with it, for consideration as possible usability or docs changes.)

In short, OpenSCAD doesn't make it simple for a user to say "give me a cylinder where this cube is" or "give me a spheroid that fits in this cube". They use different argument formats, different center/origin rules, etc.

There are 3 different behaviours available in the 3 documented primitive shapes: uncentered cube, uncentered cylinder, and centered which is the same for cube/cylinder/sphere:

* `cylinder(..., center=false)`: Still centered on `x` and `y` despite `center=false`, extends +`z`
* `cube(..., center=false)`: extends +`x`, +`y`, +`z` from origin
* `cylinder(..., center=true)`: Centered on `x`, `y`, `z`.
* `cube(..., center=true)`: Centered on `x`, `y`, `z`.
* `sphere(...)`: Centered on `x`, `y`, `z`

Additionally, the parameter specifications are inconsistent:

* A `cube` is really a cuboid: it takes a scalar for a true cube `cube(l)` or a vector of dimensions `cube([x,y,z])`. Naïve use to get a cube works as user might expect, though it projects +x,+y,+z so it behaves differently to `cylinder` and `sphere`.
* `cylinder` takes named or positional arguments, requiring that radius/diameter be specified twice if you actually want a cylinder. Naïve use to get a cylinder with `cylinder(h,r)` produces an almost-cone with a seemingly arbitrary top radius instead, x/y centered and projecting +z.
* `sphere` is always centered, unlike the other two. It only makes true spheres.
* `cube` has a fully spelled out `size=` argument, wheras `cylinder` and `sphere` take abbreviations

I strongly suggest that the tool and the documentation emphasise one consistent model for origins and parameters of all primitives:

* In the docs, consistently use named parameters and centered objects so everything gets consistent origins:
  * Always use `cube(size = x, center=true)` or `cube(size = [x,y,z], center=true)`, i.e. named parameters and centered
  * Always use `cylinder(h = z, d = x, center = true)` or `cylinder(h = z, r = x, center = true)`, i.e. named parameters and centered.
  * Avoid using the 2-diameter or 2-radius form of `cylinder` where the 1-radius or 1-diameter form will do; prefer `cylinder(h = 1, d = 2, center = true)` to `cylinder(h = 1, d1 = 2, d2 = 2, center = true)`.
* In the docs for `cylinder` note that for historical reasons the positional parameters are unintuitive and confusing to new users, so named parameters should always be used.
* In the docs for `cube` note that it's a cuboid, and the name is an anachronism that's not worth the BC pain to change
* Link to MCAD's geometric primitives in the main docs and bless them as if they were core functionality, so users don't wonder why OpenSCAD can't model a pyramid, torus, etc for them out of the box

For usability on the code level I suggest the following new interfaces to these primitives, to eventually supplant the old ones:

* `cuboid(dimensions = [x, y, z])` - always centered
* `cuboid(dimensions = d)` - same as `cube(d, centered=true)`, for consistency
* `cylinder(height = h, radius = r` *or* `diameter = d)` - unchanged, just emphasise this form in docs
* `spheroid(dimensions = [x, y, z])` - spheroid that fits within specified cuboid, centered
* `spheroid(diameter = d` *or* `radius = r)` - same as `sphere()`, convenience

but I understand that's unlikely to happen, so I'm mostly focusing on how to make the docs easier to approach for new users by using consistent coordinates.

Here's what happens right now when a user tries basics, with translucency added so you can see what's going on:

```
color("#3333FF44")
{
    cube(10);

    cylinder(10, 10);

    sphere(10);
}
```

[image]

I dunno about you, but I was surprised by that.

Here's what I expected

```
color("#3333FF44")
{
    cube(size=10, center=true);

    cylinder(h=10, d=10, center=true);

    sphere(d=10);
}
```

[image]

i.e. everything is concentric and has consistent specification of dimensions, so you can only see the cube.

ringerc, 2019-11-12:
FWIW I use a set of wrappers in my own code that handles giving everything consistent origins, and provides support for placing a cylinder or sphere in the projected from an origin for when that's convenient too.

t-paul, 2019-11-13:
That are all good suggestions, but that can't be fixed in the language itself without breaking every existing script. I've added a [pointer](https://github.com/openscad/openscad/wiki/Documentation) in the wiki in case someone tries to improve documentation.

closed 2019-11-13 by t-paul