openscad/openscad #3129: Usability: Fully contained union()ed objects or fully disjoint difference()ed objects completely eliminated from thrown-together view

ringerc, 2019-11-12:
(Submitting some issues for things that've given me trouble as I learned OpenSCAD, now I've become more familiar with it.)

When I'm working on a model I'm often `union()`ing things into it or `difference()`ing things from it. If this isn't working as expected, I rely heavily on thrown-together mode with non-100% alpha to see what might be going on.

However OpenSCAD appears to completely eliminate objects that are completely disjoint from what they're being `difference()`d from, or completely contained by the object they are being `union()`ed with, even in thrown-together view or in the wireframe view for rendered objects. Examples:

```
color("red",0.2)
difference()
{
    cube(size=10);
    
    // Visible
    translate([5,0,0])
    cube(size=10);
    
    // Vanishes completely
    translate([20,0,0])
    cube(size=10);
}
```

and

```
color("red",0.2)
union()
{
    cube(size=10);
    
    // Visible(ish); has common edge. Interior portion lost to union().
    translate([-1,0,0])
    cube(size=5);
    
    // Completely contained; vanishes
    translate([2,2,2])
    cube(size=5);
}
```

I understand that it's probably not feasible to change the behaviour as these operations fundamentally change the operated-on objects; they really are gone, or parts cut off, etc.

So instead I'm suggesting that the docs emphasise that when debugging set operations like unions, intersections, and differences, users may find it useful to:

* Enable thrown-together mode (View->Thrown-Together, F12)
* Set object `color()`s with alpha transparency e.g. `color("red",0.2)` or `color("#FF000033")`
* Use different colors for different objects where convenient, but even a `color(...) difference(...) { ...}` can be informative with an alpha set
* **Replace the set-operation with a no-op to observe the objects and their placement**, e.g if you can't tell why `bar()` isn't doing anything in:

     difference() {
       foo();
       bar();
     }

  you can:

        color("yellow", 0.2)
        {
            foo();
            bar();
        }

  and inspect; this way any wholly contained or wholly disjoint objects are retained, and you can see where objects overlap.

Furthermore as a usability enhancement I think it'd be *wonderful* if F12 view, or a new view mode closely related to it, could do this for users automatically: have turn set-ops into no-ops and apply an alpha automatically, so no objects are eliminated.

MichaelAtOz, 2019-11-12:
These (color ones) are similar to the z-fighting in the FAQ mentioned in the other issue.
Also OpenCSG preview artefacts, the order of the objects impacts which color you get to see.
OpenCSG doesn't actually know of the geometry, it simulates it, that is how it is so fast, compared to Render.

Also note the [modifiers](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters), the `#` is particularly useful for the subtracted objects in `difference()`.

t-paul, 2019-11-13:
Yes, it seems the suggestions are covered by the existing modifiers. Anything missing? Otherwise I'd close the issue.

closed 2020-01-29 by t-paul