openscad/openscad #3130: Floating point anomalies on common faces in preview mode

ringerc, 2019-11-12:
(Filing some issues for things I found challenging when learning OpenSCAD, now I've better got the hang of the basics, so they can be considered as possible usability enhancement topics).

When learning OpenSCAD I found the prevalence of weird rendering abberations very confusing. How two objects that share a common edge will render appears to depend a great deal on the exact view angle, zoom, etc. This is very confusing for new users, and I notice it happening a lot even in published files from people who seem to generally know what they're doing.

It's especially offputting when learning basic set-alike operations like `difference()`, when they just don't seem to work properly at first - sometimes your bore through a cylinder goes all the way through, sometimes it seems to have vanished or have a roof across part of it.

I *now* understand that what I am seeing is the effect of floating point imprecision causing abberations, especially in preview rendering. But I don't really see the docs touching on this much.

e.g. this trivial example:

```
difference()
{
    color("red", 0.2)
    cube(size=10, center=true);

    color("blue", 0.2)
    cylinder(h=10, d=5, center=true);
}
```

will in preview show crisscross lines of red (solid) and blue (hollow) in the top of the borehole, and on my system won't render the borehole through the solid at all even in thrown-together mode.

[image]

or without colours

[image]

but in this case at least, the final render looks fine:

[image]

It hasn't been clear to me if this is an issue limited to preview rendering or if OpenSCAD doesn't defend against floating point aberrations in CGAL operations either. So I have tended to add a floating point slop factor to ensure that the renders are sensible, and guard against surprises in final objects. It gives me decent previews too that way, e.g.

```
float_slop = 0.0001;

difference()
{
    color("red", 0.2)
    cube(size=10, center=true);

    color("blue", 0.2)
    cylinder(h=10+2*float_slop, d=5, center=true);
}
```

[image]

It's possible this is an artifact of my system's graphics that others don't experience, but I tried my partner's (admittedly even older) laptop with similar effects. I'm on Fedora 30 with Intel HD Graphics 520 using Mesa 19.1.7 with the OpenGL 4.5 Core Profile. I'm on OpenSCAD OpenSCAD version 2019.05 (`openscad-2019.05-3.fc30.x86_64.rpm`).

MichaelAtOz, 2019-11-12:
This is mentioned in the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview.3F). 
It is an artefact of how OpenCSG renders the preview of co-planar faces. See the wikipedia link on the FAQ.

While it isn't floating-point 'slop' as such, your approach is good for other reasons too.
In CSG CAD (ie the real render that GCAL does) it is good practice to avoid co-planar faces, co-linear edges, or co-indecent points.
This means for `difference()` you want the subtracted objects' faces to be either within (embeded) or protruding out of the primary object. Like your example, it is protruding out both ends, if you wanted a non-through hole, you would ensure the cylinder base was inside the bottom of the cube by a small amount.
For `union()` you may get away with butting two faces together, if they are perpendicular to an axis, but once angles get involved it will have issues. With `union()` faces should always be embedded into a joined object, e.g.

```
// trouble will follow
color("red")
  union() {
    cube(10);
    translate([10,2,0])
      cube(10);
  }

// safer
eps=0.001;
color("green")
  translate([0,20,0])
    union() {
      cube(10);
      translate([10-eps,2,0]) // embeds face into above
        cube(10);
    }
```
eps==epsilon, commonly used terminology. I use `s=` for smidgeon, less typing...

Some of these concepts may be best introduced after new users grasp the basic functionality.

t-paul, 2019-11-13:
I agree that this is an annoying problem, but so far there seems to be no automatic solution.

* [Bender](https://docs.blender.org/manual/en/latest/modeling/modifiers/generate/booleans.html) just notes _"avoid any co-planar faces (or co-linear edges)"_
* For [Rhino](https://wiki.mcneel.com/rhino/booleanfaq) (which is not exactly a cheap software package) there's pretty much the same suggestion we have in the FAQ: _"or try moving one object just a tiny, tiny bit"_

So there's unfortunately not much we can do at this point. It would be great to have a very detailed write-up of this topic for the documentation.

closed 2019-11-13 by t-paul