openscad/openscad #3131: Should assert and echo really generate CSG nodes and allow children?

thehans, 2019-11-14:
I was looking into some buggy preview rendering of intersections, and realized an issue caused by `echo` module calls.  We've been putting them into the CSG Tree as `group` nodes.  But they don't group anything.  Here is one example test of whether intersection still functions properly when an echo is there:
https://github.com/openscad/openscad/blob/0d4fa3e648b485db58b5c36a3a294113030dc87b/testdata/scad/3D/features/intersection-tests.scad#L69-L74

The problem this creates is when we call `createCSGNode` for an intersection, it checks the bounding box of left/right terms and returns `nullptr` if it determines they are disjoint, **but** there's no way to tell the difference between an actual empty geometry, and one of these fake empty groups from echo or assert.

For example, I would expect this to return an empty geometry, but it doesn't:
```
intersection() {
  union() {}
  sphere(10);
  cube(10);
}
```

So I'm proposing that echo and assert don't generate CSG nodes at all, just return nullptr.

These modules are handled in `ControlModule::instantiate` , the related code is here and would be a pretty quick change to make:
https://github.com/openscad/openscad/blob/0d4fa3e648b485db58b5c36a3a294113030dc87b/src/control.cc#L267-L281

This change would require an update most of the expected dump test results without the empty groups there anymore which I don't think is much of an issue.  And it doesn't affect any actual geometry output... except for this test: [assert-tests.scad](https://github.com/openscad/openscad/blob/master/testdata/scad/misc/assert-tests.scad) which gives children to assert and expects them to be rendered.  

So I guess its intentional behavior as originally developed, but it seems an odd use case that I don't really see much reason for, and inconsistent with echo, which "allows" children, but doesn't render them.

It also appears to be undocumented based on the manual description: 
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#assert
So I don't think it would be missed if we removed this feature?

I would sort of expect both echo and assert to give a warning or syntax error much like trying to have a child of a primitive module: `cube() cube();`  => `WARNING: module cube() does not support child modules, in file ./, line 1`

Thoughts?

nophead, 2019-11-14:
I think it is so you can insert an assert to debug something without having to add braces as well. Seems like echo should be the same. I.e. be allowed children but also render them.

nophead, 2019-11-14:
Perhaps in both cases if they don't have children then they shouldn't insert an empty node.

t-paul, 2019-11-14:
I would not restrict the child node behavior, I'd vote for allowing the child nodes to move more into the direction of unifying the behavior of functions and modules. However I don't think that means we can't get rid of the useless group nodes.

I'm wondering if it would make sense trying to change the interface between the nodes such that the child nodes don't get the `GroupNode` but instead return a list (like step 0.5 towards removing implicit unions). Initially the processing would then need to do the union resulting in unchanged behavior. With that in place it should be possible to drop the forced union in some of the place that should be safe (like `hull()`?).

It's impossible to give a syntax error from the parser, it can't have knowledge about the behavior of modules with specific names. We will need to put that into a second step (for which I want to have a visitor pattern for the AST nodes). There's an older PR by @kintel going into that direction I believe.

thehans, 2019-11-20:
Found another highlight & background modifier issue from testing this complex assembly: https://github.com/CarlosGS/Cyclone-PCB-Factory 

I ended up simplifying the issue down to this `previewChildren` example which is supposed to make it so children only show during preview, not render:
```
module previewChildren() { %children(); }
$fn=20;
previewChildren() 
  minkowski() { 
    translate([-5,-5]) cube(); 
    translate([5,5]) cylinder();
  }
translate([-5,5]) sphere(2);
```

I thought I had solved the issue of modifiers on ListNodes by passing these flags via the `State &state` parameter of `NodeVisitor::visit(...)`.  But this ends up applying to all children recursively, when we actually need to apply it to just the *immediate* children of a ListNode.

So during preview, the above example ends up showing the minkowski result as a background obejct, but also the individual minkowski components backgrounded by themselves.  F6 Render output is fine though.

`State` already had these highlight and background flags defined but I noticed they weren't being used in code anywhere.  Maybe because it doesn't work!  

I'm not sure a good way around this because child nodes are basically stored as a list of pointers to `const AbstractNode`.  
`AbstractNode` class doesn't store modifier flags and even if we added that, its still const.
Maybe we can apply the modifiers directly to the `ModuleInstantiation` objects of the ListNode children instead?

CSG dumps of above example:
(without lazy union)
```
group() {
%  group() {
    minkowski(convexity = 0) {
      multmatrix([[1, 0, 0, -5], [0, 1, 0, -5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
        cube(size = [1, 1, 1], center = false);
      }
      multmatrix([[1, 0, 0, 5], [0, 1, 0, 5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
        cylinder($fn = 20, $fa = 12, $fs = 2, h = 1, r1 = 1, r2 = 1, center = false);
      }
    }
  }
}
multmatrix([[1, 0, 0, -5], [0, 1, 0, 5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
  sphere($fn = 20, $fa = 12, $fs = 2, r = 2);
}
```
(with lazy union)
```
group() {
%  minkowski(convexity = 0) {
%    multmatrix([[1, 0, 0, -5], [0, 1, 0, -5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
%      cube(size = [1, 1, 1], center = false);
    }
%    multmatrix([[1, 0, 0, 5], [0, 1, 0, 5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
%      cylinder($fn = 20, $fa = 12, $fs = 2, h = 1, r1 = 1, r2 = 1, center = false);
    }
  }
}
multmatrix([[1, 0, 0, -5], [0, 1, 0, 5], [0, 0, 1, 0], [0, 0, 0, 1]]) {
  sphere($fn = 20, $fa = 12, $fs = 2, r = 2);
}
```

thehans, 2019-12-22:
The inconsistency between `echo` and `assert`: where `echo` allowed children but they were not rendered, was addressed as part of the lazy-union changes.  Both now allow children which get rendered.

Also both will now return ListNode if lazy-union is enabled.  I'm considering this fixed now.

closed 2019-12-22 by thehans