openscad/openscad #3132: Slashes printed to terminal if any string contains odd backslash

thehans, 2019-11-15:
When I start openscad from command line on linux(still using the gui interface though), if the file loaded has a string with escaped characters, extra slashes can be printed to the console from which openscad was started.  This only seems to happen for some (invalid?) escaped characters, so `\t` and `\n` don't seem to cause it, but `\0` and `'\s` do for example.  

I noticed this from looking at `example023.scad`, which uses `MCAD/fonts.scad`, which has many strings containing odd backslash escaped characters.
A more simplified example which also to reproduces the issue:
```
x = "\0";
cube();
```
Save to a file, then open it from openscad gui, started on command line.  After closing openscad, there will be trailing slashes in its output.

It seems that a slash occurs once for every time the file is loaded, previewed, or rendered.
Also the slashes don't appear for me until after openscad is closed, which means I guess they are buffered.

Its a pretty minor, harmless issue, just a bit strange and unexpected.

MichaelAtOz, 2019-11-15:
Happens in Windows too. The short example produces `\\` and possibly CRLF, with 2019.05.

thehans, 2020-06-19:
Closed as duplicate of #331

closed 2020-06-19 by thehans