openscad/openscad #3136: Investigate alternative Windows Installer (instead of NSIS)

t-paul, 2019-11-18:
_Moving this here from #2358 which was about a false alarm possibly triggered by NSIS._

> NSIS itself might be a problem. https://bugzilla.mozilla.org/show_bug.cgi?id=709738
> https://www.scmagazine.com/microsoft-malicious-nsis-installers-used-to-disperse-ransomware-evolve-and-grow-in-number/article/644849/

we really should be using more modern system... and signing our code somehow

- [ ] eval alternatives https://wiki.gnome.org/msitools | https://sourceforge.net/projects/osslsigncode/ | https://github.com/GNOME/msitools
- [ ] find other floss projects using this https://github.com/auth0/nodejs-msi | https://github.com/cavaliercoder/zabbix-msi
- [ ] implement experimental x-build
- [ ] test
- [ ] integrate w build scripts

martin-frbg, 2019-11-24:
How about innosetup ? http://jrsoftware.org/isinfo.php with source code available from https://github.com/jrsoftware/issrc , has a fairly readable script syntax too.

sbridger, 2020-01-13:
I'm not sure why NSIS is the problem here?

When signing  you first sign all binary executables (openscad.exe) and all DLL's, and for the paranoid all PDF documentation.
Then you build the installer, and sign the installer itself.

Then when (my) installed program runs it calls the windows service that checks its signature, and the name in the signers key (my name), and it displays a warning message if it the signature is invalid (exe has been patched/corrupted, or trust database is not available). 
This protects against someone just prepending a little extra to the front of my program and re-signing the patched exe using another valid signing key, and making a new signed installer.