openscad/openscad #3140: Disable HTML support in console window

t-paul, 2019-11-19:
This was never intended as a user visible feature, just a side effect of the GUI control supporting partial HTML. Changing the component in the last release now produces different behavior. If we want formatted output, that should be a specific feature with clear behavior, maybe using a separate output function.
Console output should write the input as given, not translate according to Qt5 for GUI mode and untranslated in command line mode.

See #1765
See http://forum.openscad.org/echoing-lt-and-gt-characters-td27978.html

MichaelAtOz, 2019-11-19:
I understand there are several well developed 'fancy' formatted things out there which thehans change will break.
I only used it as a debug aid, flagging certain messages, so don't particularly care.
But expect some screams.

thehans, 2019-11-19:
OK, I will leave it up to @t-paul for final decision / merge, so the blame can go to him :stuck_out_tongue:

kintel, 2019-11-19:
My instinct also tells me that killing this feature creep and potentially doing it properly later is preferable.

t-paul, 2019-11-19:
As linked above, we are breaking things already with no clear definition of what is supported. I think there's even a couple more tickets related to specific formatting cases.

There's requests for "fixing" `echo()` and adding formatting there anyway. So I'm thinking the only way out of this mess is to provide an actual output window and `echo()` stays as logging style thingy as it is right now.

Considering that [`{fmt}`](https://github.com/fmtlib/fmt) is going to be included in C++20 we might as well try getting that in (via external lib for now) for the formatting + think about the attributes part.

t-paul, 2019-12-20:
Looks like #3141 did not yet catch all cases. Simple `echo()` can still output HTML.

t-paul, 2019-12-20:
No, auto-build failed due to Debian already switching to CGAL5. I'll update for that separately. I think we can close this issue, latest master works as expected with the #3141 fix.

closed 2019-12-20 by t-paul