openscad/openscad #3142: Backwards compatibility with experimental or optional features

thehans, 2019-11-20:
This is a continuation of some discussion from issue #350 regarding how to handle backwards compatibility as new/experimental features are added.

Today we have 2 experimental features in the nightlies "lazy-union" and "function-literals".  We manage these through "Experimental" option checkboxes on nightly releases which are toggled by the user in GUI preferences(Qt settings), and default to disabled. They can also be enabled per command line call with 
```--enable=feature-name``` (how our tests work).  

This is useful so that users can opt-in to testing experimental code, but there are some aspects to compatibility that this scheme doesn't address.

1) User A shares a script which they developed while making use of experimental features.  User B downloads script and but is using an older release or has not specifically enabled the same feature, potentially causing errors in parsing or geometry.
2) User A wants to develop a script using a new feature, but also depends on an old library from User B which is not aware of this feature.  On the one hand, if the feature is a new builtin module, this wouldn't be a problem since the if the old library just doesn't use it.  But in the example of "lazy union", its not really an addition to the parser, but a change in overall behavior which could break this older library in insidious ways.

So the main concern is that, as more features get added, it becomes more difficult to share and re-use older code, and to even be able to tell what features or OpenSCAD version any given script relies on.

Here is one suggestion which I think would help in both of these cases:
_Originally posted by @nrdvana in https://github.com/openscad/openscad/issues/350#issuecomment-554807479_
> Best way to solve back-compat is with a opt-in directive that affects *only* the current source file.  To borrow design from Perl,
> ```
> use feature 'lazy_union';
> ```
> 
> or since 'feature' already means something in this codebase that can't be toggled on the fly (or can it?)
> ```
> use options implicit_union=false;
> ```
> 
> Then if people decide it should be the default:
> ```
> use 2020.xx;  // opt-in to default options as of version 2020.xx
> ```
> 
> 

So, instead of having the user globally enable "lazy union" in their preferences, they would need to specify per scad file.  This would make scripts self-documenting about the feature set they require.  
So when user B tries to run this on an older release, it would at least "fail fast" by giving a parser error, which would be more obvious to troubleshoot than some weird geometry difference.

Looking further ahead, lets say we do implement such a feature-enabling scheme.   Then later another release adds an experimental feature called "magic".  The releases before "magic" could at least recognize `use feature` (or whatever final syntax we decide on), and print an even more useful error message than a parser syntax error: 
> UNSUPPORTED: Feature "magic", required by file "wizard.scad" is not supported by this version (OpenSCAD 202X.XX).  You may need to update OpenSCAD to the latest  release, or even a nightly snapshot.

Other ideas for keywords that could be used: `require`, `depends`, `enable_feature`, `min_version`, `compatible`, `compatibility`.

Also I feel like using angle brackets like `use` and `include` do would help to sort of indicate that this directive applies to the whole file and should go at the top level, eg: `require <lazy-union>`.

A couple examples where fixing issues or adding features would be fairly straightforward but have been held back due to backwards compatibility concerns:
* spheres with top,bottom vertex or other alternatives: #462 and #1983 
* rotate_extrude starting on -x axis or not... #1886 
   This sort of subtle/obscure change would probably make more sense to bundle by associating them with a specific release version rather than requiring to specifically enable the feature by name.  

Overall I think this should be done in such a way that older scripts don't need to be updated when a new release changes behavior in some way, but new scripts can just say "give me all the good stuff for OpenSCAD version XXXX".

t-paul, 2019-11-20:
I think the experimental flag is the correct solution for new features that are available in a snapshot version and might be missing in the next relase.

The whole point is to mark them as unstable, so they can change while they are in experimental mode. Encoding that into scripts does not mean much then. Which state of the evolution does the script require? We really don't want to have even more details tracking of those features. The point of making them experimental is to be able to change them.

A notation inside the script like described above was discussed and we may need that at some point when we actually need to release biger features that are not backward compatible. The preferred way should be to enable things so that the compatibility does not break while users can opt-in to new features via some imported scad code. So a different sphere can already be done using a script (as it does not require the use of the built-in sphere in the code replacing the built-in).

thehans, 2019-11-20:
> The whole point is to mark them as unstable, so they can change while they are in experimental mode. Encoding that into scripts does not mean much then. 

I think it still helps to clarify when an experimental feature is being relied on by any given script.

> Which state of the evolution does the script require?  We really don't want to have even more details tracking of those features.

For experimental features that are still in flux, I think its fine to just say the latest nightly version is the only relevant implementation.  Users of experimental features can expect that that specific feature's behavior might change.

> The point of making them experimental is to be able to change them.

That's fine, but what do we when we decide to promote "lazy union" from "experimental" to "stable"?  I would say it should then be made "optional".  Otherwise we risk potentially breaking scads of older scads.  I feel like this problem leads to indecision and unwillingness to make changes to releases in general.

> The preferred way should be to enable things so that the compatibility does not break while users can opt-in to new features via some imported scad code. So a different sphere can already be done [...]

This general approach feels more like a stopgap than a true solution.  It requires new users to hunt down these obscure snippets just to get what they expected the correct result to be.  I will concede that alternative sphere geometries like (icosahedral etc.) are probably best left to userspace.  But the original builtin sphere implementation(no center vertex on top/bottom) was a mistake in my opinion, which should be fixed.  How could something like this be changed while keeping backwards compatibility?  Specifying release version features (eg: `using OpenSCAD XXXX.XX`) is one way that I think could solve this.  And if a script does not specify one, it could default to whatever feature set existed before such "compatibility versions" were implemented.

I'm not necessarily saying that *all* new features would need to be optional, but in cases where backwards compatibility is holding back overall progress I think trying a new approach like this would help.

t-paul, 2019-11-20:
Unfortunately `using OpenSCAD xxx` does not really help. General rule of versioning like that is if you missed the train to mark version 1, you are pretty much out of lack to catching that train ever again.
Suppose you declare a new openscad version and then include an old file which does not have any version marker. What now? Assume the other one is the old version? We have to, otherwise that script will break. But... a user writing the new script just relied on the latest behavior, not bothering to add the using for the new version because it's just working fine. So we can't make the new behavior the default, which is not good.

Anyway, I think that specific topic is not so much about the backward compatibility but about how code sharing works. It's only cumbersome because we don't have an established simple way to discover, get and import external scripts.

Also for the lazy union, looking at other existing feature requests, I'm not so sure we really need that flag. So people would like to export separate parts of the model to different files. Or separate parts should go to different layers / constellations in one single file. That's pretty much asking for a declarative way to specify what is not unioned _at top level_.

So something like (stealing the c++ attribute syntax just for a change)
```
module mydesign(flag) { ... }

[[layer("base")]]
mydesign(flag = 1);

[[layer("top")]]
mydesign(flag = 2);
```
could declare where to not do the top level union plus even give the export the information to output separate volumes or give the UI the opportunity to ask if the parts should be exported to a folder instead with the given names as base file name.
Due to being explicit, it does not create a backward compatibility issue.

As I said before, I'm totally for something like the `using` directive if we need that. But it really should not be an easy escape for everything. We don't want to have scripts starting with 20 `using this` and `using  that` just to enable all the new features in a couple of years. It's very desireable to evolve the default behavior so simple things can stay simple.

nrdvana, 2019-11-21:
> Assume the other one is the old version? We have to, otherwise that script will break. But... a user writing the new script just relied on the latest behavior, not bothering to add the using for the [...]

Well, yes, if you are committed to backward-compatibility you can't ever change the default behavior of something that was already being used unless you have the source file opt-in to using it.

It doesn't have to be lots of lines of `using` though.  You can split those out for the initial testing, and then when there is consensus on which ones everyone wants to keep, roll them into a version number. 

Using the perl example, in Perl 4, using a non-exiting variable name was perfectly fine.  Then they decided it was such a source of bugs that in version 5 they added `use strict;` which makes referencing an un-declared variable a fatal error, and everyone had to write that line at the top of their scripts to get the new behavior.  Later on they made `strict` the default if you say `use 5.12.0;`.  So, everyone needing new features writes a single line of boilerplate at the top of their script and gets the package of new features and everyone else experiences seamless backward compatibility.  There are dozens of other changes to Perl guarded by these same kinds of flags (`use feature 'signatures`), and some eventually get promoted into the new defaults for some specific version number.

https://perldoc.perl.org/feature.html

Javascript is headed the same direction - they added a strict feature guaraded by `"use strict";` (a string written as a statement).  They can't change the default without breaking millions of websites, so every javascript source file who wants strict behavior has to opt-in.  Sadly they didn't have a keyword available to extend, so they had to use that clumsy looking string constant.

Part of the idea of starting with the `use` keyword (for OpenSCAD, but was also true for Perl) is that you don't need to add a new keyword.  Any time you start adding new keywords you run the risk of breaking someone's program if they used that as a variable name or something.

The OpenSCAD experimental features could be added to a system like this, or left separate.  It appears that OpenSCAD development has treated features differently than toggleable-opt-in-features, and this suggested change would require them to be toggled on the fly during parsing (as the parser included other files) so probably better to leave it separate.  That's why I suggested `use options ...`.  Also it might be neat to have NAME=VALUE options instead of just boolean flags.

Comparing to Perl again, they did actually make some experimental features usable like this.  They came out with `use experimental 'smartmatch';` and then two versions later they decided it was problematic and should be deprecated and then removed.  So, now every script that enables smartmatch prints a warning about the deprecation.  Eventually they will remove it and those script will be broken, but that's ok by their official policy since it was part of "experimental".

MichaelAtOz, 2019-11-21:
I'm unsure whether this is proposing to have a specific OpenSCAD.exe A) modify its behaviour depending on the `require <version|feature>`* , or B) JUST to produce an error.
\* (please kill `using` or `use`- to close to `use <>`)

e.g. new feature, spheres are shaped like pumpkins, released with 2019.10.31
A)
```
require <pumpkins>
sphere(10);
```
Produces pumpkins, but just
```
sphere(10);
```
Produces normal spheres, the same .exe has two behaviours.
or
B)
Using a version < 2019.10.31,
```
require <pumpkins>
sphere(10);
```
> WARNING: What the hell are pumpkins - you probably need a later OpenSCAD version.

and produces a normal sphere.

I don't think A) is a good strategy.

MichaelAtOz, 2019-11-21:
Also for things like changes to the face layout of spheres, wouldn't something like `sphere(10, type=2)` be better than global changes?

nrdvana, 2019-11-21:
@MichaelAtOz except what about the person who writes a `.scad` module that assumes spheres are spheres, then sends that to someone with a newer copy of OpenSCAD, and they write a script that requires <pumpkins> (for the bit of code they added) but call out to the module, and the module generates pumpkins instead of the expected spheres?  I would argue that the newer OpenSCAD.exe must generate both behaviors in order to not break for someone.

Meanwhile, taking a new `.scad` making use of new features and running it on an old OpenSCAD.exe would generate an error, because there's no way to solve that, and people need to be told to upgrade.

thehans, 2019-11-21:
> the same .exe has two behaviours.

Sure, when given two different inputs.  I don't really see how that in itself is a problem;

`cube(10);` and `sphere(10);` are also two different inputs.  

And I would suggest, even for a ridiculous joke feature, that at least the feature naming would indicative about what behavior is affected.  So instead of ambiguous `require <pumpkin>` it could be more like `require <sphere-pumpkin>` or `require <sphere-style=pumpkin>`

MichaelAtOz, 2019-11-21:
That basically is scope dependent behaviour, with potentially multiple behaviour changes for the same syntax. You cut/paste some code and it behaves differently.

ATM it seems to be file scope, but to do it properly you need inline `requires <>`, so you could do it e.g at `module()` level, say you want to use (cut/paste) a module from a file from Thingiverse with `requires <type1>` into your new program main file with `requires <type2>`, you do
```
requires <type2>
...
module thing() {
  requires <type1>
  ...
}
```
 or even 
```
translate() 
  requires <sphere-style=pumpkins> 
    some_spheres(); // doesn't know about pumpkins but this forces them
``` 
Where `some_spheres()` is inside an `include <>`, which may have various `requires <>`.

I can't see good in this, let alone the wiki complexities of documenting it.

nrdvana, 2019-11-21:
The good that came of it in Perl is that you can take a 25-year-old script and run it with the latest interpreter and get the exact same behavior.  You can also add one line of boilerplate to the start of your script to get a bunch of new features, even while including libraries written decades ago.

But, it's very true that it requires more development effort to maintain this.  Though in general, you can just pile on more test cases for the new behavior and let the old test cases check the old behavior.

And certainly, this is only for the features that *can't* be un-ambiguously merged into the existing syntax for back-compat reasons.  Well, I mean ``extrude() for(x=[1,2], union=false) ...`` fits in the existing syntax, but it would be ugly to keep writing that every time.

t-paul, 2019-11-21:
Coming back to the specific topic about lazy union for a bit, the forum post lists 4 cases to consider. I think we should make an effort to see if there's a strategy that will cover most if not all of the real world scenarios.

* Top Level union -> check via bounding box (unless we have specific annotations in addition as mentioned above), and in case of overlaps, union the overlapping objects. Export should be able to handle the remaining, non-overlapping objects

* Difference -> I'm thinking we may want to keep the implicit union of the first module instantiation, this is a special case as it's not commutative and it could be helpful to statically see what "the first operand" is, instead of relying on the dynamic object count of the first call.

* Intersection, Minkowski -> Maybe it would help to find actual cases which would result in different geometry?

revarbat, 2019-11-22:
We already have the concept of special reserved variables like `$vpr`, `$fa`, `$fn`, and `$fs`.  Wht not have a `$feature_lazy_unions=true;` that every `union()` etc can pay attention to?  That way, if you use an old library which it is a problem for, you can turn it off for those calls by setting it to false.  Even turn it off for whole code blocks.

t-paul, 2019-11-22:
`$` variables are special by being dynamically scoped. This makes them especially bad for defining things like that as they would automatically be observed in old code instantiated by newer code which has the variable set.

revarbat, 2019-11-22:
```openscadinclude <newerlib.scad>
include <oldlib.scad>
$lazy_unions = true;

revarbat, 2019-11-22:
I was thinking something like this:
```openscad
include <newlib.scad>
include <oldlib.scad>
$lazy_unions = true;

newlib_module_a(size=[1,2,3]);
oldlib_module_b(r=10,l=23,$lazy_unions=false);

union(){
    $lazy_unions=false;
    oldlib_module_c("foo");
    oldlib_module_d(a=8,b=5);
}
```

t-paul, 2019-11-22:
And it still would break with use of `children()`. Also if that's the required style, I don't want to image how it looks once we have > 3 of those flags.

revarbat, 2019-11-23:
I would think that `children()` would respect the variable as well, but you make a very good point about multiple flags.

We really do need a way to turn this feature on/off in-language, though. Otherwise we'll have to keep going into preferences to turn the feature on and off.  I have dozens of old OpenSCAD scripts sitting around that I periodically revisit.  I'm also developing a rather major library that I would like to use newer features in. Switching features on and off gets annoying pretty fast.  My current solution is to keep two different versioned OpenSCAD binaries around.

MichaelAtOz, 2019-11-23:
I've been thinking of libraries as well, particularly with object/function literals and how best to package it all, perhaps this topic should not be _solidly_ locked in until we get some experience with the whole bundle.

t-paul, 2019-11-25:
That's the whole point of the experimental flag. It means there may be changes, even breaking changes to earlier versions without additional warning. While the feature is tagged experimental it would automatically disable in a release build.