openscad/openscad #3143: $children is not computable?

thehans, 2019-11-23:
This is a problem related to work being done for lazy union, where we would like to be able to loop over individual children from a ListNode, when that ListNode is instantiated via `children()` from a user-defined module.

Here is my original example:
> Giving a `for` loop as the child in a module call, results in `$children==1` rather than one for each iteration.
> ```scad
> // Note: Enable "View -> Animate"
> 
> // "particle" module with short name
> module p() { square(); }
> 
> module explode() {
>   echo($children);
>   for(j=[0:1:$children-1]) {
>     // angle is multiple of golden ratio to get "sunflower" pattern
>     a = j*1.61803398875*360; 
>     translate(j*$t*[sin(a),cos(a)]) children(j);
>   }
> }
> 
> // single child, does not move
> explode() for(i=[0:1:200]) p();
> 
> // vs. specifying each child (works as intended, with or without lazy union)
> /*
> explode() {
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
>   p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();p();
> }
> //*/
> ```
_Originally posted by @thehans in https://github.com/openscad/openscad/issues/350#issuecomment-554828820_

So, the problem is OpenSCAD currently bases `$children` off of the number of `ModuleInstantiation` under the call to a user defined module.  In this case the `explode()` call has 1 child which is `for(i=[0:1:200])`.   
What we'd like to be able to do is actually loop over the list of nodes created by this `for` loop.  

So in order to know the value of `$children`, it seems we would have to evaluate the children's `ModuleInstantiation`: `for(i=[0:1:200])` *before* the `UserModule`: `module explode()`.  

Its not clear to me how special (dynamically scoped) variables "should" work in conjunction with `children()`
It looks like the existing behavior is for node to pick up the scope from where `children()` is called, and not where the corresponding child node is instantiated:
```
module a() { $foo = 1; children(); } 
module b() { echo($foo); }  
group() {
  $foo = 0;
  a() b(); // ECHO: 1
}
```
If we want to keep that functionality, then I don't think it is possible to compute the number of children, since that could depend on a special variable from the body of `module a()`, which could in turn possibly depend on value of `$children`.  Its a chicken and egg issue here.

If we decide instead that the expected behavior is for `module b()` above to echo "0", then I think it would at least be theoretically possible to calculate `$children` before evaluating the body of its parent module.  Still probably won't be easy though.

Any other ideas for how this could/should work?

nophead, 2019-11-23:
I can't see how you can change how dynamic scope works, or what alternative description it would need to echo "0".  A common use of $variables is to pass arguments to children.

MichaelAtOz, 2019-11-23:
That's how it's supposed to work, if you want 0 then you do
```
module a() { $foo = 1; children(); } 
module b() { echo($foo); }  
group() {
  a() b($foo = 0); // ECHO: 0
}
```
Not helping re $children, sorry...

thehans, 2019-11-24:
> I can't see how you can change how dynamic scope works, or what alternative description it would need to echo "0". A common use of $variables is to pass arguments to children.

The unique and confusing thing about `children()` is that it seems have two possible scopes:
1) where `children()` is literally called and 
2) where the corresponding child module is called

Either of which could be considered "the actual legitimate scope", without necessarily changing the meaning of dynamic scope, in my mind at least.
But I guess you're right, changing it would leave no way to pass arguments down to children in that way.

thehans, 2019-11-24:
But the passing of arguments down to children, and the circular dependency that causes when the number of children is dependent on such an argument, is the crux of the issue.

@t-paul suggested in IRC that one way around this would be to leave `$children` evaluating to 1 as in my original example, but to be able to assign the geometry result to a variable.  

I guess this relates to Geometry Objects from #3088 (I haven't thoroughly read through that yet).  So you could get a list of objects from `x = children();`, then x would be a vector of Geometry objects which could be looped over.

t-paul, 2019-11-25:
Yeah, I'm not yet sure that actually solves the issue, or if it just moves it to a different place. Unfortunately I fear it's not solving the circular dependency.

nophead, 2019-11-25:
Seems like a sledgehammer to try to fix an obscure edge case. Can we just
document that if you try to use the result of $children to change the
number of children it won't work. Or better still make it an error somehow.

thehans, 2019-11-25:
> Can we just document that if you try to use the result of $children to change the number of children it won't work. Or better still make it an error somehow.

Its not just that obscure edge case, but more like being able to define some non-paradoxical order of evaluation.  In the original example, the `explode()` module needs to resolve `$children` before it can even start evaluating its own loop.  So as far as I can tell, this mean that if we expect `$children` to be accurate (in that its not just the static # of AST nodes), then there is no way to pass down different dynamic-scoped values per child from `explode()`s loop, because the children would have already been instantiated.

nophead, 2019-11-25:
I see, yes a big problem.