openscad/openscad #3147: DXF is derived with precision error

finale-z, 2019-11-29:
I created a rectangle in the drawing and rotated it by 135 degrees, but the Angle becomes 134.9999 after being exported to the DXF drawing, and the two sides of the rectangle are not parallel anymore, with a deviation of 0.0002 degrees.How can I solve this problem?

nophead, 2019-11-29:
When you rotate by 135 degrees the vertex coordinates become irrational numbers which can't be represented exactly with floating point numbers, so yes the sides will not be perfectly parallel, etc.

What are the actual coordinates in the DXF file?

t-paul, 2019-11-29:
If the question is only if this can be completely fixed, the answer is: no. I believe there's a way to specify the precision in the output file, so that would not get rid of the error, but higher precision might reduce it.

If you would like to keep the issue open for looking at the output precision, please amend the title accordingly.

finale-z, 2019-12-02:
The current accuracy is quite sufficient for machining, except that it is a little uncomfortable to see the error of 0.0001 degree when looking at the drawings.Thank you for your answers.

closed 2019-12-02 by finale-z