openscad/openscad #3148: Allow setting export filename in code, based on variables.

UrielGuy, 2019-11-29:
This is a feature suggestion. 
A good example might be: 
`
WIDTH=10;
HEIGHT=30;

$filename = concat("model_", str(WIDTH), "x", str(HEIGHT));
`

When exporting a file, the filename will default to the value of $filename set at the last render.
An even better option is if we get an actual string formatting functionallity for openscad.

MichaelAtOz, 2019-11-29:
For string formatting see https://www.thingiverse.com/thing:526023, I haven't used it but it looks OK.

t-paul, 2019-11-29:
An imperative way of defining file names does not fit how OpenSCAD is processing scripts and it would not easily allow multiple file names. See https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases for some ideas how to add this information in a declarative style.

closed 2019-11-29 by t-paul