openscad/openscad #3150: GUI crash with CGAL asertion

MichaelAtOz, 2019-11-30:
Combining details from Forum & #2351

I'm running version 2019.11.19.nightly (git eedfcc4) on Ubuntu. The error message is the one when rendering with F6 from the GUI ; 

the following two lines:

    
polyhedron(points=[[1,0,0],[0,0,0],[-0.1,1,0],[0,2,0]],faces=[[0,1,3,2]]);                                           
     cube([1,1,1]);                                                                                                       

give the following output (with version 2019.11.19.nightly):

CGAL error: assertion violation!
Expression : itl != it->second.end()
File       : /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line       : 1152
Explanation: 
Refer to the bug-reporting instructions at
http://www.cgal.org/bug_report.html
Aborted (core dumped)

for the command line I get (suprisingly) a distinct error:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 227

but the exit code is 0 and an STL is created, with the cube.

If ou don't reproduce the "windows closed" part with this nightly build, I can try to provide more details (maybe I changed some setting).

_Originally posted by @tturpin in https://github.com/openscad/openscad/issues/2351#issuecomment-560017797_

t-paul, 2019-11-30:
Might be #2847?

MichaelAtOz, 2019-11-30:
That was quick, _possibly_ diagnosed while I was reformatting the OP...

MichaelAtOz, 2019-11-30:
My edit didn't make it to the mailing list, so posting this bit again.

@tturpin, I'm not a linux guru, but we have some here.
It looks like the GUI is not catching the exception, I get ERROR: CGAL error in in the GUI console without loosing the window.
Please post the output from Help/Library-info or openscad --info.

tturpin, 2019-12-01:
Here is the output of openscad --info (removing font path)
```
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2019.11.19.nightly (git eedfcc4)
System information: Linux 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018 x86_64 Ubuntu 18.04.1 LTS 4 CPUs 15.53 GB RAM
User Agent: OpenSCAD/2019.11.19.nightly (git eedfcc4) (Linux x86_64; Ubuntu 18.04.1 LTS)
Compiler: GCC "7.3.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.5
QScintilla version: 2.10.2
InputDrivers: 
GLib version: 2.56.1
lodepng version: 20180910
libzip version: <not enabled>
fontconfig version: 2.12.6
freetype version: 2.8.1
harfbuzz version: 1.7.2
lib3mf version: (not enabled)
Application Path: /usr/bin
Documents Path: /home/tifn/.local/share
User Documents Path: /home/tifn/Documents
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/tifn/.local/share/OpenSCAD/libraries
User Config Path: /home/tifn/.config/OpenSCAD
Backup Path: /home/tifn/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/tifn/.local/share/OpenSCAD/libraries
  /home/tifn/3d/openscad

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  [...]
GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 18.0.5
GL Renderer: Mesa DRI Intel(R) HD Graphics 530 (Skylake GT2) 
GL Vendor: Intel Open Source Technology Center
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018
Machine: x86_64
```
The assertion mentioned in #2847 is commented in my polygon_mesh_to_nef_3.h so this is definitely a duplicate of this  #2847.

MichaelAtOz, 2019-12-24:
Note, the two CGAL ERROR: assertion violations above (with CGAL 4.11) are different to #2847 which is,
 `pe != 0`, the commented out assertion in CGAL 4.11.
This is a different flavour.
GUI OpenSCAD does not catch the assertion `itl != it->second.end()` and aborts, but command line does catch, but a different assertion to the GUI `target(s1).vertex()==target(s2).vertex()`.

thehans, 2020-12-31:
Fixed via #3526.

closed 2020-12-31 by thehans