openscad/openscad #3151: Code hinting shortcut?

OnkelTem, 2019-11-30:
I've just installed openSCAD for the first time and I wonder how can I get those cute hints on function parameters? They appear when I edit the code, but after that I couldn't find a shortcut to show them again. And if there is a shortcut (I hope it is) it would be great to have it here: https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts

thehans, 2019-12-10:
They are triggered by opening parenthesis on a builtin function/module.  If you go back and edit a call, they won't show again unless the `(` is deleted and re-typed.

OnkelTem, 2019-12-11:
@thehans That I know, but how to trigger it manually? For example, in Intllij products I just hit `Alt+p`

closed 2020-02-16 by t-paul

t-paul, 2020-02-16:
CTRL+Space for autocomplete, SHIFT+CTRL+Space for calltips (function parameters).

commit a2c8b92b (2020-02-16): Add key shortcuts for autocomplete and calltips (fixes #3151).