openscad/openscad #3153: Export as Image: Chinese filename unreadable codes

sonichy, 2019-12-07:
Windows 7, OpenSCAD 2019.5
Save to 一层3a.png but get һ��3a.png

Mailing list fail:
This message was created automatically by mail delivery software.
A message that you sent could not be delivered to one or more of its
recipients. This is a permanent error. The following address(es) failed:
discuss-request@lists.openscad.org
Messages from this domain are blocked.

t-paul, 2019-12-07:
duplicate of #2658.

closed 2019-12-07 by t-paul

t-paul, 2019-12-09:
@MichaelAtOz do you know if it is possible to only use the forum if the mailing list part is blocked? Looks like the ISP is blocking the mail delivery.

MichaelAtOz, 2019-12-09:
@t-paul  No, the forum sends 'draft' posts to the mailing-list, the forum is subscribed to the list, when the forum post is received again it makes it an actual post.
I'm not sure if that message means everything is blocked, I'll do a test. Could be the china firewall maybe.

MichaelAtOz, 2019-12-09:
@t-paul Looks like it works at my end.
@sonichy You may need to check with your ISP about that email block.