openscad/openscad #3154: polygon multiple paths incorrect, or documentation misleading?

thehans, 2019-12-07:
From https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#polygon :
> **paths**
> - ...
> - multiple vectors
>    - Creates primary and secondary shapes. Secondary shapes are subtracted from the primary shape (like difference). Secondary shapes may be wholly or partially within the primary shape.
> 

This makes it sound like any secondary paths act as purely "negative" objects, but in fact they are all XOR'd together.

Not sure if the documentation should be updated to reflect the actual behavior, or if the behavior should be changed.
One issue with this behavior is when they partially intersect, you can get surfaces that meet at a single edge which I don't believe is manifold

```
// Example of partial intersection resulting in non-manifold geometry
polygon(
  points=[
    [-1,-1],[-1,1],[1,1],[1,-1],
    [0,-2],[0,2],[2,2],[2,-2]
  ], 
  paths=[[0,1,2,3],[4,5,6,7]]
);

// Example where second path is completely disjoint from first. 
// If it were subtracted from first, then it would not display at all
translate([0,4])
  polygon(
    points=[
      [-1,-1],[-1,1],[1,1],[1,-1],
      [2,-1],[2,1],[4,1],[4,-1]
    ], 
    paths=[[0,1,2,3],[4,5,6,7]]
  );
```

thehans, 2019-12-07:
Also self intersection does the same sort of XOR thing:
```
polygon([[-1,-1],[-1,1],[1,1],[0,-0.5],[0,0.5],[1,-1]]);
```

I think we are defaulting to  "Even-Odd" [PolyFillType](http://www.angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Types/PolyFillType.htm) when calling Clipper, which could be changed to something more appropriate if we decide the behavior should be different.

Although maybe its considered a feature for ex if you want an island inside of a hole, etc.
```
polygon(points=[
  [-4,-4],[-4,4],[4,4],[4,-4],
  [-3,-3],[-3,3],[3,3],[3,-3],
  [-2,-2],[-2,2],[2,2],[2,-2],
  [-1,-1],[-1,1],[1,1],[1,-1]],
  paths=[[0,1,2,3],[4,5,6,7],[8,9,10,11],[12,13,14,15]]
); 
```
So "fixing" the other issues above would require dropping this "feature" I think.

thehans, 2019-12-16:
Any thoughts?  I'm curious what other's would expect behavior to be in this situation.

mMerlin, 2020-11-22:
I found this issue when coming to report that either the documentation was wrong, or the code was for exactly this case. Since the documentation (currently) says `like difference`, I was expecting (and wanting) that with initialization of:

```txt
tst_coords = [
    [0,0], [0,20], [10,20], [10,0],
    [-1,2], [-1,4], [30,2], [30,4],
];
path1 = [0, 1, 2, 3];
path2 = [4, 6, 7, 5];
```

```txt
polygon(tst_coords,[ path1, path2 ]);
```

would produce the same rendering as

```txt
difference() {
    polygon(tst_coords,[ path1 ]);
    polygon(tst_coords,[ path2 ]);
}
```

My project wants true difference, not xor. The test sample is (of course) much simpler than the case the prompted that research. It seems like it is going to take some 'twisting' to get the functionality I want from the current project state. adding extra paths to the passed parameters is easy. conditionally doing a difference, not so much.

mMerlin, 2020-11-22:
Here is a work around, to get the *difference* functionality described in the documentation. Once I walked through a couple of sequences, it turned out to not be as messy as I expected. Each of these handled all of the cases I threw at them. That was not extensive, but should have covered the *normal* situations.

```c
module my_polygon(points, paths, convexity=undef)
{
    let(
        have_path = is_list(paths),
        num_path = have_path ? len(paths) - 1 : 0
    ) {
        if (num_path > 1) {
            difference() {
                polygon(points, [paths[0]], convexity);
                for (pth = [1:num_path]) {
                    polygon(points, [paths[pth]], convexity);
                }
            }
        } else {
            polygon(points, paths, convexity);
        }
    }
}
```

Or the shorter, but (to me) not as obvious

```c
module my_polygon(points, paths, convexity=undef)
{
    let(
        have_path = is_list(paths),
        last_diff = have_path ? len(paths) - 1 : 0,
        first_path = have_path ? [paths[0]] : paths
    ) {
        difference() {
            polygon(points, first_path, convexity);
            for (pth = [1:1:last_diff]) {
                polygon(points, [paths[pth]], convexity);
            }
        }
    }
}
```